olikraus/u8g2 #1815: Crox Cyrillic fonts dont print Cyrillic...

andreydk1981, 2022-03-17:
Hello. I'm trying to display text in Cyrillic using fonts from the Crox group: Cyrillic fonts. None of the fonts print Cyrillic characters. Fonts used with _tf ending (for example u8g2_font_crox1hb_tf).

I tried other Cyrillic fonts (for example u8g2_font_6x13_t_cyrillic) - everything works fine - Cyrillic is displayed on the screen.

What could be the problem?

olikraus, 2022-03-18:
Crox fonts are encoded according to ISO 8859-5 https://en.wikipedia.org/wiki/ISO/IEC_8859-5
All other fonts with cyrillic glyphs are encoded according to unicode https://en.wikipedia.org/wiki/Cyrillic_script_in_Unicode
For u8g2 it means, you need to turn off unicode suppport or use the draw string function.

andreydk1981, 2022-03-20:
I am make some function... it's work

void PrintRusStr(int x, int y, String source) // печатает русскими буквами
{
  int i, k;
  String target;
  unsigned char n;
  char m[2] = {'0', '\0'};

  k = source.length();
  i = 0;

  while (i < k)
  {
    n = source[i];
    i++;

    if (n >= 0xC0)
    {
      switch (n)
      {
      case 0xD0:
      {
        n = source[i];
        i++;
        if (n == 0x81)
        {
          n = 0xA8;
          break;
        }
        if (n >= 0x90 && n <= 0xBF)
          n = n + 0x30;
        break;
      }
      case 0xD1:
      {
        n = source[i];
        i++;
        if (n == 0x91)
        {
          n = 0xB8;
          break;
        }
        if (n >= 0x80 && n <= 0x8F)
          n = n + 0x70;
        break;
      }
      }
    }
    m[0] = n;
    target = target + String(m);

    u8g2.drawStr(x, y, target.c_str());
  }
}

closed 2022-04-03 by olikraus