olikraus/u8g2 #1816: Support for SSD 1320 160*80

AristotleJunior, 2022-03-17:
I've got a new OLED screen with chip SSD 1320 and dimension 160*80, and can't find relative config in u8g2, and I tried to work it out by myself but failed, so woud you please help me supporting this screen
[SSD1320.pdf](https://github.com/olikraus/u8g2/files/8287072/SSD1320.pdf)

AristotleJunior, 2022-03-18:
both I2C and SPI
I will really appreciate your help

olikraus, 2022-03-18:
reference: #1569 #1351

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1320.c

Maybe I find some time on the weekend to work on this.

olikraus, 2022-03-19:
I will order this display: https://de.aliexpress.com/item/1005003510267760.html

AristotleJunior, 2022-03-21:
Yeah, that is exactly my display.
Looking forward to your good news

olikraus, 2022-03-21:
Shipping delay is at least 4 weeks...

olikraus, 2022-04-17:
[image]

Constructor looks like this: 
`U8G2_SSD1320_160X80_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);`

I have created beta 2.33.2 for u8g2...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

AristotleJunior, 2022-04-21:
Thank you for your great work. I really appreciate it. I wish u8g2 to become better and better.

closed 2022-04-21 by AristotleJunior