olikraus/u8g2 #1817: Why can u8g2 be slow inside an app?

shlyakpavel, 2022-03-21:
I included u8g2 into my app to display info. The app does nothing special, just a couple of sleeping Qt threads.
However, u8g2 becomes extremely slow in terms of displaying buffer. I use hardware SPI & linux. It takes about 2 seconds for sendBuffer() to finish. The unicode text or graphics are displayed line by line, like animated. The same code runs fine in a separate non-multithreaded demo.

olikraus, 2022-03-21:
Might be a problem of the SPI driver. I once had a problem with rasPi where root access was required. Problem is, that any access to SPI resources in memory are very slow from user space.

olikraus, 2022-04-03:
no feedback, closing

closed 2022-04-03 by olikraus