olikraus/u8g2 #1822: Mystery Quest font not declared in this scope

matsutakehoyo, 2022-03-27:
Hi fist of all thank you so much for the superb library!

I am building a project with an OLED in an Arduino (Pro Micro). While trying different font options I encounter an error when trying to use Mystery Quest

for example when I try to 

`  u8g2.setFont(u8g2_font_mystery_quest_48_tn); `

I get the message 'u8g2_font_mystery_quest_48_tn' was not declared in this scope

My code works with other fonts such as u8g2_font_osb41_tn.
Is there something else I need to do to use this font family?

olikraus, 2022-03-27:
Yes, that is kind of a problem with github wiki. mystery_quest will be part of the next release, but documentation is already online.

You can use the latest beta instead...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

matsutakehoyo, 2022-03-27:
I updated the library and it worked, Thank you!

closed 2022-03-27 by matsutakehoyo

olikraus, 2022-03-27:
:)