olikraus/u8g2 #1828: Has anyone gotten the STST7528_NHD_C160100 display to work?

HavocRC, 2022-04-05:
I've done as much research and troubleshooting as I can before posting here. I haven't seen anyone else using this display, so I wasn't sure if it was set up correctly - I saw a thread from a few years ago and it looks like it was in development at the time, never got any updates on it. I'm assuming I'm doing something wrong. 

Here's the display I'm using: https://newhavendisplay.com/content/specs/NHD-C160100DiZ-FSW-FBW.pdf

Wired up as such:
 
```
Disp || Arduino Uno
SDA --> A4 (4K7 pullup to 3.3V)
SCL --> A5 (4K7 pullup 3.3V)
Vin --> 3.3V
RST --> 1K pullup to 3.3V
CSB --> 1K pulldown
```

I'm not sure what to do with CSB and RST as I2C is supposed to only use 2 wires. I used the [I2C scanner](https://newhavendisplay.com/content/specs/NHD-C160100DiZ-FSW-FBW.pdf) and found that with the current pinout, I am able to read an address, so I2C is definitely working.

Here is the code I'm trying: 

```
#include <Arduino.h>
#include <Wire.h>
#include <U8g2lib.h>

U8G2_ST7528_NHD_C160100_1_HW_I2C u8g2(U8G2_R0);

/* u8g2.begin() is required and will sent the setup/init sequence to the display */
void setup(void) {
  u8g2.begin();
  u8g2.setI2CAddress(0x3F);
}

/* draw something on the display with the `firstPage()`/`nextPage()` loop*/
void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(0,20,"Hello World!");
  } while ( u8g2.nextPage() );
  delay(1000);
}
```

I'm not sure if I'm using the library wrong, using an incorrect constructor, or if this particular display hasn't been verified as it seems quite obscure. Thank you very much for any help.

olikraus, 2022-04-05:
The client address is wrong, you need to multiply it with 2: https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress

(This is an often discussed topic, but whatever i would have implemented in u8g2, people would discuss whether the current u8g2 implementation makes sense or not)

HavocRC, 2022-04-05:
I did read that part but _glossed right over the multiplication by 2._ Thank you for the extremely fast reply I will give it another go with 0x7E as soon as I get home today.

HavocRC, 2022-04-05:
Made some headway! I connected reset from the display to my Uno and with the new address (CSB still pulled LOW), I'm seeing something....The screen is just flickering all black on/off at a rate slow enough to obviously tell it's flickering. 
My new constructor: `U8G2_ST7528_NHD_C160100_1_HW_I2C u8g2(U8G2_R1,11);`

I've tried full display tests, hello world tests from both full_buffer and page_buffer, along with 1_HW/2_HW page buffers, and all the results are the same, the screen just flickers everything on/off.

Triple checked circuitry. Any ideas?

olikraus, 2022-04-06:
Maybe try the SW_I2C constructor. The SW_I2C emulation is more reliable sometimes. Maybe bus speed is too high at the moment.

HavocRC, 2022-04-06:
Same outcome :(

olikraus, 2022-04-06:
I think support was added here:
https://github.com/olikraus/u8g2/issues/986

I can confirm, that it did work with this display:
https://github.com/olikraus/u8g2/wiki/gallery#28-oct-2020-st7528-erc16064

But it looks like, that noone ever tested the NHD.

Can you use the U8G2_ST7528_ERC16064 constructor. Will it also flicker?
Please execute the contrast test: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/ContrastTest/ContrastTest.ino

Will you see something on the screen?

HavocRC, 2022-04-06:
I tried the U8G2_ST7528_ERC16064 constructor - still flickers, but it looks like this [(Flikr)](https://flic.kr/p/2ncReNY) - you can see the smaller LCD size inside the C160 size. No lines move.

Tried the contrast test with both constructors - still flickers, but some lines are missing [(Flikr)](https://flic.kr/p/2ncXSCQ). There's not a contrast fade; the messed up lines you see in the photo are sequentially filled then disappear and repeat. With the ERC15064 constructor it looks the same as the first photo uploaded, then that gap is sequentially filled and repeats.

olikraus, 2022-04-07:
Hmm, could it be a soldering/wiring/hardware issue?
How are caps connected to the flex cable? Which caps did you use? You mentioned an Uno Board: Did you add a level shifter? The display may not work with the 5V logic of the Uno.

olikraus, 2022-04-07:
Here is my own setup:

[image]

The IC in the right side (HC4050) does the signal level shift from Arduino UNO 5V logic to 3.3V. Ceramic caps (orange) are used on the very left close to the flex cable connector. Everything is soldered.

HavocRC, 2022-04-07:
So I'm not using a level shifter...because I assumed since I needed pullups to 3V3 that 3V3 would be my reference. I hooked it up to my oscope to verify [(Flikr)](https://flic.kr/p/2ncYn1j) and it looks like we are maxing out at 3.52V. The signal looks kinda crappy to me (probe comp is set properly). 

 I connected everything according to this schematic [(Flikr)](https://flic.kr/p/2nd5RWf), note CSB is now NC as I found it didn't make any difference whether it was high or low. 

I don't want to waste your time. I despise breadboards and I'm not confident about that signal I saw on the scope. I'm going to order a level shifter and get everything soldered together. I'll let you know what the results are in a few days.

olikraus, 2022-04-07:
Level shifting: Note, that I2C requires bidirectional level shifting. However if you use the u8g2 SW_I2C constructor unidirectional level shifters will also work.

CSB: I think you must connect CSB to ground according to data sheet.

You have seen the address with the scanner, so I assume the display will work even without level shift. I personally would assume there is a wiring / connection / soldering issue.

Locking at your pictures the display looks very strange. In fact it looks like that it is already broken. 

One more thing: It looks like you have uses electrolytic capacitors. The caps for the display are used to generate the LCD voltage (you can measure this at vout). So we need higher voltage with some amount of current. However only ceramic capacitors have the required low ESR (resistance) at the switching speed of the internal charge pump. I mean, the electrolytic capacitors may not be able to deliver the required current.

Vout: Note that the charge pump has to be enabled by software (u8g2 init sequence sent with u8g2.begin(), so only after sending that init sequence you will be able to measure >10V there.

olikraus, 2022-04-07:
Oh, one more thing: You should add a ceramic capacitor between VSS and VDD close to the flex cable connector. At least a 100nF and maybe additionally a 2.2uF should be there.

HavocRC, 2022-04-12:
After soldering everything together, adding some logic converters, it's all working fine! I was able to get the graphics test to work with page buffer, however it looks like everything is shifted off by 20 pixels or so. Too tired to figure it out tonight I'm just super happy that I got the screen showing content! Thank you for the support and sorry it was just a dumb issue on my end, I know better! Tomorrow I'll have to see about this pixel shift.

Just in case anyone else has this screen, here are all specs: (p/n from Digikey)

- LCD: NHD-C160100DIZ-FSW-FBW
- Connector: FH12-14S-0.5SH(55)
- Breakout board: FPC050P020
- TTL converter: BOB-12009
- Constructor: `U8G2_ST7528_NHD_C160100_1_HW_I2C u8g2(U8G2_R0,11);  // reset pin on 11`
- Page buffer
- Schematic below

[image]
[image]
[image]

olikraus, 2022-04-12:
Nice pictures, looks good.

For the initial line problem: Maybe try this:

u8g2.sendF("caca", 0x038, 0x064, 0x040, x);

where x is a number between 0 and 99 to fix the shift problem.
Is there any number x which ould fix the wrong shift?

olikraus, 2022-04-12:
Yes, I think it should be added to the init code inside u8g2. It is somehow missing. Maybe noone tested this display. At least I do not own this display, so I need to wait for feedback. So: Thanks for testing this device.

HavocRC, 2022-04-12:
Are there anymore tests you like me to run on the screen while I have it that would help you out?

olikraus, 2022-04-12:
Can you run the flip mode example? There should be a solid frame all the time at the border of the display.

HavocRC, 2022-04-13:
It works great! It takes the screen about 500ms to update all pixels from one side to the other.

olikraus, 2022-04-13:
Great, thanks for the feedback

olikraus, 2022-04-13:
I will close this topic once I did the repo update...

HavocRC, 2022-04-13:
> I will close this topic once I did the repo update...

Sounds good! Sorry I wasn't sure how you wanted it...on stack nobody ever accepts an answers as solved after it solves their problems :).

Thanks again for the help!

olikraus, 2022-08-13:
> It takes the screen about 500ms to update all pixels from one side to the other.

hmmm 500ms... did you use the hardware HW_I2C constructor? SW_I2C might be too slow.

olikraus, 2022-08-13:
closing this now. Hope the init change will work...

closed 2022-08-13 by olikraus

AJ-Smoothie, 2022-08-15:
> closing this now. Hope the init change will work...

I've been waiting for the update! I'll try it out and let you know if it works.

olikraus, 2022-08-15:
There is also a beta release available which includes this change:

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

blankenshipbotron, 2023-06-13:
Hello,
If anyone is trying to get this to work on a Feather M0 32 bit Adalogger module here is the code. This module has a CS (Chip select), but it doesn't seem to matter weather its driven low. Also make sure  setI2C is used before the begin

u8g2.setI2CAddress(0x7E);
 u8g2.begin();

``
#include <Arduino.h>
#include <U8g2lib.h>
#include "Wire.h" // KEYPAD

#define CS      A1 
#define RST     A0

#define SDA     20
#define SCL     21

U8G2_ST7528_NHD_C160100_F_HW_I2C u8g2(U8G2_R0,RST, SCL, SDA);

void setup() {
  // put your setup code here, to run once:
 //digitalWrite(CS, LOW);  
 u8g2.setI2CAddress(0x7E);
 u8g2.begin();
 

}

void loop() {
  // put your main code here, to run repeatedly:
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(0,15,"Thanks olikraus !");
    u8g2.drawStr(0,30,"USING u8g2 !");
    u8g2.drawStr(0,45,"NHD_C160_100 !");
  } while ( u8g2.nextPage() );
  delay(1000);

}
``