olikraus/u8g2 #1835: How do i use u8g2 library in raspberry pi

Eldho1416, 2022-04-11:
I have been using a KS0108 (JHDE128x64E) display and i have interfaced it with Arduino and it is working so, i need someone to help to interface it with raspberry pi like how to install this library and use it

thanks in advance

@olikraus

olikraus, 2022-04-11:
Some documentation is here:
https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#raspberry-pi

Note, that raspi is a 3.3v device, but your display expects 5v levels.

Eldho1416, 2022-04-11:
[image]

I'm sure i2c is not possible with this but i'm confused about SPI so can you help me here

olikraus, 2022-04-11:
Ks0108 doesn't support SPI.

Eldho1416, 2022-04-11:
so it means i can't interface it with raspberry pi?

olikraus, 2022-04-11:
> so it means i can't interface it with raspberry pi?

Like in most other domains, this is more a question of time and removal of obstacles. If you expect a ready made solution, then you are right, such a solution does not exist and we are also not able to provide such a solution. If you want to work on this you are welcome to contribute here.

Obstacles what I see here:
 - U8g2 is an Arduino Library, however, Raspberry Pi is not supported by Arduino, so it is also not supported by u8g2. The link above referes to unsupported contributed ports. You are on your own..
 - Existing ports assume I2C or SPI, they do not mention parallel interface. So none of the ports will work out of the box, you need to create a new port for parallel interface
 - As mentioned above, Raspberry Pi is a 3.3V device, but your display expects 5V. So you probably need voltage level shifters. Some additional hardware would be required. 

In other words: You ask the Maintainer of a Lib, which does not fit for your Hardware to solve the connection between a display and a hardware, which also do not fit together.

If you want to use your KS0108 display, then I suggest ot buy an Arduino Uno instead.

Eldho1416, 2022-04-11:
@olikraus thanks for the input. I have already used it in UNO board and works since i'm a noob in programming i raised this to get some input.

Also, in ug82 for 128x64 KS0108 displays you have included 3 chip in library as far as ik 128x64 displays have max of 2 chips

closed 2022-04-11 by Eldho1416