olikraus/u8g2 #1837: c-periphery as git submodule

shlyakpavel, 2022-04-12:
This repo includes code from https://github.com/vsergeev/c-periphery "as is", without any changes. This is helpful in small projects since there's no need to control external libraries.
However, some projects (such as the one I'm currently working on) already use c-periphery for other purposes. Bringing the same files with u8g2 means either compiling and linking two copies of u8g2 in the same project or maintaining a u8g2 fork that uses external c-periphery. Both ways seem to be impractical.
My proposal is to use external c-periphery (you can fork it to be sure you control the repo) and to get it downloaded and built as a git submodule. Not only can I benefit from that, but also other users will do as long as c-periphery gets updated and fixed in the future.
Thank you for your nice work!

olikraus, 2022-04-12:
> This repo includes code from https://github.com/vsergeev/c-periphery "as is", without any changes. 

Can you specify the exect location in u8g2?

shlyakpavel, 2022-04-12:
@olikraus , [/sys/arm-linux/drivers](https://github.com/olikraus/u8g2/tree/master/sys/arm-linux/drivers)

shlyakpavel, 2022-04-12:
> So if you need to update something, just get the files from that folder.

So would you mind if I created a pull request replacing these files with a git submodule?
It is convenient to use u8g2 repo "as is" without having to maintain a fork, so I try to upstream as much as possible from what we do here at dataprint.

olikraus, 2022-04-12:
> So would you mind if I created a pull request replacing these files with a git submodule?

No issue as long as this is limited to the "sys" folder.

closed 2022-04-13 by olikraus