olikraus/u8g2 #1839: Using U8G2 on Arduino Core STM32

technik-gegg, 2022-04-13:
Hey Oliver,

I'm currently in the process of porting my firmware from the Maple framework to the Arduino Core STM32 framework and I came across a major change in the Core STM32. All GPIO pin definitions have changed from **uint8_t** to **uint32_t**, which I guess has been made to reflect some of the newer MCUs from STM.
Until now, my firmware used the default uint8_t type for internal storage / parameters / return values and this caused a problem in exactly one specific function, in which the GPIO pin didn't do what was expected. It took me 2 days fiddling to figure what the exact reason was.
So, I needed to change all of my source code in order to reflect the new type. 
Instead of simply replacing *uint8_t* by *uint32_t* I did the following: I introduced a custom typedef "pin_t" which I applied to all of my internal GPIO pins / parameters / return values and added the following conditional instruction in some header file:

```cpp
#if !defined(STM32_CORE_VERSION)
  typedef uint8_t     pin_t;
#else
  typedef uint32_t    pin_t;
#endif
```
The *STM_CORE_VERSION* definition is being set by the framework automatically.
This enables me to switch easily between frameworks if I need to, but is also future proof (in case some other framework needs a different type, such as *uint16_t*, for example).

Just to be clear: So far, I haven't seen any malfunctions in the U8G2 library, but with the pace of new MCUs hitting the market, I imagine that this problem could arise in the near future. So, I just wanted to let you know.

I thought about opening an PR, but for one, the U8G2 has many, many, many pin definitions :o) and for the other, I figured that you create the U8G2 header files using a script. So, I guess, it's far easier to handle it this way, if you ever decide to move in this direction.

olikraus, 2022-04-13:
The core part of u8g2 is hardware independet. 
U8g2 for Arduino contains an hardware abstraction layer which the obviously has to use the datatypes of the target system.
So.... I do not see any need for changes in U8g2.

technik-gegg, 2022-04-13:
That's true. Though in U8G2 Arduino you've stated all issues/PR's have to go in here. So...

closed 2022-04-13 by technik-gegg