olikraus/u8g2 #1842: SSD1306 stops working after a while

zakzag, 2022-04-15:
Hi guys, 

I am working on a project, where I just want to show on a screen of SSD1306 connected to an ESP32 and U8G2. , using Arduino environment,(but not the Arduino software).
As I progress with my project, I realized that after including some libraries and started using them, the screen stopped working.
After some investigation in the disassembled code (it's an interesting journey by the way) and the stack trace, I find I2C communication suspicious as the reason why it stops working.
The libraries I am using: ArduinoJSON, AsyncTCP, ESPAsyncWebServer, Preferences, PubSubClient, SPI, SPIFFS, U8G2, Wifi, Wire.
I think the screen stops working when too long code in loop, but I am not really sure, so I wrote a small program that also makes screen work weird:

`#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ 19, /* data=*/ 23, /* reset=*/ U8X8_PIN_NONE);   // ESP32 Thing, pure SW emulated I2C

void setup(void) {
    Serial.begin(115200);
    u8g2.begin();
}

void loop(void) {
    u8g2.setFont(u8g2_font_6x10_tr);
    u8g2.setFontDirection(0);
    u8g2.setFontRefHeightAll();
    u8g2.firstPage();
    do {

        u8g2.drawBox(rand()%128, rand()%64, rand()%30, rand()%20);
        for (uint8_t i = 0; i < 50; i++) {
            u8g2.drawStr(rand() % 128, rand() % 64, "X");
        }
    } while (u8g2.nextPage());
    delay(30);
}`

When I change the delay from 30 to 0 everything is ok, when it's 100, the screen stops working
Using the value 30 results the way this video shows:

https://www.youtube.com/watch?v=Kx64WjW462Q

Do you have guys any idea what's happening? It does not matter which pin I use for I2c, standard SCL, SDA or anything else, the screen does the same.

The same happens when I started include and use all the libs mentioned above, because I need them. I can't paste the whole code here, because it's more than 4-5000 lines and has several classes, libs and files, but the video above shows exactly how it stops working.

olikraus, 2022-04-15:
Maybe the ESP watchdog triggers.

zakzag, 2022-04-15:
No, the device is not restarting

olikraus, 2022-04-15:
Maybe you need to call yield() in the u8g2 picture loop.

closed 2023-02-22 by olikraus