olikraus/u8g2 #1844: Request for ST7567S I2c 128/64 LCD support

he77PC011, 2022-04-15:
Request for ST7567S I2c 128/64 LCD support

https://www.buydisplay.com/download/ic/ST7567S.pdf

olikraus, 2022-04-15:
I have added I2C versions to the code:
for example:
U8G2_ST7567_JLX12864_1_SW_I2C
U8G2_ST7567_ENH_DG128064_1_SW_I2C
U8G2_ST7567_ENH_DG128064I_1_SW_I2C
U8G2_ST7567_OS12864_1_SW_I2C

olikraus, 2022-04-15:
I have created beta v2.33.1.
Can you test the above constructors?
You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

he77PC011, 2022-04-29:
failed the test
Using https://github.com/mworkfun/ST7567A_128X32DOT_LCD
Only 128*32 pixels LCD.
After checking, it seems that there is no
addr = 0x3f; //Address of LCD device.

olikraus, 2022-04-29:
I am confused. What has the referenced lib Todo with u8g2

RockKenny, 2022-08-05:
[image]

`#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_ST7567_ENH_DG128064_1_SW_I2C u8g2(U8G2_R0, SCL, SDA, U8X8_PIN_NONE);

void setup(void) {
  u8g2.setI2CAddress(0x3F * 2);
  u8g2.begin();
  u8g2.clearBuffer();          // clear the internal memory
}

void loop(void) {
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0, 10, "Hello World!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(1000);
}
`

Hello. How to fix?

olikraus, 2022-08-06:
Use U8G2_ST7567_ENH_DG128064_F_SW_I2C

Thomas6500, 2022-10-12:
HI
to use less memory , How i can configure  ST7567_ENH_DG128064I_F_SW_I2C  to use at the same time "u8g2.drawStr" and "u8x8.print"?
thanks

olikraus, 2022-10-12:
You can not use u8x8.print() because the C++ is notavailable in your case. However you could use the plain C u8x8 functions  instead of u8x8.print() (see https://github.com/olikraus/u8g2/wiki/u8x8reference#drawstring):
``` C++
void u8x8_DrawString(u8x8_t *u8x8, uint8_t x, uint8_t y, const char *s);
void u8x8_Draw1x2String(u8x8_t *u8x8, uint8_t x, uint8_t y, const char *s);
void u8x8_Draw2x2String(u8x8_t *u8x8, uint8_t x, uint8_t y, const char *s);
```

In order to do this, you need the u8x8_t pointer as first argument, which is returned by the following (undocumented) U8g2 function:
``` C++
u8x8_t *U8g2:getU8x8(void)
```

So a u8x8 call with a given u8g2 object will look like this:

``` C++
u8x8_DrawString(u8g2.getU8x8(), 0, 0, "Upper Left");
```

closed 2023-02-22 by olikraus