olikraus/u8g2 #1851: t6963_240x128_f u8x8_byte_8bit_8080mode: Confusion about U8X8_MSG_GPIO_E; Custom device driver?

carlk3, 2022-04-29:
I'm using a [STM32L496xx/STM32L4A6xx](https://www.st.com/resource/en/datasheet/stm32l4a6ag.pdf) MCU and a [NHD-240128WG-BTFH-VZ#](https://newhavendisplay.com/240x128-graphic-lcd-module-fstn-display-with-white-backlight/) LCD display, which has a [RA6963 ](https://support.newhavendisplay.com/hc/en-us/article_attachments/4414786705559/RA6963.pdf) controller (similar to T6963). 

So, the pins available to me are:
[image]

In setting up my "uC specific" GPIO and Delay callback, I'm not sure what to do with 
	`case U8X8_MSG_GPIO_E:				// E/WR pin: Output level in arg_int`
I don't have an "enable" pin. (I presume `U8X8_MSG_GPIO_CS: // CS (chip select) pin` maps to the **/CE** line in the table.) But I need to do something with **/WR** and **/RD**.

Here is my current attempt, which doesn't seem to work:
```
		if (arg_int) {
			LL_GPIO_ResetOutputPin(LCD_nWR_GPIO_Port, LCD_nWR_Pin);
			LL_GPIO_SetOutputPin(LCD_nRD_GPIO_Port, LCD_nRD_Pin);
			// Release data pins
			LL_GPIO_SetOutputPin(LCD_D0_GPIO_Port,
					LCD_D0_Pin
					| LCD_D1_Pin
					| LCD_D2_Pin
					| LCD_D3_Pin
					| LCD_D4_Pin
					| LCD_D5_Pin
					| LCD_D6_Pin
					| LCD_D7_Pin);
		} else {
			LL_GPIO_SetOutputPin(LCD_nWR_GPIO_Port, LCD_nWR_Pin);
			LL_GPIO_ResetOutputPin(LCD_nRD_GPIO_Port, LCD_nRD_Pin);
		}
```
(The `// Release data pins` part is because we're using open collector (pull down, no pull up) on the data GPIOs (5 V tolerant) to do 3.3 V to 5 V level conversion.)

As an alternative to figuring this out, I have an optimized low-level device driver for this display left over from an earlier project. It essentially provides two functions: Initialize, and Send Buffer. (It sends the entire 3840 byte [full framebuffer] in one swell foop. We only use graphics mode. We also take advantage of the large SRAM on the controller to do a sort of double buffering.) Is there a nice way that I could integrate that into u8g2?

olikraus, 2022-04-29:
E pin is the same as WR. RD pin needs to be connected to logic high

carlk3, 2022-04-29:
> E pin is the same as WR. RD pin needs to be connected to logic high

Oh wow, that fixed it! Does that mean that u8g2 never does any reads?

olikraus, 2022-04-30:
> > E pin is the same as WR. RD pin needs to be connected to logic high
> 
> Oh wow, that fixed it! Does that mean that u8g2 never does any reads?

Correct

carlk3, 2022-04-30:
Interesting. I'm surprised, since I read statements like "When using the MSB=0 command, a Status Read must be performed.
If a status check is not carried out, the RA6963 cannot operate normally, even after a delay time." I guess that only applies to Registers Setting commands? Anyway, I won't argue with success! It's good to know I don't need to worry about releasing the data lines (although I am now doing it in case U8X8_MSG_GPIO_CS when CS is set high, in order to save power).

I think u8g2 is going to be a great fit for this project. Thank you!

closed 2022-04-30 by carlk3

olikraus, 2022-04-30:
There are displays which doesn't allow reading at all. So I made this a habit for u8g2 in general: u8g2 will never read from the display. It works indeed nice for all the supported devices.

AndreAhmed, 2024-05-01:
@carlk3 
can you share your code please, I have the same issue

carlk3, 2024-05-01:
This is what I'm running now for an ST [NUCLEO-L4A6ZG](https://www.st.com/en/evaluation-tools/nucleo-l4a6zg.html) connected to a [Newhaven 240X128 Graphic LCD Module](https://newhavendisplay.com/240x128-graphic-lcd-module-fstn-display-with-white-backlight/):
```
static uint8_t u8x8_gpio_and_delay_stm32L4(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int,
                                           void *arg_ptr) {
    switch (msg) {
        case U8X8_MSG_GPIO_AND_DELAY_INIT:  // called once during init phase of u8g2/u8x8
            break;                          // can be used to setup pins
        case U8X8_MSG_DELAY_NANO:           // delay arg_int * 1 nano second
            // (arg_int ns) / (cycle time ns) NOPs
            // (cycle time ns) = (1 / SystemCoreClock) * 1E9
            //      = 1E9 / SystemCoreClock
            {
                size_t cyc = (size_t)arg_int / (1000000000 / SystemCoreClock + 1);
                // Function call itself takes 100
                if (cyc < 100)
                    return 1;
                else
                    cyc -= 100;
                for (size_t i = 0; i < cyc; ++i) {
                    __NOP();
                }
            }
            break;
        case U8X8_MSG_DELAY_100NANO:  // delay arg_int * 100 nano seconds
            delay_25ns(4 * arg_int);
            break;
        case U8X8_MSG_DELAY_10MICRO:  // delay arg_int * 10 micro seconds
            delay_25ns(10 * 40 * arg_int);
            break;
        case U8X8_MSG_DELAY_MILLI:  // delay arg_int * 1 milli second
            vTaskDelay(pdMS_TO_TICKS(arg_int));
            break;
    //    case U8X8_MSG_DELAY_I2C:				// arg_int is the I2C speed in 100KHz,
    //    e.g. 4 = 400 KHz
    //      break;							// arg_int=1: delay by 5us, arg_int = 4:
    //      delay by 1.25us
        case U8X8_MSG_GPIO_D0:  // D0 or SPI clock pin: Output level in arg_int
            if (arg_int)
                LL_GPIO_SetOutputPin(LCD_D0_GPIO_Port, LCD_D0_Pin);
            else
                LL_GPIO_ResetOutputPin(LCD_D0_GPIO_Port, LCD_D0_Pin);
            break;
        case U8X8_MSG_GPIO_D1:  // D1 or SPI data pin: Output level in arg_int
                                //  case U8X8_MSG_GPIO_SPI_DATA:
            if (arg_int)
                LL_GPIO_SetOutputPin(LCD_D1_GPIO_Port, LCD_D1_Pin);
            else
                LL_GPIO_ResetOutputPin(LCD_D1_GPIO_Port, LCD_D1_Pin);
            break;
        case U8X8_MSG_GPIO_D2:  // D2 pin: Output level in arg_int
            if (arg_int)
                LL_GPIO_SetOutputPin(LCD_D2_GPIO_Port, LCD_D2_Pin);
            else
                LL_GPIO_ResetOutputPin(LCD_D2_GPIO_Port, LCD_D2_Pin);
            break;
        case U8X8_MSG_GPIO_D3:  // D3 pin: Output level in arg_int
            if (arg_int)
                LL_GPIO_SetOutputPin(LCD_D3_GPIO_Port, LCD_D3_Pin);
            else
                LL_GPIO_ResetOutputPin(LCD_D3_GPIO_Port, LCD_D3_Pin);
            break;
        case U8X8_MSG_GPIO_D4:  // D4 pin: Output level in arg_int
            if (arg_int)
                LL_GPIO_SetOutputPin(LCD_D4_GPIO_Port, LCD_D4_Pin);
            else
                LL_GPIO_ResetOutputPin(LCD_D4_GPIO_Port, LCD_D4_Pin);
            break;
        case U8X8_MSG_GPIO_D5:  // D5 pin: Output level in arg_int
            if (arg_int)
                LL_GPIO_SetOutputPin(LCD_D5_GPIO_Port, LCD_D5_Pin);
[... 51 lines cut ...]
    //    case U8X8_MSG_GPIO_I2C_CLOCK:		// arg_int=0: Output low at I2C clock pin
    //      break;							// arg_int=1: Input dir with pullup high for
    //      I2C clock pin
    //    case U8X8_MSG_GPIO_I2C_DATA:			// arg_int=0: Output low at I2C data pin
    //      break;							// arg_int=1: Input dir with pullup high for
    //      I2C data pin
    //    case U8X8_MSG_GPIO_MENU_SELECT:
    //      u8x8_SetGPIOResult(u8x8, /* get menu select pin state */ 0);
    //      break;
    //    case U8X8_MSG_GPIO_MENU_NEXT:
    //      u8x8_SetGPIOResult(u8x8, /* get menu next pin state */ 0);
    //      break;
    //    case U8X8_MSG_GPIO_MENU_PREV:
    //      u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ 0);
    //      break;
    //    case U8X8_MSG_GPIO_MENU_HOME:
    //      u8x8_SetGPIOResult(u8x8, /* get menu home pin state */ 0);
    //      break;
        default:
            u8x8_SetGPIOResult(u8x8, 1);  // default return value
            break;
    }
    return 1;
}
```