olikraus/u8g2 #1857: Query For other language

I-AM-Tushar-chauhan, 2022-05-04:
Hello sir..
I need some other languages like Hindi,Punjabi,Gujrati etc, but I don't know how the encoding or decoding are done in it.
can you explain me how is it working in u8g2 with english language??

olikraus, 2022-05-04:
U8g2 support UTF-8, which is the default encoding of gcc and Arduino IDE. In other words you can just enter Hindi or any other glyph as they are and it will be displayed with u8g2.

First step for any language support is to figure out the unicode range of your desired language: For Hindi you need Devanagari glyphs as described here: https://en.wikipedia.org/wiki/Devanagari_(Unicode_block)
The range would be U+0900..U+0954. The next step is to select a suitable font which contains those glyphs.
Unifont is such a font. Once you have there two things, then we can construct the font for u8g2.
In your case Hindi font is already included in u8g2: 

[image]

I-AM-Tushar-chauhan, 2022-06-10:
Hello sir,
Thanks for the reply, I can't find out how this (u8g2_font_unifont_t_devanagari) is working because I was accessing the unifont.bdf file and I was making changes into it but I can't find out this type of file (u8g2_font_unifont_t_devanagari)
for unifonts. can you please guide me on how to make it or how to make changes to it??

/*
  Fontname: -gnu-Unifont-Medium-R-Normal-Sans-16-160-75-75-c-80-iso10646-1
  Copyright: Copyright (C) 2014 Roman Czyborra, Paul Hardy, Qianqian Fang, Andrew Miller, et al. Licensed under the GNU General Public License; either version 2 or (at your option) a later version, with the GNU Font Embedding Exception.
  Glyphs: 305/57086
  BBX Build Mode: 0
*/
const uint8_t u8g2_font_unifont_t_devanagari[6002] U8G2_FONT_SECTION("u8g2_font_unifont_t_devanagari") = 
  "\61\0\4\2\5\5\5\5\6\20\20\0\376\12\376\13\377\1\270\3y\5k \6\0@\217\2!\10A"
  "Q\210\302AB\42\11\205H\214B\306M\0#\21FE\210JO\6\203D\244\305`\20\351\11\0$"
  "\24GE\210Nh\240\210D$\241\331(\42\251\30\210\62\0%\27GE\210\206F\222(ID\24\251"
  "X*!ID\22%\32\5\0&\24GE\210\312\246$S\21\22%\42\232\204(\221\221L\2'\7"
  "\201P\214\302 (\15\203\315\207J\42\222\210\364&\222\11)\16\203\311\207B&\222\211\364\42\222\210\0"
  "*\16\347\304\210NIE\233M\242IM\6+\15\347\304\210N\254f\60\310\304j\0,\11\202L\207"
  "\202\42\241\0-\7$H\212\302 .\7BL\210\302 /\15FE\210VQU(\25\252J\1\60"
  "\17FE\210\212&\222\10\371(\21\311H\0\61\14EI\210JF\222\10\365\311@\62\20FE\210\306"
  " \21\22\245\62\232PUj\60\63\22FE\210\306 \21\22\245\42\263\224H\224\30$\0\64\22FE"
  "\210RH\223\210Td\22\231\304`\224\252\0\65\20FE\210\302A\252j K\245D\211A\2\66\20"
  "FE\210\312$\224J\15\24!G\211A\2\67\14FE\210\302UQ\252(\325\4\70\21FE\210\306"
  " \21\62J\14\22!G\211A\2\71\20FE\210\306 \21\62J\14T\215\42\23\0:\11\342\314\210"
  "\302 \66\10;\13\42\315\207\302 \246H(\0<\11%I\210R\246\253\16=\11\246D\211\302=f"
  "\60>\12%E\210B\252\233\216\0\77\17FE\210\306 \21\22\245BU\331T\4@\25FE\210\312"
  "$#I(\22\213\210\42\242\210H\26\261A\0A\17FE\210\212&\322\42$\32\14D\216\2B\20"
  "FE\210\302@\21\62\32,B\216\6\13\0C\17FE\210\306 \21\262\352H\224\30$\0D\17F"
  "E\210\302 \222I\204\374&\61\210\0E\16FE\210\302A\252j\240H\265\32\14F\15FE\210\302"
  "A\252j\240Hu\5G\20FE\210\306 \21\262*\31\204l\24\213\0H\14FE\210B\310\321`"
  " \362(I\14EI\210\302@\22\352'\3\1J\17GE\210\312@\24\353I&\222\311l\0K\23"
  "FE\210BH\223\210TdD\242D&R\223\10\5L\12FE\210B\252\277\32\14M\16FE\210"
  "BH\64\31D\26\36\71\12N\22FE\210Bh\263QD\24\21IB\222\320\34\5O\15FE\210"
  "\306 \21\362G\211A\2P\16FE\210\302@\21\62\32,R]\1Q\31g\305\207\306 \22J\204"
  "\22\241D(\21J\204\22\11EB\42\31\344\4R\22FE\210\302@\21\62\32,\42\65\211L\42$"
  "\12S\17FE\210\306 \21\62s$J\14\22\0T\13GE\210\302A&\326\337\0U\14FE\210"
  "B\310\77J\14\22\0V\23GE\210B\312*\221\211d\42\231\212P\42\25\313\0W\16FE\210B"
  "\310\13\27\223Ad$\12X\21FE\210BH\224\210\264\21i\42-B\242\0Y\20GE\210BJ"
  "\225\310D\62\25\251X\67\0Z\14FE\210\302UQ\257R\203\1[\12\203\321\207\302 \322\77\31\134"
  "\15FE\210B*V\25K\305\252\2]\12\203\305\207\302\244\177\62\10^\12f\304\214\212&\222\10\5"
  "_\10'\304\207\302A\0`\10cH\215B\246\0a\17\6E\210\306 \21J%\6#\33\305\42b"
  "\17fE\210B\252\305B#r\264Q,\0c\16\6E\210\306 \21Ru\224\30$\0d\15fE"
  "\210V\213\205f\344\215b\21e\20\6E\210\306 \21\22\15\6\252\242\304 \1f\15eE\210\216$"
  "T\62\220\204z\2g\25fE\207Vb\241Id\22\231\310&\65H\204D\211A\2h\15fE\210"
  "B\252\305B#\362Q\0i\14eI\210J((\352\223\201\0j\15\245E\207R\241\250\217$\21\11"
  "\0k\22fE\210B\252M\42R\221\21%\62\221\232D(l\12eI\210\206\250\177\62\20m\24\7"
  "E\210\302B\21\211H\42\222\210$\42\211H\42\222\2n\13\6E\210Bb\241\21\371(o\15\6E"
  "\210\306 \21\362Qb\220\0p\20FE\207Bb\241\21\71\332(\26\251\24\0q\15FE\207\306B"
  "\63\362F\261H\25r\14\6E\210Bb\241\21\251\272\2s\16\6E\210\306 \21\222\71J\14\22\0"
  "t\14EE\210J\250d \11u%u\13\6E\210B\310o\24\213\0v\15\6E\210B\310(\21"
  "\351\215H\2w\23\7E\210BJ\22\221D$\21ID\22\221TX\0x\20\6E\210BH\224\210"
  "dD\232H\42$\12y\17FE\207B\310\243D\304\42U\61H\0z\13\6E\210\302U\250W\203"
  "\1{\17\203\315\207\306$\23ID\62\221D$#|\10\301Q\207\302\7\2}\20\203\311\207\202&\222"
  "\210d\42\211H&\262\0~\13gD\214\206F\22\321(\0\177&\20B\7C\242\337#\364\4z\304"
  "d\220\310L\22\241T$\61Hd&\211Pj\62H\14\366\4z\2=\242\77\200'\20B\7C\242"
  "\337#\364\4z\304F\262\230$\42\211H\344 \21\31\204\42\211HQ$\261\320\23\350\11\364\210\376\0"
  "\0\0\0\10\11d\11\206\377\377\11\0\16\210\320\14\317(\243H(&\32\0\11\1\16\210\320\14CD"
  "\224\60\311h&\0\11\2\10B\334\15\303 \11\3\12\242\360\11\303 \62\10\11\4$\356\311\7K|"
  " \317M\6\203D&\242\214\10%\262\215\60#\12\15D!\205*\262\220\15\42bu\10\11\5 \217"
  "\305\7\313\301@\221\211H#J\211n\243\314\250B\3UH!\213,t\203\210Z\236\2\11\6'\217"
  "\305\7\313\301@\221\211\230E\254$\66\33\211*#Q\204\6\22EH!\221D\26\22\315 b)\221"
  "f\22\0\11\7\34\260A\7\303\17\202r\351`)\327\3\16\362\0\251D\231\260\34\314\363\210\24\0\11"
  "\10!\20B\7\333 \253\11\7\7\37\4\345\322\301R\256\7\34\344\1R\211\62a\71\230\347\21)\0"
  "\11\11\32\220\301\7\303\17rz\200\134=\35\10S\302\230\60\245\14I#\333A\16\11\12\36\220\301\7"
  "\303\17rz\200\134=\35H\66)\205\244lS%*\322\250\42\233\334 \7\11\13$\260A\7\303\17"
  "r\312\211P\222\220('\302\301@\66\330\244&\232TH\262TL\25\351D\36\61H\0\11\14!\220"
  "\301\7\303\17\242y@n\25[D\6\42\315 \242\61\62T\245\6\71\211\70\221G\14$\0\11\15!"
  "\20B\7cD\274G\15>\10\31\212\204\42\241H\250\221\346\21y\204\36\241G\344\1r\15\0\11\16"
[... cut ...]
  "\17\2\250\374,\316\305\7\303\17B\223\201\302\305 \64\31(\134\14\22\212\311\340\377h\63H\14\24\203"
  "Af\61\30$\6\212\301 \61\330\14>\10\250\375,\316\305\7\303\17B\223\201\302\305 \64\31(\134"
  "\14\22\212\311\340\377H\63P\14\24\212Af\241\30$\6\12\305 \61\320\14>\20\250\376+\316\305\7"
  "\303\17B\223\201\302\305 \64\31(\134\14\22\212\311\340\377H\64H\14\24\203A\346@\61P\14\6\211"
  "\201h\360A\0\250\377+\316\305\7\303\17B\223\201\302\305 \64\31(\134\14\22\212\311\340\377H\64H"
  "\14\24\203A\346@\61P\14\6\211\201b\360\203\0\0";

olikraus, 2022-06-12:
>  I can't find out how this (u8g2_font_unifont_t_devanagari) is working 

use `u8g2.setFont(u8g2_font_unifont_t_devanagari)`

I-AM-Tushar-chauhan, 2022-06-14:
file-path
u8g2->csrc->u8g2_fonts.c 

> use `u8g2.setFont(u8g2_font_unifont_t_devanagari)`
this Devanagari fonts are coming from the file path which I mentioned above. basically, I need to know how that file is generated so I will change the text's height and width and shape.

olikraus, 2022-06-14:
> I need to know how that file is generated ...

https://github.com/olikraus/u8g2/blob/2005154785e5c9431770348b93a399c928622052/doc/faq.txt#L252-L272

I-AM-Tushar-chauhan, 2022-06-16:
> 

Thank you, sir,
you gave me your precious time.

closed 2022-06-16 by I-AM-Tushar-chauhan