olikraus/u8g2 #1860: Core Dump LoadProhibited Exception on firstPage()

Ricky6633, 2022-05-09:
Hello i have a issue where on the firstPage line the ESP32 throws an Exception occasionally, not always but like every 30 loop runs

`Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e0d7f  PS      : 0x00060730  A0      : 0x800d39bb  A1      : 0x3ffb1e80  
A2      : 0x00000038  A3      : 0x3f405dcb  A4      : 0x3ffc2b38  A5      : 0x3f400de4  
A6      : 0x00000000  A7      : 0xfcb2ee80  A8      : 0x800e4cb4  A9      : 0x3ffb1e60  
A10     : 0x00000002  A11     : 0x3f400d50  A12     : 0x00000000  A13     : 0x40f51800  
A14     : 0x7ff00000  A15     : 0xbbfef134  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000a8  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

ELF file SHA256: 0000000000000000

Backtrace: 0x400e0d7f:0x3ffb1e80 0x400d39b8:0x3ffb1ea0 0x400d45f0:0x3ffb1f20 0x400d295e:0x3ffb1f90 0x400e6239:0x3ffb1fb0 0x4008a246:0x3ffb1fd0`

My Code looks as follow:

`void UC1701X_BME680::classSetup()
{
  u8g2->begin();
  u8g2->setContrast(180);
  u8g2->setFontPosCenter();
  u8g2->enableUTF8Print();
}

void UC1701X_BME680::setDisplay(float Temperature, float Humidity, float Pressure, float IAQ, uint8_t IAQStatus, float CO2, float VOC, 
                                float TemperatureMin, float TemperatureMax, float HumidityMin, float HumidityMax)
{
  Serial.println("Print Display");
  u8g2->firstPage();
  Serial.println("firstPage");
  do {
    char buffer[20];
    //u8g2->drawStr(0, 45, "Hello World!");
    //u8g2->drawLine(0, 9, 128, 9);
    //u8g2->drawLine(63, 12, 63, 45);
    //u8g2->drawLine(0, 45, 128, 45);

    u8g2->setFont(u8g2_font_siji_t_6x10);
    u8g2->setCursor(58, 5);
    u8g2->print(""); // WLAN3
    //u8g2->drawGlyph(5, 20, 0xE21A);
    //u8g2->drawUTF8(0, 5, 0xE21A);

    u8g2->setFont(u8g2_font_8x13B_tf);
    u8g2->setCursor(0, 25);
    u8g2->print(Temperature);
    u8g2->print("°C");

    u8g2->setCursor(-4, 3);
    u8g2->setFont(u8g2_font_siji_t_6x10);
    u8g2->print(""); // Arrow up
    u8g2->setFont(u8g2_font_6x12_mf);
    u8g2->print(TemperatureMax);
    u8g2->setCursor(-4, 13);
    u8g2->setFont(u8g2_font_siji_t_6x10);
    u8g2->print(""); // Arrow down
    u8g2->setFont(u8g2_font_6x12_mf);
    u8g2->print(TemperatureMin);

    u8g2->setFont(u8g2_font_8x13B_tf);
    sprintf(buffer, "%s%%", String(Humidity, 2));
    u8g2->setCursor(u8g2->getDisplayWidth() - u8g2->getUTF8Width(buffer), 25);
    u8g2->print(Humidity);
    u8g2->print("%");

    u8g2->setFont(u8g2_font_6x12_mf);
    sprintf(buffer, "%s", String(HumidityMax, 2));
    u8g2->setCursor(u8g2->getDisplayWidth() - (u8g2->getUTF8Width(buffer) + 10), 3);
    u8g2->print(HumidityMax);
    u8g2->setFont(u8g2_font_siji_t_6x10);
    u8g2->print(""); // Arrow up
    u8g2->setFont(u8g2_font_6x12_mf);
    sprintf(buffer, "%s", String(HumidityMin, 2));
    u8g2->setCursor(u8g2->getDisplayWidth() - (u8g2->getUTF8Width(buffer) + 10), 13);
    u8g2->print(HumidityMin);
    u8g2->setFont(u8g2_font_siji_t_6x10);
    u8g2->print(""); // Arrow down
    
    u8g2->setFont(u8g2_font_6x12_tf);
    float Pressure_hPa = Pressure / 100;
    sprintf(buffer, "%s hPa", String(Pressure_hPa, 2));
    u8g2->setCursor((u8g2->getDisplayWidth() - u8g2->getUTF8Width(buffer)) / 2, 37);
    u8g2->print(Pressure_hPa, 2);
    u8g2->print(" hPa");

    u8g2->setFont(u8g2_font_8x13B_tf);
    sprintf(buffer, "%s", String(IAQ, 0));
    u8g2->setCursor((u8g2->getDisplayWidth() - u8g2->getUTF8Width(buffer)) / 2 + 24, 54);
    u8g2->print(IAQ, 0);

    u8g2->setFont(u8g2_font_6x12_tf);
    u8g2->setCursor(0, 50);
    u8g2->print(VOC, 0);
    u8g2->print(" eVOC");
    u8g2->setCursor(0, 60);
    u8g2->print(CO2, 0);
    u8g2->print(" eCO2");

    u8g2->setFont(u8g2_font_emoticons21_tr);
    u8g2->setCursor(106, 54);
    if (IAQStatus == 0) {
      u8g2->print("\x0034");
    } else if (IAQStatus == 1) {
      u8g2->print("\x0034");
    } else if (IAQStatus == 2) {
      u8g2->print("\x0034");
    } else {
      if (IAQ < 100) {
        u8g2->print("\x0022");
      } else if (IAQ < 200) {
        u8g2->print("\x0024");
      } else if (IAQ < 300) {
        u8g2->print("\x0026");
      } else {
        u8g2->print("\x0029");
      }
    }
  } while ( u8g2->nextPage() );
}`

Im not sure if the error really is in the firstPage function, but it happens right between the 2 Serial calls, "Serial.println("firstPage") never gets called

olikraus, 2022-05-09:
How is *u8g2 constructed?

Ricky6633, 2022-05-09:
UG82_UC1701_MINI12864_1_4W_HW_SPI *u8g2;
u8g2 = new UG82_UC1701_MINI12864_1_4W_HW_SPI(U8G2_R0, CS_PIN, RS_PIN, RST_PIN);

Tho, i changed a bit on my code and the exception gets throws now (consistently) at another point in my code (another class that has nothing todo with the display), unfortunaly im quite a noob specially if it comes down to debugging, could this be a general heap fragmentation issue?

olikraus, 2022-05-09:
I personally would do something like this:
Maybe you could try this:
``` C++
UG82_UC1701_MINI12864_1_4W_HW_SPI u8g2_obj(U8G2_R0, CS_PIN, RS_PIN, RST_PIN);
UG82_UC1701_MINI12864_1_4W_HW_SPI *u8g2 = &u8g2_obj;
```
"new" should work, but I never used true.

> if it comes down to debugging

Unfortunately noone else could do the debugging for your. 

> could this be a general heap fragmentation issue

U8g2 does not use any new/delete/malloc/free. So u8g2 can not change memory heap in any way.

closed 2023-02-22 by olikraus