olikraus/u8g2 #1861: Grayscale implementation

ReinBentdal, 2022-05-11:
There has previously been an issue about grayscale implementation, #421.

What is the current status on a grayscale implementation? Has there been any work on an extra lib, as discussed in the referenced issue?

olikraus, 2022-05-11:
I never started work on such a Lib ☹️

ReinBentdal, 2022-05-11:
I see. I haven't yet looked at the implementations in your library. How difficult do you think it would be to implement this for one specific driver? For example if I where to override `setContrast`.

olikraus, 2022-05-11:
You mean to change U8g2 to support graylevel displays? 
I would say: Almost impossible. U8g2 is optimized for two colors. That's exactly why I wanted to create a new lib for grayscale displays.

ReinBentdal, 2022-05-11:
Not exactly. What I meant was just to mutate the library myself and only implement grayscale for my specific needs and driver, as a workaround.
That is: modify the library so I can set grayscale for the ssd1309 driver (maybe in a hacky way) with no intentions for it to be a general library feature

ReinBentdal, 2022-05-11:
oh okay, that's on my. Thought it had support for it..

closed 2022-05-11 by ReinBentdal