olikraus/u8g2 #1862: linux i2c example code warning

JamesGKent, 2022-05-11:
in the linux-i2c example in the sys directory there is a line that triggers a warning under GCC [here](https://github.com/olikraus/u8g2/blob/29c6b797072dda4c1a3e8e28aeab96ea4a5b525d/sys/linux-i2c/common/linux-i2c.c#L40)

it currently looks like this:
`data[idx] = *(uint8_t *)(arg_ptr+i);`
I think it should look like:
`data[idx] = *((uint8_t *)(arg_ptr)+i);`

in the current implementation the pointer arithmetic is happening on the void pointer instead of the uint8_t pointer.
I think this means if used with a display that sends multiple bytes in one transaction that this implementation would fail.

olikraus, 2022-05-11:
Well, it is contributed code and never got reviewed.
Can you specify the exact file and line number?

JamesGKent, 2022-05-11:
no worries, link is in fist comment and [here](https://github.com/olikraus/u8g2/blob/master/sys/linux-i2c/common/linux-i2c.c#L40)

but its sys/linux-i2c/common/linux-i2c.c Line 40

olikraus, 2022-05-11:
Ah, thanks. I missed the link.

https://github.com/olikraus/u8g2/blob/29c6b797072dda4c1a3e8e28aeab96ea4a5b525d/sys/linux-i2c/common/linux-i2c.c#L40

olikraus, 2022-09-03:
changed as suggested...

closed 2022-09-03 by olikraus