olikraus/u8g2 #1864: How to prepare commit to release?

pikot, 2022-05-15:
Hi! I try to adapt my code which I added to u8g2_arduino library https://github.com/olikraus/U8g2_Arduino/pull/10 to main u8g2 library.
Is any good practices to prepare code to release exist?   Correct me if i'm wrong about workflow:
1)  I add  my driver to tools/codebuild/codebuild.c, 
 2)  I prepare release for arduino by this script tools/release/arduino/create_release.sh
3) I test it on my device

maybe some uint tests exist, which I haven't find? )

olikraus, 2022-05-16:
Pull requests should go to u8g2.
I can do the code build topics for you.

pikot, 2022-05-17:
I created pull request -> https://github.com/olikraus/u8g2/pull/1866. It tested with generated Arduino library in my project)

olikraus, 2022-05-18:
I like the work on codebuild.c
But I probably need some time to investigate this. Unfortunately I have not much time for this project at the moment..

olikraus, 2022-08-13:
Due to increased code size, I decided not to include the PR. Please excuse.

closed 2022-08-13 by olikraus