olikraus/u8g2 #1867: Differential sendBuffer()

vanniaz, 2022-05-21:
In some situations (e.g. ESP32 connected to 128x64 OLED via I2C) there is plenty of RAM and computing power, so full screen buffer mode is the fastest and most versatile option, but the I2C connection to the display becomes a bottleneck.
While the screen buffer content can be updated quickly, the I2C transfer of the buffer can take tens of milliseconds. Using SPI is a possible solution but requires more I/O pins that are not always available.

I think that a simple enhancement could exist: keeping track of the first and last modified bytes in the screen buffer while the buffer is being updated, and sending only the bytes included in this range via I2C.
Here "first" and "last" mean in display memory order, so that the modified range can be sent with a single write command, with the proper intial screen location.

Of course this is not a perfect optimization, because if for example the first and last bytes of the buffer have been modified the full buffer would be sent anyway instead of just two bytes. But it has the advantage of being simple and not requiring several write commands to different locations, just a start location and a length.
In most cases this should allow a significant reduction in the screen update time.

I am proposing this as a feature request. I did not look at the source code so maybe that I am overlooking some possible issues, but if it is not too complicated it would be a nice improvement.

olikraus, 2022-05-21:
Thanks for the feature request.

Initial question: Did you use the hardware I2C constructor ("HW" inside the constructor name)?

Here are my arguments against this :-/

### No speed improvement for slow controllers.

Let me assume the following situation for the ESP32 with 240 MHz:
 * I2C transfer time 50ms
 * Rendering time 10ms
In total this would require 60ms.
Now let us assume, that your approach can reduce the transfer time to by 20%. Additionally we may need 5% extra code to check and update the min and max modified position  (first and last modified byte if I understood your approach)
So we have:
 * I2C transfer time 40ms
 * Rendering time 10.5ms

Obviously this is faster by 9.5 ms
Now let us assume a 8 MHz ATMega system, which is 30 times slower. The same task would take:
 * I2C transfer time 50ms
 * Rendering time 300ms
With the same speed up asumption, we get:
 * I2C transfer time 40ms (same as above)
 * Rendering time 315ms

Overall this is slower by 5ms. So: On the ATMega the new code will have a negative effect on performance. Additionally it will consume more flash memory. The penalty will be even mode for page mode updates.

### Embedded vs PC development

A windows developer is used to have so much computation power. We let the computer do on the fly / live computation also in cases where it wouldn't be necessary. A good example are true-type fonts: It is convenient to provide a font size and let the PC do the scaling of the font. Of course it would be much faster to calculate the bitmap image of each glyph instead of doing an on the fly scaling. But we got used to this, because there is so much cpu power available. This is different with on embedded systems with limited cpu power. 

The developer has the choice:
 * Let the controller figure out the highest and lowest row which was changed in order to transfer only the area between these two rows.
 * The developer knows which area will be updated. Why wasting cpu power to calculate this area? Just transfer only this reduced area: Use https://github.com/olikraus/u8g2/wiki/u8g2reference#updatedisplayarea

### Github fork

U8g2 does not focus on speed for a specific system. It tries to support many displays for a wide range of systems.
If a speed up is required for a specific system, then a fork of the project is a great idea to improve speed for a specific system like ESP32. 

### Conclusion

So... I hope it is ok for you that I will not work on this request.

vanniaz, 2022-05-22:
Thank you for the very detailed reply.

I did a quick test by writing a string "ABCD" with a 33 pixel high font in the middle vertical area of the 128x64 screen.
This is close to a real situation where you have some small texts in the top and bottom areas of the screen, and a bigger value in central position which gets updated frequently.

These are the times measured with millis() on the ESP32 (using the hardware I2C constructor):

- rendering time 1 ms
- I2C transfer time 36 ms

Since the font is 33 pixel high, only 33 horizontal screen lines could be transferred instead of the full 64. The amount of data would be 33/64 of the buffer size (but the time would be somewhat higher due to the I2C messages overhead).
A conservative estimate could be a 35% total update time reduction.

I do not know whether the transfer process can be trimmed to single horizontal lines or it needs to work in multiple of 8 (like updateDisplayArea), but even rounding to multiples of 8 there will be a significant saving.

If instead of the large text we only needed to update a smaller line of text, e.g. a 12 pixel high status bar at the bottom of the screen, **the time reduction could reach 75%**.

If you estimate that an ATMega is 30 times slower, and you multiply the rendering time in the above examples by 30 you still have some reduction in the first case (33 pixel height) and a good reduction in the second case (12 pixel height).
I do not have an ATMega board ready for tests at the moment, so if somebody would like to do these test it could be useful. The updateDisplayArea function can be used to simulate a differential sendBuffer.

Moreover, the "differential sendBuffer" feature could be switched on/off with a define, so the user could decide whether to use it or not depending on the situation.

Of course I do not expect you to work on it, but at least I think it is something worth considering for future enhancements.

I could think about the idea of creating a fork but I do not know whether I have the required programming skills and enough spare time. Anyway I will take a look at the code, to check how difficult this would be.

vanniaz, 2022-05-22:
I simulated the differential sendBuffer() by computing the bounding rectangle of a string and then finding the "tiles" to pass to updateDisplayArea().
To test it I wrote a 4 character string "ABCD" with a 33 pixel high font first, and then with a 12 pixel high font. The rendering and transfer times are taken for sendBuffer and for the computed updateDisplayArea.

The results with an ESP32 are the following:
|  Font Height  | Transfer Type | Render Time (ms) |Transfer Time (ms)|
| ------------- | ------------- | ------------- |------------- |
|  33 | sendBuffer  | 1  | 36  |
|  33 | updateDisplayArea  | 1  | 25  |
|  12 | sendBuffer  | 1  | 36  |
|  12 | updateDisplayArea  | 1  | 4 |

Here is the small test program:

```
uint8_t x=0, y=50, w;
int8_t ascent, descent;
uint8_t delta;
char textMessage[] = "ABCD";
	
uint32_t stopWatch = millis();
u8g2.clearBuffer();

// test with 33 pixel high font
//u8g2.setFont(u8g2_font_inb33_mr);

// test with 12 pixel high font
u8g2.setFont(u8g2_font_freedoomr10_mu);

w = u8g2.getStrWidth(textMessage);
ascent = u8g2.getAscent();
descent = u8g2.getDescent();
u8g2.drawStr(x,y, textMessage);

delta = millis() - stopWatch;
Serial.println("Rendering time:");
Serial.println(delta);

stopWatch = millis();

// transfer full buffer with sendBuffer
//u8g2.sendBuffer();

// transfer only modified tiles with updateDisplayArea
u8g2.updateDisplayArea(x/8, (y-ascent)/8, w/8 + 1, (ascent-descent)/8 + 1);

delta = millis() - stopWatch;
Serial.println("Buffer transfer time:");
Serial.println(delta);
```
This approach could be used as a replacement for a true built-in differential sendBuffer, but it is probably much less efficient.
However, if somebody has an Arduino Uno or another ATMega board and an I2C display on hand, it could be interesting to repeat the same tests with the ATMega to see how this approach works for slower processors.

_**Note:**_ I did these tests very quickly and I may have overlooked something or made mistakes, feel free to let me know!

olikraus, 2022-05-23:
Thanks for the measures. So it looks like, that you can already improve speed by using updateDisplayArea().
Whouldn't this be good enough then?

Small topics with your test:
 - ATMega probably needs to use the page mode render method
 - My argument had not only been, that the ATMega is slower, but also that the speedup from the communicaton might be eaten up by the overhead (5%) of calculating first and last byte.

|  Font Height  | String Length  | Transfer Type | Render Time (ms) | Transfer Time (ms) | Total Time Reduction |
| ------------- | ------------- | ------------- |------------- |------------- |------------- |
|  33 |  4 | sendBuffer  | 40  | 36  | |
|  33 |  4 | updateDisplayArea  | 40  | 25  | 14%|
|  33 |  4 | sendBuffer with auto area calculation  | 42  | 25  |  |

olikraus, 2022-05-23:
> Yes, but updateDisplayArea() is a bit awkward to use, so I will write a function "drawStrDirect" that computes the bounding rectangle based on getStrWidth, getAscent, getDescent, converts the rectangle to tile positions and finally sends the modified tiles via I2C by calling updateDisplayArea.

Sounds good, this is exactly the purpose of updateDisplayArea.

> that are complicated to handle with updateDisplayArea(). Do you think that it would be possible?

This is indeed complicated. Do-able, but difficult, because a different functions is required for each rotation / mirror combination.

> Since you work with "tiles", there could be a "dirty" bit associated to each tile

Well, yes...but still complicated. The related procedures are highly optimized already and the calculation of the dirty flag itself would require a good amount of extra time:
One procedure is this:
https://github.com/olikraus/u8g2/blob/29c6b797072dda4c1a3e8e28aeab96ea4a5b525d/csrc/u8g2_ll_hvline.c#L72
So the initial dirty tile flag has to be calculated here:
https://github.com/olikraus/u8g2/blob/29c6b797072dda4c1a3e8e28aeab96ea4a5b525d/csrc/u8g2_ll_hvline.c#L106

And here for any tile hit to the right:
https://github.com/olikraus/u8g2/blob/29c6b797072dda4c1a3e8e28aeab96ea4a5b525d/csrc/u8g2_ll_hvline.c#L118

Then here with every tile, that is hit into y direction:
https://github.com/olikraus/u8g2/blob/29c6b797072dda4c1a3e8e28aeab96ea4a5b525d/csrc/u8g2_ll_hvline.c#L138

Time consuming....

vanniaz, 2022-05-23:
OK, thank you very much for your explanations!
I agree with you that updateDisplayArea (based on getAscent, getDescent and getStrWidth) is already a good solution, so I am closing this issue.

closed 2022-05-23 by vanniaz