olikraus/u8g2 #1869: COG Serial SPI 132x32 Graphic LCD Display No Backlight ST7567A

olikraus, 2022-05-25:
### Discussed in https://github.com/olikraus/u8g2/discussions/1868

<div type='discussions-op-text'>

<sup>Originally posted by **markhen57** May 23, 2022</sup>
BuyDisplay have this nice small (low cost!) LCD using the ST7567A with 132x32 glass. It works well with "U8G2_ST7567_PI_132X64_F_4W_SW_SPI u8g2(...", except half the display is not visible (and the RAM usage is 2x what is required).

Would it be possible to request this config be added to the release?

Thanks for great work

Mark</div>

olikraus, 2022-05-25:
Goal is to add support for this display:
https://www.buydisplay.com/cog-serial-spi-132x32-graphic-lcd-display-no-backlight-st7567a

U8G2_ST7567_PI_132X64_F_4W_SW_SPI works, but only the upper part is required.

@markhen57 please remind my on this topic in the view days by adding a comment here (otherwise I might forget this topic until the next release)

olikraus, 2022-05-26:
I found some time to implement the new constructor. It is now called 
U8G2_ST7567_ERC13232_1_4W_SW_SPI

Can you test the u8g2 page mode "FlipMode" example? How will it look like? 
There should be a solid frame all the time.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

markhen57, 2022-05-27:
Hi - thanks for getting to this so fast!

I have followed your instructions to remove the old library and install the latest library. 

I have run the "FlipMode" example and am attaching all the pictures here. I think the frame is correct in all cases?

The only change I made for this test was to change the constructor pinouts to match my hardware:

 "FlipMode" example has this line uncommented:

`U8G2_ST7567_ERC13232_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);` 

I changed to:

`U8G2_ST7567_ERC13232_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 7, /* dc=*/ 9, /* reset=*/ 8);`

However, when I run "HelloWorld" example I do not get the expected result -  I will post MORE info and pictures in my NEXT POST.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

markhen57, 2022-05-27:
As I mentioned above, when I use the new 132x32 constructor (adjusted for my hardware pins) with the "HelloWorld" example, I do not get the expected result. 

With the previous constructor (132x64) and code, the result is shown in first picture - "Hello World!", as expected:

`U8G2_ST7567_PI_132X64_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 7, /* dc=*/ 9, /* reset=*/ 8);`

`void setup(void) {`
` u8g2.begin();`
`}`
`void loop(void) {`
`  u8g2.clearBuffer();`
`  u8g2.setFont(u8g2_font_ncenB08_tr); `
`  u8g2.drawStr(0,10,"Hello World!"); `
`  u8g2.sendBuffer();`
`  delay(1000); `
`}`

[image]

With the new 132x32 constructor, and the same code, the result is shown below (the text is "cut"):

`U8G2_ST7567_ERC13232_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 7, /* dc=*/ 9, /* reset=*/ 8);`

`void setup(void) {`
` u8g2.begin();`
`}`
`void loop(void) {`
`  u8g2.clearBuffer();`
`  u8g2.setFont(u8g2_font_ncenB08_tr); `
`  u8g2.drawStr(0,10,"Hello World!"); `
`  u8g2.sendBuffer();`
`  delay(1000); `
`}`

[image]

markhen57, 2022-05-27:
I have done another experiment that may help to locate the issue above?

The following code gives the CORRECT"Hello World!" display WITH the new 132x32 constructor:

`U8G2_ST7567_ERC13232_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 7, /* dc=*/ 9, /* reset=*/ 8);`

`void setup(void) {`
`  u8g2.begin();`
`  u8g2.setFont(u8g2_font_ncenB08_tr); `  
`}`
`void loop(void) {`
`  u8g2.clearDisplay();`
`  u8g2.firstPage();`
`  do {`
`  u8g2.drawStr(0,10,"Hello World!"); `
` } while (u8g2.nextPage() );`
`  delay(1000); `
`}`

markhen57, 2022-06-01:
You asked me to remind you about this topic in a few days :-)

olikraus, 2022-06-04:
> You asked me to remind you about this topic in a few days :-)

yes, thanks... back again

> the result is shown below (the text is "cut"):

Yes you used a one page constructor with sendBuffer: this will not work

> The following code gives the CORRECT"Hello World!" display WITH the new 132x32 constructor:

Yes this is the correct code for the one page constructor.

So all works as expected?

markhen57, 2022-06-04:
Thank you again for your attention to this!

I guess I should have read the (great) documentation better! I had missed the "1", "2", "F" buffer description. In the updated examples I only saw a 132x32 "1" constructor example, so I used that one. I didn't know/think to try the same constructor with an "F" option. When I do that, it works as I expect (I prefer to use "sendBuffer()" as this is more compatible with a code base I am porting to u8g2).

This is the constructor I am using successfully now (with "sendBuffer()"):

`U8G2_ST7567_ERC13232_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 7, /* dc=*/ 9, /* reset=*/ 8);`

Thanks again!!!

Best regards -

Mark

olikraus, 2022-06-04:
The sendBuffer approach is simpler to use, but will require more RAM. Thanks for the feedback.

closed 2022-06-04 by olikraus