olikraus/u8g2 #1873: Support st7920 with 160x32

BrainHunter, 2022-05-29:
Hi there,

Can you please add support for the ST7920 with 160x32 Pixels?

I tested the helloworld example with the 192x32 Mode which works, but this wastes some RAM and i have No idea If there are other downsides.

For reference:
https://www.pollin.de/p/grafik-lcd-powertip-pg-16032lru-bwh-h-p2-121826

Thanks

olikraus, 2022-06-04:
Any image of the 192x32 constructor? Is everything displayed correctly?
If so, I can indeed just reduce the width from192 to 160

shlyakpavel, 2022-06-07:
Please, keep 192
We use it!

olikraus, 2022-06-07:
I would have created a 160x32 constructor ADDITIONALLY.
Is the 160x32 still needed?

BrainHunter, 2022-06-09:
Hi there,

With the 192x32 constructor everything looks fine for me, so it would be nice if you could add the 160x32 constructor. 
Thanks!

Sorry for the Closing and reopen.

olikraus, 2022-06-12:
I have created u8g2 beta release 2.33.4. It includes the 160x32 constructor. 
Can you do a test with the new constructor? If you run the flipmode example in page mode, then there should be a frame visible all the time close to the border.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

BrainHunter, 2022-07-17:
Hi there,
i finally got to test the new constructor.
Works fine!

Thanks for your fast help and of course for the great library!

olikraus, 2022-07-17:
Great, thanks for the feedback

closed 2022-08-13 by olikraus