olikraus/u8g2 #1874: strange gfx issue with esp8266

pike2k, 2022-05-30:
I have a project where I had an odd Esp8266 that took 7-12Vin and did not work with 5V except for USB. It was was marked LoLin NodeMCU v3.

This ESP8266 had no gfx issue

I then replaced this with a different model of a unit marked (you guessed it) LoLin NodeMCU v3. This unit however does NOT take 12V on Vin (magic smoke). It does however work with 5V applied to Vin (I had more units).
This 5V unit however has 1 or maybe 2,3 "stuck" pixels down in the right corner. If I swith to a different Oled (these are 128x32 btw) it shows a similar issue.

This is the line I use: U8G2_SH1106_128X32_VISIONOX_F_HW_I2C oled(U8G2_R2, /* reset=*/ U8X8_PIN_NONE);

Is it known that different ESP8266 running same code produces different output with u8g2?
Any way around this gfx glitch?

olikraus, 2022-06-04:
so we can close this issue?

pike2k, 2022-06-06:
Sadly this issue is random and comes and goes. I have not yet found a common thread :(

olikraus, 2022-06-06:
Maybe communication is instable. Did you ever try  software emulated I2C with u8g2?

closed 2023-02-22 by olikraus