olikraus/u8g2 #1875: Is the U8G2LOG class not wellworking, when used as a inheritable class?

hhuter, 2022-06-01:
My dear Oli Krauses and all the other C++ foxes around! 
1. Very THX, Oli, for sharing.
2. I need some C++ explantions about your lovely u8g2 library

* I use a SSD1306 OLED 128x64 display - I2C type, hardwired on
* the Arduino Uno's hardware I2C.
* I am using the Arduino IDE.

My problems arises, when I derive a class from U8G2LOG

Explanation:
* I want to replace the Serial Monitor, with a 128x64 display to emulate 
a 16 character x 4 lines scrolling display with the wellknown and simple print() functionality.
(mainly for steady monitoring my other projects functionality without having my PC via USB connected)
Now: _MORE_ displays should be driven on _ONE_ microcontroller.
* My intention is to hardwire _more_ then one LOG - Displays
(on an Arduino MEGA, an ESP32 ...) and to keep the SW usage as straight and simple as it could be and as it should be.

First, this works great and exactly as expected:

```c++
// #include <Arduino.h>
#include <U8g2lib.h>

U8G2_SSD1306_128X64_NONAME_2_HW_I2C u8g2(U8G2_R0);

#define U8LOG_W  16
#define U8LOG_H  4
uint8_t u8log_buffer[U8LOG_W*U8LOG_H];

U8G2LOG MyDisp;  // the U8G2LOG objact

void MyDisp_begin() {
  u8g2.begin();                     // use graphic mode
  u8g2.setFont(u8g2_font_t0_16_mr);
  MyDisp.begin(u8g2, U8LOG_W, U8LOG_H, u8log_buffer);  // auto-update
  MyDisp.setLineHeightOffset(3);
  MyDisp.setRedrawMode(0);          // Update screen with every newline 
}

void setup(void) {
  MyDisp_begin();      // Initialisation:     MyDisp
  Serial.begin(9600);  // and the 'classic':  Serial
}

void loop(void) {

  Serial.print("msec:  0x");
  Serial.println(millis(),HEX);  

  /* and now exactly the same on my SSD1306 OLED display: */
  MyDisp.print("msec:  0x");
  MyDisp.println(millis(),HEX);  // Hurra ! The same is displayed

  delay(1000);
}

```

Next: with the above given 'template' I want to have
* _all_ U8G2LOG parts packaged into _ONE_ class
* for simple creating more then one objects to drive more then one hardwired 128x64 displays

```c++
// #include <Arduino.h>
#include <U8g2lib.h>

#define U8LOG_W  16
#define U8LOG_H  4

// Plug all U8G2LOG parts into _ONE_ class:
class MYDISP : public U8G2LOG
{
    U8G2_SSD1306_128X64_NONAME_2_HW_I2C  u8g2b;  // part: graphic (must done initialised!)
    uint8_t u8log_buffer[U8LOG_W*U8LOG_H];       // part: buffer  (this needs not to be initialized)
  
  public:
    // this is the constructor (it is a default ctor too):
    MYDISP(const u8g2_cb_t *rotation = U8G2_R0) :
      // initialisation list: doing explizit (the inherited) and the new members constructor(s):
        // U8G2LOG{},     //  inheritance: init. not needed, see u8g2lib.h ... the constructor does nothing
        u8g2b{rotation}   //  initialisation of this graphic part member
    {
        u8g2b.begin();             // use graphic mode
        u8g2b.setFont(u8g2_font_t0_16_mr); 
        this->begin(u8g2b, U8LOG_W, U8LOG_H, u8log_buffer);  // auto-update
        this->setLineHeightOffset(3);    
        this->setRedrawMode(0);    // Update screen with every newline  
    }

    void test() {
      this->println("Hey man");  // even inside this classes objects: the  println()  does not work
    } 
};

// Now testing this class: ===========================================================
//  First: instantiate this class + (plus!) initialise it with (the my prefered) settings:

// MYDISP MyDisp;        // either default constructor,
// MYDISP MyDisp {};     // ... default constructor,
MYDISP MyDisp {U8G2_R0}; // ... or ctor with argument
//  but! this, too (!), makes the Serial monitor disfunctional *) too (?! Why)

void setup(void) {
  Serial.begin(9600);   // for checking the Serial monitor
  // but even this, the Serial is nonfunctial anyfurther, as soon I created my object see *)
}

void loop(void) {

  Serial.print("msec:  0x");    // does no longer work: it does _nothing_ anymore
  Serial.println(millis(),HEX);  

  /* and now exactly the same on my SSD1306 OLED display: */
  MyDisp.print("msec:  0x");    // it displays nothing
  MyDisp.println(millis(),HEX);

  MyDisp.test();                // also this does displays nothing

  delay(1000);
}
```

Both the above codes should behave exactly identical!

But my 2nd approach:
* does _not_ work to drive my display
* plus: it _trashes_ the Serial Monitor functionality: arghh%&/(§$%

I already checked the Oli Kraus's source code:
* he did not implement the  "print()"  functionality directly,
* but he redirected the arduinos  print()  function in some way, which my C++ knowledge do not understand.

Some explanation would help me to enhance the my C++ language knowledge better, and better, and ...

Thank you very much
Johannes

olikraus, 2022-06-05:
> he did not implement the "print()" functionality directly,

Yes, the code is reused. Otherwise the same code would exist twice in the memory of your controller, wasting flash memory.

> but he redirected the arduinos print() function in some way, which my C++ knowledge does not understand.

Not much C++ magic: I just derive from the Arduino print class (if the code is compiled under Arduino environment):
``` C++
class U8G2LOG
#ifdef ARDUINO
: public Print
```

> Some explanation would help me to enhance the my C++ language knowledge better, and better, and ...

Not sure what other explanation I can give (how class inheritance works should be explained better in some text books), but I could commend your code a little bit:

The member class initalization expects arguments in inside () not inside {}, so souldn't it be
``` C++
u8g2{rotation}
```
replaced by
``` C++
u8g2{rotation}
```

Below
``` C++
MYDISP MyDisp {U8G2_R0}; // ... or ctor with argument
```
Is the same problem: You need to use () instead of {}. In fact using {} will call a not defined default constructor of your class.

There might be another problem: Arduino is not C++. Arduino modifies your C++ so that some things which are required in C++ are not required any more. On the other side some C++ coding styles my not work any more. This not documented, only my personal experience, but it makes working with classes very hard under Arduino.

I personally do not see an issue inside u8g2 here, instead your code looks incorrect to me.

hhuter, 2022-06-08:
Hello Oliver,

1. No, not really:
I changed the curly braces against your suggested round braces:
 -->
But the Arduino binary (the "hex-file") are __exactly and bitident
the same__,  both: curly (initialisation-list) and rounded (parameter list) compiles to exactly the same nonworking code (as given in the  __hex-file__ )
(And clearly as directly uploaded to the Arduino device, it shows the same "nonbehaviour")

2. You wrote
"There might be another problem: Arduino is not C++. Arduino modifies your C++ so that some things which are required in C++ are not required any more. On the other side some C++ coding styles my not work any more. This not documented, only my personal experience, but it makes working with classes very hard under Arduino."

"Arduino is not C++": yes, somehow yes, for example, I see this
* autoincluding the arduino.h
* selfchaining the building process
* and some more stupid automatisms and restrictions (e.g. the folder must be samenamed as the .ino sourcefile, which it does contain)

But still I think it is a more or less a transparent GCC toolchain with a more or less "pure" C++ ?

Thx a lot for your guidance and the hints.
Have a good week.
Johannes

olikraus, 2022-06-08:
> But still I think it is a more or less a transparent GCC toolchain with a more or less "pure" C++ ?

Arduino applies a preprocessing step. The generted cpp file should be there in the build folder.

hhuter, 2022-06-08:
thx oliver,

i will stay on this your debugging hint. sometimes the next days or weeks. or so.
it is not an urgent problem. and - it is just a beautifing elegance problem.

liebe grüsse nach deutschland
von johannes aus österreich

! is there any other c++ gcc arduino nerd here? hello world ! where are you ?

closed 2023-02-22 by olikraus