olikraus/u8g2 #1876: Need help on displaying text data with smooth Scroll on 72x40 OLED

vijayprakash1, 2022-06-03:
Hi @olikraus,

I am not sure whether I should post my issue here but you have developed the many libraries so thought you might help me on solving my issue with displaying text data with a smooth Scroll on **72x40 OLED using nRF52832 MCU**. 
I have written my own C library with referring to U8g2 

The issue I am facing is when I try to display text data on the OLED with smooth scrolling, the only first half of the display working, and in the second half of the display no data is being displayed

e.g:  If I want to display the below data 

**Hello, how are you?
doing today vijay?**   

Text getting displayed on the OLED is 
**Hello, how a
doing today**

I am attaching my code snip for the reference, please help me where I am doing wrong

[Oled_scroll_issue.pdf](https://github.com/olikraus/u8g2/files/8832715/Oled_scroll_issue.pdf)

vijayprakash1, 2022-06-03:
void OLED_INIT(){

     send_command(SSD1306_DISPLAYOFF);        //turn off oled panel
      send_command(SSD1306_SETSTARTLINE);     // set display start line
      send_command(SSD1306_SETCONTRAST);      // contrast level 250
      send_command(0XFF);
      send_command(SSD1306_MEMORYMODE);       //set memory addressing mode, page addressing
      send_command(0X02);
      send_command(SSD1306_SEGREMAP);         //set segment remap 128 to 0
      send_command(SSD1306_DISPLAYALLON_RESUME);
      send_command(SSD1306_NORMALDISPLAY);
      send_command(SSD1306_SETMULTIPLEX);      //1/40 duty multiplex ratio
      send_command(0X27);
      send_command(SSD1306_COMSCANINC);     // column scan direction normal
      send_command(SSD1306_SETDISPLAYOFFSET); // no offset
      send_command(0X00);
      send_command(SSD1306_SETDISPLAYCLOCKDIV);   // the suggested ratio 0x80
      send_command(0X80);
      send_command(SSD1306_SETPRECHARGE);     // pre-charge period (1, 15)
      send_command(0X22);
      send_command(SSD1306_SETCOMPINS);    // 0x12 if height > 32 else 0x02
      send_command(0X12);
      send_command(SH1106_SET_PUMP_MODE);  //internal IREF setting
      send_command(0X30);

      send_command(SSD1306_SETVCOMDETECT);   // vcomh regulator level
      send_command(0X20);

      send_command(SSD1306_CHARGEPUMP);  // charge pump disabled
      send_command(0X14);
      send_command(SSD1306_DISPLAYON);
      send_command(SSD1306_COMSCANDEC);
      send_command(SSD1306_MIRROR); //mirror

         nrf_delay_us(1000);

      send_command(OLED_SET_DISPLAY_OFF); //turn off display
      is_oled_on = false;

      clear_display();
}

Here we  place our data in oled_buffer and we run the below function to display the message on OLED:  

Note: oled_buffer contains ASCII values.

[  We start printing from OLED page zero which is 0xb0 ]

void Display_Data_ON_Oled(uint8_t OLED_DATA_BUFF[OLED_MAX_PAGE][OLED_MAX_COLUMN], uint8_t __u8NextLine, uint8_t OledPage_END, uint8_t scroll, uint8_t _Final) 
{
 
  
  unsigned char OLED_PAGE, j, OledBuffIdx;
  int i;

  //JUST SCROLL CMD
  if (scroll) {

    smooth_scroll_OLED(UP, 1);
    scroll = 0;
  }

  send_command(OLED_SET_DISPLAY_ON); //turn on oled panel
  nrf_delay_us(100);

  OLED_PAGE = __u8NextLine;

  for (OledBuffIdx = 0; OledBuffIdx < OledPage_END; OledBuffIdx++) {

  
  
     send_command(0xb0 + OLED_PAGE);
      setCol(0);
    // SEG 0-63 //72
    for (j = 0; j < 72; j++) {
     
 send_data(OLED_DATA_BUFF[OledBuffIdx][j]);

    }

    OLED_PAGE++;

  }
  }
}

Using the below function we scroll the OLED pages

void scrollDisplay(int8_t lines) 
{
 setStartLine(m_startLine + lines);

}

void smooth_scroll_OLED(uint8_t ScrollDirection,uint8_t scroll)
{
  if(ScrollDirection)
  {   for(int s =0; s < 8;s++)
      {
           scrollDisplay(scroll);
           nrf_delay_ms(10);
      }
  }

olikraus, 2022-06-04:
In order to support on issues I have to invest time.
Investing time for me is ok if this is related to my projects and as a consequence many users can benefit from it. 
This is, however, not the case in your request.

Additionally you request is complicated, which increases the amount of time dramatically:
 - The error description is poor
 - The provided code is unknown to me
 - The formating of the code in the above poor (and unreadable)

closing this...

closed 2022-06-04 by olikraus