olikraus/u8g2 #1882: parallel communication with ST7920 works with u8glib but not u8g2

smeegle5000, 2022-06-14:
i'm trying to setup an st7920 128x64 display for parallel communication but after adapting the only example code i could find for u8g2 the display remains blank, but when running the original it works fine, i'm not sure if i'm doing anything wrong but if so any guidance would be greatly appreciated

display is wired as follows for both sketches 

display pin 1 (GND) gnd
display pin 2 (VCC) +5v
display pin 3 (V0) n/c
display pin 4 (RS) pin 9
display pin 5 (R/W) pin 10
display pin 6 (E) pin 8
display pin 7 (DB0) pin 0
display pin 8 (DB1) pin 1
display pin 9 (DB2) pin 2
display pin 10 (DB3) pin 3
display pin 11 (DB4) pin 4
display pin 12 (DB5) pin 5
display pin 13 (DB6) pin 6
display pin 14 (DB7) pin 7
display pin 15 (PSB) +5v
display pin 16 (NC) n/c
display pin 17 (RST) n/c
display pin 18 (VOUT) n/c
display pin 19 (BLA) +3v3
display pin 20 (BLK) GND

u8glib version
```
#include "U8glib.h"
 
//Pins on Croduino-> 8 bit communication: D0..D7: 0, 1, 2, 3, 4, 5, 6, 7, E=8, RS=9, RW=10, PSB = +5V, RST = +5V, 
U8GLIB_ST7920_128X64_4X u8g(0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10);
 
void setup() {

}
 
void loop() {
 u8g.firstPage(); //This is the function that starts drawing on the display 
 do { //The display is divided into more pages (parts).
 draw(); //We call the function that contains everything we want to display  
 } while ( u8g.nextPage() ); //After it has displayed one page, go to the next one, until it shows all pages or the entire display
 delay(990); //Wait a bit until the next display 
}
 
void draw() {
 u8g.setFont(u8g_font_fixed_v0); //Select which font you want to print out on the display (list of fonts: https://github.com/olikraus/u8glib/wiki/fontsize)
 u8g.setFontPosTop(); //We want the reference position for printing the string to be in the upper left corner of that string 
 u8g.setPrintPos(0,0); //X i Y coordinates where the string wants to print out on the display 
 u8g.print("test"); //Display of the desired string on the default position
 u8g.setPrintPos(30,50); //Changing coordinates for displaying the new string 
 u8g.print(millis()/1000); //Printing of the new string on the display 
}
```
u8g2 version
```
#include "U8g2lib.h"
 
//Pins on Croduino-> 8 bit communication: D0..D7: 0, 1, 2, 3, 4, 5, 6, 7, E=8, RS=9, RW=10, PBS = +5V, RST = +5V
U8G2_ST7920_128X64_F_6800 u8g2(U8G2_R0, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10);

 
void setup() {

}
 
void loop() {
 u8g2.firstPage(); //This is the function that starts drawing on the display 
 do { //The display is divided into more pages (parts).
 draw(); //We call the function that contains everything we want to display  
 } while ( u8g2.nextPage() ); //After it has displayed one page, go to the next one, until it shows all pages or the entire display
 delay(990); //Wait a bit until the next display 
}
 
void draw() {
 u8g2.setFont(u8g2_font_t0_16_mr); //Select which font you want to print out on the display (list of fonts: https://github.com/olikraus/u8glib/wiki/fontsize)
 u8g2.setFontPosTop(); //We want the reference position for printing the string to be in the upper left corner of that string 
 u8g2.setCursor(0,0); //X i Y coordinates where the string wants to print out on the display 
 u8g2.print("test"); //Display of the desired string on the default position
 u8g2.setCursor(30,50); //Changing coordinates for displaying the new string 
 u8g2.print(millis()/1000); //Printing of the new string on the display 
}
```

olikraus, 2022-06-14:
It should work, see here: https://github.com/olikraus/u8g2/wiki/gallery#26-nov-2016-st7920-128x64-lcd-in-8080-parallel-mode

I do not remember exactly whether the arg sequence is the same between u8g and u8g2. Additionally try 8080 mode instead of 6800.

smeegle5000, 2022-06-15:
it worked! issue was that the r/w pin needed to be set low, also it seems something was not working correctly in the code, 6800 mode worked fine as did 8080, here's the working code and a pinout for anyone that may come across the same thing later on

display pin 1 (GND) gnd
display pin 2 (VCC) +5v
display pin 3 (V0) n/c
display pin 4 (RS) pin 9
display pin 5 (R/W) gnd
display pin 6 (E) pin 8
display pin 7 (DB0) pin 0
display pin 8 (DB1) pin 1
display pin 9 (DB2) pin 2
display pin 10 (DB3) pin 3
display pin 11 (DB4) pin 4
display pin 12 (DB5) pin 5
display pin 13 (DB6) pin 6
display pin 14 (DB7) pin 7
display pin 15 (PSB) +5v
display pin 16 (NC) n/c
display pin 17 (RST) pin 10
display pin 18 (VOUT) n/c
display pin 19 (BLA) +3v3
display pin 20 (BLK) GND

```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_ST7920_128X64_F_8080 u8g2(U8G2_R0, 0, 1, 2, 3, 4, 5, 6, 7, /*enable=*/ 8 /* A4 */, /*cs=*/ U8X8_PIN_NONE, /*dc/rs=*/ 9 /* A3 */, /*reset=*/ 10 /* A1 */);  // Remember to set R/W to 0 

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();         // clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr); // choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");  // write something to the internal memory
  u8g2.sendBuffer();          // transfer internal memory to the display
  delay(1000);  
}
```
thankyou

closed 2022-06-15 by smeegle5000