olikraus/u8g2 #1883: .print function not working on Stm32 F411CEU

jesman2, 2022-06-14:
I have a problem with the u8g2.print command on my Stm32 F411(using arduino ide with stm32 extension btw). When i want to print anything, let it be just plain text or some variables. The .print command does nothing. I have tried to change the font, change draw color and some other stuff but i can't get it to work. The drawStr. command works perfectly. I made a "random_variable" to try to print that, but it also doesn't print variables. So my questions are:

1. Is this a bug?
2. Is there another possibility to display variables?
3. Did i do something wrong? :)

code below:

#include <OneButton.h>
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

//###########THERMISTOR PINS##################
#define ThermIn_RV PB1
#define ThermIn_RA PB0
#define ThermIn_LV PA7
#define ThermIn_LA PA6

//#############HEATER PINS####################
#define Heater_RV PB13
#define Heater_RA PB14
#define Heater_LV PB15
#define Heater_LA PA8

//#############BUZZER PIN####################
#define Buzzer PA0
#define Buzzer_Freq 2400

//#############BUTTON PINS###################
#define Down_Button PA5
#define Up_Button PA4
#define Back_Select PA1

int buttonState;

unsigned long timePress = 0;
unsigned long timePressLimit = 0;
int clicks = 0;
/*
  U8g2lib Example Overview:
    Frame Buffer Examples: clearBuffer/sendBuffer. Fast, but may not work with all Arduino boards because of RAM consumption
    Page Buffer Examples: firstPage/nextPage. Less RAM usage, should work with all Arduino boards.
    U8x8 Text Only Example: No RAM usage, direct communication with display controller. No graphics, 8x8 Text only.
    
*/
//#######################DISPLAY VARIABLES#######################
U8G2_SSD1309_128X64_NONAME0_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ PB8, /* data=*/ PB9, /* cs=*/ PB7, /* dc=*/ PB6, /* reset=*/ PB5); 

//#######################MENU VARIABLES#######################
int selected = 0;
int entered = -1;

//#######################THERMISTOR VARIABLES#######################
int Vo_RV, Vo_RA, Vo_LV, Vo_LA;
float R1 = 10000;
float logR2_RV, R2_RV, T_RV;
float logR2_RA, R2_RA, T_RA;
float logR2_LV, R2_LV, T_LV;
float logR2_LA, R2_LA, T_LA;
float c1 = 1.009249522e-03, c2 = 2.378405444e-04, c3 = 2.019202697e-07;

float random_variable = 200;

void setup(void) {
  pinMode(PA1,INPUT_PULLDOWN);
  pinMode(PA4,INPUT_PULLDOWN);
  pinMode(PA5,INPUT_PULLDOWN);
  u8g2.begin();
  delay(10);
  u8g2.enableUTF8Print();
  u8g2.clearBuffer();          // clear the internal memory
  u8g2.setFont(u8g2_font_tom_thumb_4x6_mf ); // choose a suitable font
  u8g2.drawStr(0,10,"CYKA BLYAT!");  // write something to the internal memory
    u8g2.setCursor(0, 20);
    u8g2.print(random_variable);
  u8g2.sendBuffer();          // transfer internal memory to the display 

}

void loop(void) {
  displaymenu();
}

void Thermistor_read(void){

//##########THERMISTOR RECHTS-VOOR##########
  Vo_RV = analogRead(ThermIn_RV);
  R2_RV = R1 * (1023.0 / (float)Vo_RV - 1.0);
  logR2_RV = log(R2_RV);
  T_RV = (1.0 / (c1 + c2*logR2_RV + c3*logR2_RV*logR2_RV*logR2_RV));
  T_RV = T_RV - 273.15;
//##########THERMISTOR RECHTS-ACHTER##########
  Vo_RA = analogRead(ThermIn_RA);
  R2_RA = R1 * (1023.0 / (float)Vo_RA - 1.0);
  logR2_RA = log(R2_RA);
  T_RA = (1.0 / (c1 + c2*logR2_RA + c3*logR2_RA*logR2_RA*logR2_RA));
  T_RA = T_RA - 273.15;
//##########THERMISTOR LINKS-VOOR##########
  Vo_LV = analogRead(ThermIn_LV);
  R2_LV = R1 * (1023.0 / (float)Vo_LV - 1.0);
  logR2_LV = log(R2_LV);
  T_LV = (1.0 / (c1 + c2*logR2_LV + c3*logR2_LV*logR2_LV*logR2_LV));
  T_LV = T_LV - 273.15;
//##########THERMISTOR LINKS-ACHTER##########
  Vo_LA = analogRead(ThermIn_LA);
  R2_LA = R1 * (1023.0 / (float)Vo_LA - 1.0);
  logR2_LA = log(R2_LA);
  T_LA = (1.0 / (c1 + c2*logR2_LA + c3*logR2_LA*logR2_LA*logR2_LA));
  T_LA = T_LA - 273.15;
  
}

void displaymenu(void) {

  int down = digitalRead(Down_Button);
  int up = digitalRead(Up_Button);
  int enter = digitalRead(Back_Select);
  int back = digitalRead(Back_Select);
  if (up == HIGH && down == HIGH) {
  };
  if (up == HIGH) {
    selected = selected + 1;
    delay(200);
  };
  if (down == HIGH) {
    selected = selected - 1;
    delay(200);
  };
  if (enter == HIGH) {
    entered = selected;
  };
  if (back == HIGH) {
    entered = -1;
  };
  const char *options[7] = {
    " Menu 1",
    " Menu 2",
    " Menu 3",
    " Menu 4 ",
    " Menu 5 ",
    " Menu 6 ",
    " Menu 7 "
  };

  if (entered == -1) {
    u8g2.clearBuffer();
    u8g2.drawStr(0,10,"TestMenu");
    for (int i = 0; i < 7; i++) {
      if (i == selected) {
        u8g2.print(options[i]);
      } else if (i != selected) {
        u8g2.print(options[i]);
      }
    }
  } else if (entered == 0) {
    u8g2.clearBuffer();
    u8g2.setFont(u8g2_font_tom_thumb_4x6_mf );
    u8g2.setCursor(0, 0);
    u8g2.print(F("Dsn menu test"));
    u8g2.print("Menu option 1");
    u8g2.print("Dsn Menu system");
  } else if (entered == 1) {
    u8g2.clearBuffer();
    u8g2.setCursor(0, 0);
    u8g2.print(F("Dsn menu test"));
    u8g2.print("Menu option 2");
    u8g2.print("Dsn Menu system :)");
  }
  else if (entered == 2) {
    u8g2.clearBuffer();
    u8g2.setCursor(0, 0);
    u8g2.print(F("Dsn menu test"));
    u8g2.print("Menu option 3");
    u8g2.print("Dsn Menu system:)");
  } else if (entered == 3) {
    u8g2.clearBuffer();
    u8g2.setCursor(0, 0);
    u8g2.print(F("Dsn menu test"));
    u8g2.print("Menu option 4");
    u8g2.print("Dsn Menu system:)");
  } else if (entered == 4) {
    u8g2.clearBuffer();
    u8g2.setCursor(0, 0);
    u8g2.print(F("Dsn menu test"));
    u8g2.print("Menu option 5");
    u8g2.print("Dsn Menu system:)");
  } else if (entered == 5) {
    u8g2.clearBuffer();
    u8g2.setCursor(0, 0);
    u8g2.print(F("Dsn menu test"));
    u8g2.print("Menu option 6");
    u8g2.print("Dsn Menu system:)");
  } else if (entered == 6) {
    u8g2.clearBuffer();
    u8g2.setCursor(0, 0);
    u8g2.print(F("Dsn menu test"));
    u8g2.print("Menu option 7");
    u8g2.print("Dsn Menu system:)");
  }
  u8g2.sendBuffer();
}

olikraus, 2022-06-14:
The print function is just called from the Arduino core library (print() is not part of u8g2). Maybe the STM32 implementation has an issue here.

jesman2, 2022-06-14:
Ah ok, sorry for bothering you. I will look into it on the stm32 side then. Thanks for making the u8g2lib though!

closed 2022-06-14 by jesman2