olikraus/u8g2 #1885: ST75256 Pixels grey that should not be active

millerp6ATWIT, 2022-06-15:
I'm having a problem with the ST75256 where it is displaying incorrectly for every HW SPI constructor. I don't know my exact version of the ST75256 as its unspecified from the manufacturer, so I've tried them all. I am just trying to run the graphics test on an Adafruit Huzzah ESP32. With some ST75256 constructors such as "U8G2_ST75256_JLX172104_F_4W_HW_SPI", I get no output, some outputs look faded and in odd parts of the screen but all other pixels are clear, and some look like they are in the right place and start out looking crystal clear but after the screen updates a few times they all become a muddled grey but the images can still be made out. I believe that is with this constructor: "U8G2_ST75256_WO256X128_F_4W_HW_SPI". They don't span the whole screen but I don't know if this is expected of the graphics test anyway. 

[Here's](https://photos.app.goo.gl/yBKuCvE1EiKgH7K28) a video with the U8G2_ST75256_WO256X128_F_4W_HW_SPI constructor. It's the one that seems to be working best although a couple other's behave similarly. I think it would be perfect if not for the display being dark. You can also see that at the beginning of the test it starts off clear and quickly becomes dark in the other pixels on the screen. 

I'm quite new to all this so your help is greatly appreciated. Thanks!

millerp6ATWIT, 2022-06-15:
Oliver, thank you for your fast response. This makes sense I think you are correct that it is a power or capacitance issue. Currently the display is being powered off of the 3.3v on my ESP32 and the backlight is from a separate power supply. There is no PCB just a ribbon cable and a breakout so likely I have to change my wiring.  Sorry for the poor video its what I managed to capture last night before throwing in the towel. I'll take another look later tonight and close the issue if this solves it. I believe you are correct. 

Thanks!

millerp6ATWIT, 2022-06-15:
Update, I switched the 3.3v in to the power supply to see if that would fix the issue, but no success there. I then played around with the capacitors but nothing really improved. I then plugged it back into the 3.3v from the ESP32 with the different capacitor configuration and BAM all better. Not sure why its worse with my desktop power supply than with the ESP but its working now. I'm curious if you might know why the esp supply looks better than with the bench supply but I'll mark this as closed if you don't have any final thoughts. 
Thanks!

millerp6ATWIT, 2022-06-15:
Oh also sorry for asking about a second topic let me know if this is something to ask in another thread, but I have a display using an IST3088 and cannot for the life of me find a library for it. I don't know if this is similar enough to the IST3020 in the u8g2 library to get working with some tweaks. Haven't even tried it yet but if you have any input it would be appreciated. There's not much info on the IST3088 out there. 
Thanks!

olikraus, 2022-06-16:
There is no support for the IST3088 at the moment. You could create another issue for the IST3088, but I would need someone for testing.

millerp6ATWIT, 2022-06-17:
Thanks, I have a display with the driver, I will wire it up and create a new issue when I get a chance.

olikraus, 2022-06-17:
Any link to the IST3088 display?

millerp6ATWIT, 2022-06-17:
It's actually a sample that I was sent by a display manufacturer for a small product I'm trying to develop. But I can't find a library that supports it. [Here's](https://ancoretraining-my.sharepoint.com/:f:/p/paul_miller/EoH0dsCtSJNIn7BHiit_AmEBDvSY_u1TcvMhd6J9wTSRTw?e=3sC2RF) (password: u8g2) all of the documentation I have on the display and the driver. I don't know where you can buy another display using the chip, but I have 3 and would be happy to post one to you if it would allow you to add support for the driver. 

I've looked at the documentation and its pretty far beyond my knowledge.

olikraus, 2022-06-17:
I could try to add the driver, if you are able to test the code. There might be several tests and recursions required, but it seems to be doable from what i saw in the datasheet.

millerp6ATWIT, 2022-06-17:
Thanks, I'll open a new issue.