olikraus/u8g2 #1886: trying to port u8g2 on stm32 nucleo board

marconcellos, 2022-06-17:
Hi,
i've tried to use a YES optoelectronics e-ink monochrome display (YMS200200 1,54" 200x200 ssd1681), I'm using a stm32f302 nucleo board connecting display by SPI.
I've imported u8g2 and used u8x8_d_ssd1681_200x200.c by Pikot.
this is my interface module:
*extern SPI_HandleTypeDef hspi3;
extern uint8_t u8x8_d_ssd1681_200x200(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);
static uint8_t u8x8_gpio_and_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);
static uint8_t u8x8_byte_4wire_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);

void display_init(u8g2_t *u8g2)
{
  if (u8g2 == NULL) {
	return;
  }

  u8g2_SetupDisplay(u8g2, &u8x8_d_ssd1681_200x200, u8x8_cad_011, &u8x8_byte_4wire_hw_spi, &u8x8_gpio_and_delay);

  static uint8_t frame_buffer[200 * 200 / 8];
  u8g2_SetupBuffer(u8g2, frame_buffer, u8g2->u8x8.display_info->tile_height, u8g2_ll_hvline_horizontal_right_lsb, U8G2_R3);

  u8g2_InitDisplay(u8g2);

  // Put the display to sleep.
  //u8g2_SetPowerSave(u8g2, 0U);

}

static uint8_t u8x8_gpio_and_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	  switch (msg) {
	  case U8X8_MSG_GPIO_AND_DELAY_INIT:
	    // Configure the CS pin.
		  u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);

		  HAL_Delay(1);
	    break;

	  case U8X8_MSG_DELAY_MILLI:
	    // A delay of 0 means wait until ready.
	    if (arg_int == 0U) {
	    	do{

	    	}while(HAL_GPIO_ReadPin(GPIOC, GPIO_PIN_2) == 1);
	    }
	    else {
	    	HAL_Delay(arg_int);
	    }
	    break;

	  case U8X8_MSG_GPIO_CS:
		  HAL_GPIO_WritePin(GPIOA, GPIO_PIN_4, arg_int);
	    break;

	  case U8X8_MSG_GPIO_DC:
		  HAL_GPIO_WritePin(GPIOC, GPIO_PIN_0, arg_int);
	    break;

	  case U8X8_MSG_GPIO_RESET:
		  HAL_GPIO_WritePin(GPIOC, GPIO_PIN_1, arg_int);
	    break;

	  default:
	    return 0U;
	  }

	  return 1;
}

static uint8_t u8x8_byte_4wire_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  switch (msg)
  {
  case U8X8_MSG_BYTE_SEND:
	HAL_SPI_Transmit(&hspi3, (uint8_t *) arg_ptr, arg_int, 10000);
	break;
  case U8X8_MSG_BYTE_INIT:
	break;
  case U8X8_MSG_BYTE_SET_DC:
	  //HAL_GPIO_WritePin(GPIOC,  GPIO_PIN_0, arg_int);
	  u8x8_gpio_SetDC(u8x8, arg_int);
	break;
  case U8X8_MSG_BYTE_START_TRANSFER:
	  u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);
	break;
  case U8X8_MSG_BYTE_END_TRANSFER:
	  u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
	break;
  default:
	return 0;
  }

  return 1;
}*

In  u8g2_InitDisplay(u8g2); init and powersave sequences are launched, just powersave sequence produce a flashing of the display, while init sequence and following drawing instructions (in the main.c) doesn't produce nothing.

marconcellos, 2022-06-17:
P.S. I've also tried to invert two instructions in
u8x8_d_ssd1681_D67_200x200_init_seq[]
to make it compliant to the order written in the datasheet of the display

olikraus, 2022-06-17:
Not sure how I can support here. There is no support for the ssd1681 in u8g2.

marconcellos, 2022-06-17:
@oliraus, sorry.
I've used the file u8x8_d_ssd1681_200x200.c written by Pikot, but I couldn't rise an issue on his repository.

closed 2023-02-22 by olikraus