olikraus/u8g2 #1887: IST3088 Support

millerp6ATWIT, 2022-06-17:
Continuing discussion in this thread [here](https://github.com/olikraus/u8g2/issues/1885) on trying to add IST3088 support. I'll put together a schematic on how I plan to wire it, would you mind giving it a look before I do so that I don't fry anything?

olikraus, 2022-06-18:
Sure, I can review the schematic...

millerp6ATWIT, 2022-06-18:
[Here](https://ancoretraining-my.sharepoint.com/:b:/p/paul_miller/EVZXKcGkBvlGmQytjRG5XWoBmbWT75G3UOFiePg8TlzZNQ?e=p31ZnI) is the schematic for how I think I should wire it based off the data sheet. For the booster circuit I picked 1uF for C1 and C3 as it was in range for both. But it was in the lower range for C1 and upper range for C3, I don't know how much this will affect things. For C2 I picked 2.2uF, again just picking out of the range given in the datasheet. Let me know if you see anything that needs to change. Thanks.

olikraus, 2022-06-18:
You should use as often as possible ceramic caps. Reset and RS (=A0) must be connected to two different GPIO pins.
Any reason why you didn't add a capacitor to C24 pins?

millerp6ATWIT, 2022-06-18:
Yeah I'll use three separate gpio pins. I wasn't sure about the caps because they're labeled positive and negative so I didn't know if I would need to use polar caps but I'll use ceramic, that makes it easier. 

On page 27 in the manual it has some wiring examples on the capacitor pins and it says to connect C24 for a 10x boost in drive voltage, which would put it at 33V if I give it a 3.3V on VCI. And the max drive voltage is 28V according to the manual so I opted for 8x.  

[Here's](https://ancoretraining-my.sharepoint.com/:i:/p/paul_miller/EbaFM5kPYOBDnnDz6ecIWDQBvRoLzqq4vLa0fgxLj_IpUQ?e=radRbu) the page.

olikraus, 2022-06-19:
> ... so I opted for 8x.

I just wonder why the display vendor decided to add these capacitor lines to the flex cable.

olikraus, 2022-06-19:
I started to work on the code. Here are some notes which are not clearly described in the datasheet. Needs to be checked once the hardware is available.

 * Index value seems to be 16 bit, so the high part is always zero, RS must be low during the transfer of the 16 bit
 * Contrast value (index 5) is not specified, needs to be figured out for the display
 * SLP/STB (index 3) are not clear to me, assigning 0 as of now. 
 * Unclear: Polarity Control, V0V4 Adjustment, Contrast Offset
 * U8g2 Flip Mode can be implemented via register 1, but exact setting is not clear

olikraus, 2022-06-19:
Here is a first version of the init sequence with the above mentioned open issues:

https://github.com/olikraus/u8g2/blob/eab9695accce46ef46a8e71ece8bdf99a8420708/csrc/u8x8_d_ist3088.c#L175-L191

millerp6ATWIT, 2022-06-19:
> I just wonder why the display vendor decided to add these capacitor lines to the flex cable.

It certainly is odd, I think since it's directly from a manufacturer its to give more options in its use, however it makes it more complex.

I have the capacitors coming in the mail, I'll get this wired up on a breadboard as soon as I have them. I'll get some code ready to test in the meantime. To include your recent changes do I need to install the library from Github or will version 2.32.15 in the installation manager have the most recent commit? I tried downloading the zip, extracting, removing the "-master" and placing in my Arduino library folder but I don't see the library showing up under libraries or examples. 

And what syntax would I use for the constructor? Say I was going to try to run the full_buffer graphics test on my ESP32, will it look something like:
`U8G2_SSD1306_ist3088_320X240_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ , /* dc=*/ , /* reset=*/ );
`

olikraus, 2022-06-19:
> And what syntax would I use for the constructor? 

There is still more todo on my end, but yes the constructor would somehow look like this (with all uppercase chars).
Additionally you need to enable u8g2 16 bit mode (see here: https://github.com/olikraus/u8g2/blob/master/doc/faq.txt)

>  I'll get some code ready to test in the meantime.

Is there any init sequence available for your display? As mentioned above, I have some doubts here...

Once the constructor is ready, my standard beta tester instructions will be valid:

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

millerp6ATWIT, 2022-06-19:
> Is there any init sequence available for your display? As mentioned above, I have some doubts here...

I reached out to the manufacturer to ask if they had an example of how to initiate the IST3088 driver and I will let you know how they respond. 

I successfully installed the beta version of the library ( I understand the code is not ready yet, just wanted to make sure I could do it), I will update it again when you're ready on your end. In the meantime I'll wait to hear from the manufacturer on a potential initiation sequence and wire up the display when the capacitors come in. I'll update you when either of those happen.

olikraus, 2022-06-19:
> Additionally you need to enable u8g2 16 bit mode

Just realized, that 16 bit mode is (hopefully) automatically enabled for ESP32.

millerp6ATWIT, 2022-06-19:
Oh nice, I noticed that the ST75256 it was working and I didn't uncomment that line in the library, ( I did add `#define U8G2_16BIT` to the setup for the main ) Not sure if it's necessary to do in the library or not.

olikraus, 2022-06-19:
> Oh nice, I noticed that the ST75256 it was working and I didn't uncomment that line in the library, ( I did add #define U8G2_16BIT to the setup for the main ) Not sure if it's necessary to do in the library or not.

The lib tries to autodetect ESP32...:
https://github.com/olikraus/u8g2/blob/926898ca238fe1b1da73d2e7505035113d51a01f/csrc/u8g2.h#L74-L79

olikraus, 2022-06-19:
I have added all the constructors, for example:
``` C++
U8G2_IST3088_320X240_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8); 
U8G2_IST3088_320X240_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
```

All examples will include the above two constructors.
Note: U8x8 is not supported (due to the architecture of the IST3088)

olikraus, 2022-06-19:
beta release 2.33.5 is now available...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

millerp6ATWIT, 2022-06-19:
Awesome, I'll install it and let you know when I'm ready to test.

olikraus, 2022-06-20:
Looks like a piece of program code. Maybe they can provide the corresponding source code.

millerp6ATWIT, 2022-06-21:
Hmm, I just asked if they had source code to reference I'll let you know.

millerp6ATWIT, 2022-06-23:
Hi Oliver, they did respond with some C code, hopefully this helps some: [link](https://ancoretraining-my.sharepoint.com/:u:/p/paul_miller/EaegGK1CMlNAnL-zyvnAdz0BA2KLxHgfW0IW2hzPhb29FQ?e=gRef0h).

On the testing note I received the capacitors today so I will begin wiring the display for testing.

olikraus, 2022-06-23:
Super, this helps
``` C++
void MainAddset() { 
    SdCmd(0x0d);                
	SdData(0x13); SdData(0x00);
    SdCmd(0x0e);                
	SdData(0xef); SdData(0x00);
    SdCmd(0x08);                
	SdData(0x00); SdData(0x00);
	SdCmd(0x09); 
}
void stanby() {
	 SdCmd(0x03);                
	 SdData(0x00); SdData(0x01);
}
void exit_stanby() {
	 SdCmd(0x03);                
	 SdData(0x00); SdData(0x10);
}
void initLCDM(void) {
    SdCmd(0x01);                // driver control
	SdData(0x00); SdData(0x00); // c1-->c240 s1-->s240 duty=1/240
    SdCmd(0x02);                // driver control
	SdData(0x01); SdData(0x4c);	//0x15
    SdCmd(0x03);                // power control
	SdData(0x00); SdData(0x70); // VC VR Vf=1
    SdCmd(0x04);                // power control
	SdData(0x04); SdData(0x61); //0x61  //1/14b
    SdCmd(0x05);                // CT control
	SdData(0x00); SdData(0x6a); // 0X76 
    SdCmd(0x06);                // CT control
	SdData(0x00); SdData(0x03); 
    SdCmd(0x23);                // B/W MODE,16G NEED REMOVE
	SdData(0x00); SdData(0x04); //04:mono
    SdCmd(0x28);                // OSC
	SdData(0x00); SdData(0x0c);
    SdCmd(0x37);                
        SdData(0x00); SdData(0x12);
    SdCmd(0x07);                
        SdData(0x00); SdData(0x01);
 }
```

millerp6ATWIT, 2022-06-23:
Cool, excited to test I'll let you know when I have something.

olikraus, 2022-06-23:
I have not yet updated the code with the above vendor information... I will create another beta release

olikraus, 2022-06-24:
I took over the init sequence from above. There is one strange issue: The datasheet you provided requests bias as 1/16, however the above sequence configures bias as 1/13:
```
    SdCmd(0x04);                // power control
	SdData(0x04); SdData(0x61); //0x61  //1/14b
```
The second byte 0x04 selects bias as 1/13. The comment seems to say bias 14 (//1/14b) and the actual request in the datasheet is 1/16 . Well, I tookover the 1/13 value.

olikraus, 2022-06-24:
beta release 2.33.6 is available... 

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

millerp6ATWIT, 2022-06-26:
Hey oliver sorry for the delay, I have the screen wired up exactly like the schematic and I tried running the graphics test full buffer example. Its displaying something, just not something readable, [here's](https://photos.app.goo.gl/HoxwGyYu5NUmiW567) a video of the output and some pictures of the wiring. I tried to be as neat with it as I could.

millerp6ATWIT, 2022-06-26:
Also [here](https://photos.app.goo.gl/qJuqfC4JBet5GrFN7) is the hello world example, maybe that will offer more insight.

olikraus, 2022-06-26:
The hello world example is a good picture. I think I need to change two topics...

olikraus, 2022-06-26:
I have created beta release 2.33.8. Can you create a picture of the hello world example again with the latest beta?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

millerp6ATWIT, 2022-06-26:
Wow, that looks great now. [Heres](https://photos.app.goo.gl/husBPd4FfaxPVb4k6) the graphics test running beautifully. Is there anything else you can think of we should test?

olikraus, 2022-06-26:
oh, i am impressed. I didn't expect such a progress right now.
Yes please do a test with the FlipMode example from the "page mode" section. It will draw a frame at the border of the display. The frame should be there all the time. There should be no missing section of the frame and there should be no pixel between the frame and the outer border of the display.

olikraus, 2022-06-26:
BTW: Nice display with excellent contrast.

millerp6ATWIT, 2022-06-26:
Hmm interesting, it does seem like the border opposite the flex cable is [flickering](https://photos.app.goo.gl/PDRHqJMDt1RrLhcX9) on and off somewhat. 

Yes I'm quite pleased with this display! I'm glad the progress has been so quick. I was worried the wiring would need to be revised but it seems the driving voltage circuit is working well.

olikraus, 2022-06-26:
Thanks for all the nice videos: I also see the flickering in the video. I personally would say, that the u8g2 lib works correctly and that this flickering is somehow caused by hardware. Maybe wires are too long between ceramic caps and the flex cable connector. I personally also do not trust those proto boards. Maybe also the power supply is not stable enough. I could imagine that the flickering is gone once everything is soldered.

Ok, I think it looks good from my perspective. Note, that u8x8 will not work with your display. 
Thanks for the cooperation, testing and videos :-)

millerp6ATWIT, 2022-06-26:
Great yeah its quite a finicky wiring setup, it is likely hardware related. 

Thank you for all your help, you've been very knowledgeable and responsive to work with. Let me know if there's anywhere I can throw something your way for all your work.

closed 2022-06-26 by millerp6ATWIT

olikraus, 2022-06-26:
Bugfix for setContrast will be included in next release...