olikraus/u8g2 #1888: Blank screen with Teensy 3.2 / Adafruit SSD1306?

bimac, 2022-06-18:
Hi,

I'm trying to use u8g2 on a Teensy 3.2 with an Adafruit 128x64 SSD1306 ([this one](https://www.adafruit.com/product/326)). It works fine with the Adafruit library, but for the life of me, I can't get it to work with u8g2. Here's my code:

```
#include <Arduino.h>
#include <SPI.h>
#include <Wire.h>
#include <U8g2lib.h>

U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);
// U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0, 19, 18, U8X8_PIN_NONE);

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB08_tr);
    u8g2.drawStr(0,20,"Hello World!");
  } while ( u8g2.nextPage() );
}
```

Is there anything that comes to mind?

Thanks

bimac, 2022-06-18:
Uhm, nevermind - I switched the display to SPI mode and things work fine now :-)

Have a nice weekend!

closed 2022-06-18 by bimac

olikraus, 2022-06-18:
Great 👍