olikraus/u8g2 #1893: Possible documentation error in "Porting to new MCU platform"

spi5498, 2022-06-23:
I'm currently attempting to port the library to a new MCU and is using the great documentation for this.

However, it seems that there might be a "demonstration error" in the i2c byte callback procedure:

```
case U8X8_MSG_BYTE_END_TRANSFER:
      i2c_transfer(u8x8_GetI2CAddress(u8x8) >> 1, buf_idx, buffer);
      break;
```

I think the above "demo" of function "i2c_transfer" should be in the case "U8X8_MSG_BYTE_SEND:" instead?

Currently this case is just demoed as:
```

    case U8X8_MSG_BYTE_SEND:
      data = (uint8_t *)arg_ptr;      
      while( arg_int > 0 )
      {
	buffer[buf_idx++] = *data;
	data++;
	arg_int--;
      }
```

Maybe I am just interpreting it wrong?

Kind regards,
Daniel

olikraus, 2022-06-23:
U8g2 will always send this sequence:
U8X8_MSG_BYTE_START_TRANSFER
U8X8_MSG_BYTE_SEND
U8X8_MSG_BYTE_SEND
U8X8_MSG_BYTE_SEND
...
U8X8_MSG_BYTE_SEND
U8X8_MSG_BYTE_END_TRANSFER

So you can rely on u8g2 to receive the END Transfer message. Now the idea is, that the data is collected only with Byte Send, but later physically sent with END Transfer.

As a real example you could check the code here: 
https://github.com/olikraus/u8g2/blob/fd2a1edd27205668f92af3b71d7f515811405641/cppsrc/U8x8lib.cpp#L760

spi5498, 2022-06-24:
Hello!

Thanks for your reply.
I think this is just a issue on the documentation part, and not on the library functionality. 

I'll add further samples here:

In the Arduino callback sample posted on the porting docs.
The call is to the "Wire" library, 

And the calls is:

```
    case U8X8_MSG_BYTE_SEND:
      Wire.write((uint8_t *)arg_ptr, (int)arg_int);
      break;
    case U8X8_MSG_BYTE_INIT:
      Wire.begin();
      break;
    case U8X8_MSG_BYTE_SET_DC:
      break;
    case U8X8_MSG_BYTE_START_TRANSFER:
      if ( u8x8->display_info->i2c_bus_clock_100kHz >= 4 )
      {
	Wire.setClock(400000L); 
      }
      Wire.beginTransmission(u8x8_GetI2CAddress(u8x8)>>1);
      break;
    case U8X8_MSG_BYTE_END_TRANSFER:
      Wire.endTransmission();
      break;
    default:
      return 0;
```
In the above sample, we can clearly see where procedures for begin, transfer, and end is supposed to be

Meanwhile the "boilerplate" callback procedure on the porting docs is stating an example of a function called "i2c_transfer" only under the case "U8X8_MSG_BYTE_END_TRANSFER:" while the other cases are empty of any what the cases should execute.

This is no show stopper in any way, I just wanted to chime in that this part of the porting documentation might need some clarification. But maybe I still got it backwards?

olikraus, 2022-06-24:
Hmm... From my perspective there are two different approaches provided by the low level HAL procedures, which are both mentioned in the section after this header: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-i2c-communication

So the idea is: You check your HAL documentation: Based on this documentation you either implement option 1 or I do not know how to express this better (maybe because i am not a native English speaker).

spi5498, 2022-07-18:
Hey!
Thanks for your reply. 

Feel free to close this, it was just something i found while working on my port. 
It was just my thought on how to further elaborate a bit on this specific segment of the documentation.

closed 2022-07-18 by spi5498

olikraus, 2022-07-18:
Ok...