olikraus/u8g2 #1894: number string display small glitch with 0,1,2 and other work well

actg, 2022-06-23:
I'm working with ssd1306 chip and oled 0.96inch(128*64) and newest version u8g2 [2.29.11](https://github.com/olikraus/u8g2/releases/tag/2.29.11) ,but zero,one,and two number display have some residual ghost,but other number work well,the following was review video

[number of string display glitch](https://www.youtube.com/shorts/OA6woTp8SDo)

- some initlize souce code segment as following:
```c
    u8g2_Setup_ssd1306_128x64_noname_f(&u8g2, rotation, u8x8_byte_cb, u8x8_gpio_and_delay);

    u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,
    u8g2_ClearDisplay(&u8g2);
    u8g2_SetPowerSave(&u8g2, 0); // wake up display
```

olikraus, 2022-06-23:
Hmmm... difficult to say without seeing the code for drawing this display.

In general you probably want to use mode 0: https://github.com/olikraus/u8g2/wiki/u8g2reference#setfontmode

actg, 2022-06-24:
The initialize source code extended,I have been test again with diffirent fond(*.bdf file),I can confirm the unifont caused number zero,one,two display small glith,but other display normally,it's so strange, oops... ^_^

```c

/*
  Fontname: -gnu-Unifont-Medium-R-Normal-Sans-16-160-75-75-c-80-iso10646-1
  Copyright: Copyright (C) 1998-2019 Roman Czyborra, Paul Hardy,  Qianqian Fang, Andrew Miller, Johnnie Weaver, David Corbett, et al.  License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>  with the GNU Font Embedding Exception.
  Glyphs: 582/57086
  BBX Build Mode: 0
*/
const uint8_t u8g2_font_t12[20488] U8G2_FONT_SECTION("u8g2_font_t12") = 
{
// init byte array too verbose,we ignore at present
}

    u8g2_SetFont(&u8g2, u8g2_font_t12);
    u8g2_SetFontDirection(&u8g2, 0);
    u8g2_SetFontMode(&u8g2,0);
    u8g2_SetFontPosTop(&u8g2);

```

-  decrease veriable from 30 to 0 in a timer callback every 100ms period
```c
char buffer[32]={0};
static int value=30;

u8g2_t *u8g2_instance_get(void)
{
    return &u8g2;
}

/**
 * C++ version 0.4 char* style "itoa":
 * Written by Lukás Chmela
 * Released under GPLv3.
 */
static char* custom_itoa(int value,char* result, int base) {
    // check that the base if valid
    if (base < 2 || base > 36) { *result = '\0'; return result; }

    char* ptr = result, *ptr1 = result, tmp_char;
    int tmp_value;

    do {
        tmp_value = value;
        value /= base;
        *ptr++ = "zyxwvutsrqponmlkjihgfedcba9876543210123456789abcdefghijklmnopqrstuvwxyz" [35 + (tmp_value - value * base)];
    } while ( value );

    // Apply negative sign
    if (tmp_value < 0) *ptr++ = '-';
    *ptr-- = '\0';
    while(ptr1 < ptr) {
        tmp_char = *ptr;
        *ptr--= *ptr1;
        *ptr1++ = tmp_char;
    }
    return result;
}

if(value > 0)
{
    value--;

    if(value == 0)
       value=30;
}
custom_itoa(value,buffer,10);

    u8g2_DrawStr(u8g2_instance_get(), 80, 38, buffer);
    u8g2_SendBuffer(u8g2_instance_get());

```

olikraus, 2022-06-24:
Maybe you missed to clear the buffer before drawing something to it.

actg, 2022-06-25:
I have found the issue,If used tools bdfconv.exe,we should give -b parameter to 1,not 0,common height work well.

closed 2022-06-25 by actg