olikraus/u8g2 #1897: TK078F288 and HJR_OEL1M0201 OLEDs

olikraus, 2022-06-25:
@Tiky169
I create an issue here for your Pull Request https://github.com/olikraus/u8g2/pull/1896

uint8_t u8x8_d_sh1107_HJR_OEL1M0201_96x96(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);
uint8_t u8x8_d_sh1107_TK078F288_80x128(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);

olikraus, 2022-06-25:
```
U8G2_SH1107_HJR_OEL1M0201_96X96_F_4W_HW_SPI(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
U8G2_SH1107_TK078F288_80X128_F_4W_HW_SPI(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
```

olikraus, 2022-06-25:
I have created beta 2.33.7

Will it work? Can you test this release?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

Tiky169, 2022-06-26:
I am very happy to tell you that this beta 2.33.7 is running very well. Both modules have been tested with a few examples and both are working now. Thank you very much!

olikraus, 2022-06-26:
Super, thanks for the feedback.

closed 2022-06-26 by olikraus