olikraus/u8g2 #1900: ERC12864FS-1-S7

martinbu, 2022-06-28:
Hi,

the Display ERC12864FS-1 got an update due to global chip shortage and is now named ERC12864FS-1-S7. In the update information [0] they write it should be compatible, except the pin assignment. But as it uses a different IC (ST7567S instead of ST7565R) and ST7567S is listed seperately in the u8g2 source code I want to ask if a new constructor with a different configuration is needed for it?

[0] https://www.buydisplay.com/1-4-inch-graphic-128x64-lcd-module-serial-spi-st7567s-black-on-white#product.info.description

olikraus, 2022-06-28:
Can you test some of the ST7567S constructors? will they work? Which constructor has best results?

martinbu, 2022-06-30:
None of the ST7567S constructors worked for me, but the issue was a different one. I have set a fixed spi speed of 48MHz in my spi driver. This works fine for the display with the st7565R but not for st7567S. After reducing the spi speed to 42MHz everything works fine and the "old" constructor (u8g2_Setup_st7565_erc12864_alt_f) works absolutly fine for me now.

But another question is, does the st7567S has different functionalities that should be configured for some reason?

olikraus, 2022-06-30:
The differences are indeed minimal: https://github.com/olikraus/u8g2/blob/master/doc/controller_cmds.txt
The features/commands where the ST7565 and ST7567 differ, are usually not used by u8g2: The boost level I do not remember, but of the contrast is fine for you, then the boost level cmd is anyways not required.

not-fl3, 2023-02-08:
@martinbu sorry for bumping an old thread, but I am desperate :) 

I have the same ERC12864FS-1-S7 display (https://www.buydisplay.com/1-4-inch-graphic-128x64-lcd-module-serial-spi-st7567s-black-on-white)

connected like this: 
[image]

with the following code: 
```
U8G2_ST7565_ERC12864_ALT_F_4W_HW_SPI u8g2(U8G2_R0, 10, 8, 9);

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(0,24,"Hello World!");
  } while ( u8g2.nextPage() );
}
```

When I plug in the arduino, for a split second it shows correct text on the screen, but than it goes all black :(

I found a (sort of) relevant problem: https://forum.arduino.cc/t/128x64lcd-with-nt7534-control-lsi/131714/7, but it was completely different display.

I really am desperate, I have zero clues how to proceed and what else to try, I've tried all the different u8g2 constructors, different contrast values, I've runned (ported to arduino) buydisplay's reference code - its always the same, split second of an image and than it immediately goes black :( 

So would be super happy for any advice :pray:

olikraus, 2023-02-19:
Sounds like a power supply problem. Did you connect backlight to the arduiono? It so, just disconnect backlight and try your sketch again. Usually text should be readable without backlight. If all is ok, then backlight draws too much of current. In such a case, use an external power supply or add a resistor into the backlight power supply (remember to use a high current resistor otherwise it might become to hot and might burn).

not-fl3, 2023-02-19:
> Sounds like a power supply problem. Did you connect backlight to the arduiono? It so, just disconnect backlight and try your sketch again. Usually text should be readable without backlight. If all [...]

Thanks for reply!! Actually, yes, backlight fixed the problem - turned out pixels are not just black, but super high contrast. After further invesatigation I found wiring problem around V0/XV0 pins. Sorry for your time, it was totally my bad!

olikraus, 2023-02-20:
Happy to see that your problem is fixed