olikraus/u8g2 #1901: wrong spi_mode for sh1106_128x64_noname?

bimac, 2022-06-29:
Hey there,

I just switched from an SSD1306 to an SH1106 using HW SPI on a Teensy 3.2 with several other SPI slaves on the bus along with the display. The SSD1306 worked fine (a change was necessary for other reasons). After swapping in the SH1106 and changing the constructor respectively I am getting artifacts when the other slaves are active. Also, the picture is flipped by 180 degrees.

```
//U8G2_SSD1306_128X64_NONAME_1_4W_HW_SPI u8g2(U8G2_R0, OLED_CS, OLED_DC, OLED_RST);
U8G2_SH1106_128X64_NONAME_1_4W_HW_SPI u8g2(U8G2_R2, OLED_CS, OLED_DC, OLED_RST);
```

When using the constructor for SSD1306 with the SH1106 hardware installed it works fine (except for the 2px offset). It also works fine with the SH1106 constructor after modifying the library to use spi_mode 0 instead of spi_mode 3.

Does u8g2 use an incorrect spi_mode for SH1106 or could this rather be an issue on my end? Is there a way to change spi_mode without hacking it into the library? I found #1695 which seems to be related - but I can't figure out how to use it.

Thanks!

olikraus, 2022-07-01:
Thanks for that hint. I guess I need to look into it more deeper.

bimac, 2022-07-01:
Thanks! Let me know if there's anything I can do on my end ...

olikraus, 2022-07-01:
Actually I don't know how to continue here. Probably i don't own a sh1106 display with spi to reproduce the issue.

bimac, 2022-07-01:
I could send you one if that would help?

olikraus, 2022-07-01:
I will search in my lab first 😃

olikraus, 2022-07-03:
I still haven't search in my lab, but I have a request

> Let me know if there's anything I can do on my end ...

Do you own some other Arduino-compatible board? For example Arduino Uno?
Will the problem also appear with an Arduino Uno or any other none-Teensy 3.2 Board?

bimac, 2022-07-03:
I'll have to search _my_ lab tomorrow ;-) I should have a Leonardo and a Due somewhere ... I'll try with a MWE and report back!

bimac, 2022-07-04:
OK, I found the Due. Here's the MWE:

```
#include <U8g2lib.h>
#include <SPI.h>

U8G2_SH1106_128X64_NONAME_1_4W_SW_SPI u8g2(U8G2_R0, 2, 3, 4, 5, 6);
//U8G2_SH1106_128X64_NONAME_1_4W_HW_SPI u8g2(U8G2_R0, 4, 5, 6);

void setup() {
  u8g2.begin();
}

void loop() {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(0,24,"Hello World!");
  } while ( u8g2.nextPage() );
}

```
I can confirm: With HW SPI the text is flipped upside-down. Changing https://github.com/olikraus/u8g2/blob/3d895d12a07cb68d2fba8c52ecd6cd6c1f0dbb62/csrc/u8x8_d_ssd1306_128x64_noname.c#L467 from 3 to 0 resolves the issue. With SW SPI, the text is upright in either case.

olikraus, 2022-07-04:
Wow super. So I should really change the mode...

bimac, 2022-07-04:
Well, at least that does the trick for the two displays on my desk. Its weird though, because this whole thing came up while working on a fork of [fwachsmuth/Synkino](https://github.com/fwachsmuth/Synkino). He is also using an [SH1106 with HW SPI](https://github.com/fwachsmuth/Synkino/blob/aaa44afb5fbb15bb562e0726b7246e61c530c88c/Synkino_Frontend/Synkino_Frontend.ino#L209) but in combination with a different MCU (ATMEGA328P). I actually received my SH1106s from @fwachsmuth - so they should be the same exact displays he is successfully using in his project.

olikraus, 2022-07-04:
Your feedback at least shows, that the issue is probably not related to the hw spi implementation for a specific board

bimac, 2022-07-04:
Yes, well, that's true :)

fwachsmuth, 2022-07-04:
@bimac I sourced my ~100 modules from a variety of sellers on aliexpress, it is well possible that you got one that behaves (or is) different than mine. I also have some that say "white" but are actually blue... you get what you pay for. :)

bimac, 2022-07-05:
Haha :) Well, for reference, these are the two modules I have here - they both behave the same way:

[image]
[image]

olikraus, 2022-09-03:
Not sure why I initially assigned mode 3. Looking at the datasheet doesn't show any difference in the SPI timing diagramms.
I will assign mode 0 to sh1106.

closed 2022-09-03 by olikraus

bimac, 2022-09-08:
Sorry, only see your reply now. Thanks for looking into this & the commit, @olikraus!

olikraus, 2022-09-08:
Its already available in Arduino IDE.