olikraus/u8g2 #1902: Merging the fonts

I-AM-Tushar-chauhan, 2022-06-29:
Hello, olikraus,
Now I know how to generate my font bdf file. but it has a block system, I mean if I merge two Unicode fonts so it will cut out the first Unicode fonts.

Do you know any function that merges these two fonts without cutting out the first Unicode fonts?
The Unicode 0x0930 use in three different ways for the Hindi language with three different structures of fonts, examples are the following:-
1)पर्व
2)पेट्रोल
3)आर
these three examples create a problem for me with how I write these three different structure in a single bdf file with different access.
 /*STARTCHAR U+0930   // this is the example of 2.
ENCODING 2352
SWIDTH 1000 0
DWIDTH  11  0
BBX 5 5 -11 -7
BITMAP 
18
38
70
E0
C0
ENDCHAR*/

Because I want to run it on P10 tiles.

Do you know the significance of U8G2.setfontmode??

olikraus, 2022-06-29:
hmm I don't understand your problem that good, but regarding the merging:
You need to copy the content between STARTCHAR and ENDCHAR into a single font.
I mean, you use a copy of one font. Then insert the desired content between STARTCHAR and ENDCHAR from the second font into first font.

I-AM-Tushar-chauhan, 2022-06-29:
"You need to copy the content between STARTCHAR and ENDCHAR into a single font."
but it's not possible, because if we use 0x094D with 0x0930 then it's meaning will change according to Hindi language.
we have 3 possible fonts of 0x0930 unicode in hindi language:
)र्
2)ट्
3)र
that is the reason why I need to merge and define different fonts in same file.
do you have any suggestion about it??

olikraus, 2022-06-29:
The construction of a new glyph, based on two specific characters is not possible with .bdf and u8g2.
See also here: https://github.com/olikraus/u8g2/issues/1015

closed 2023-02-22 by olikraus