olikraus/u8g2 #1908: U8G2_SED1330_240X128_1_8080 u8g2 only using 240 out of 256 pixel?

sm5oem, 2022-07-12:
Hi is there any way to get all pixels working on the  U8G2_SED1330_240X128_1_8080 u8g2, To 256X128 ?
It's missing the last 16 on the right side of the display now.
Thanks,
Ronny

olikraus, 2022-07-12:
We probably need to add a 256x128 constructor. Can you attach the display datasheet or any link to the display description?

sm5oem, 2022-07-13:
HiThanks for your reply.this is what I have found so far.Richard Teel / LCD-101 · GitLab
https://www.lcd-module.de/eng/pdf/zubehoer/sed1330.pdf
I hope this is some thing you can use.Thanks for a great job on all the displays you are working on.  Thanks,Ronny Larsson

 
 

  
  /************ INTERFACE PIN ASSIGNMENT NANO ************
#define d0 6      //  9 FC 16p    d0             10 pin on dispay board
#define d1 7      // 10 FC 16p    d1             11 pin on dispay board
#define d2 8      // 11 FC 16p    d2             12 pin on dispay board
#define d3 9      // 12 FC 16p    d3             13 pin on dispay board
#define d4 10     // 13 FC 16p    d4             14 pin on dispay board
#define d5 11     // 14 FC 16p    d5             15 pin on dispay board
#define d6 12     // 15 FC 16p    d6             16 pin on dispay board
#define d7 13     // 16 FC 16p    d7             17 pin on dispay board
#define cs 2      //  8 FC 16p    CS              8 pin on dispay board orange
#define reset 5   //  5 FC 16p    RESET           5 pin on dispay board green
#define enable 3  //  7 FC 16p    WR = enable     7 pin on dispay board purple
#define dc 4      //  6 FC 16p    dc JUMPED TO A0 9 pin on dispay board yellow
                                  RD = HIGH + 5v  6 pin on dispay board
                  //  1 FC 16p    FG / GND      1&2 pin on dispay board
                  //  2 FC 16p    RESERV not used
                  //  3 FC 16p    + 5v            3 pin on dispay board
                  //  4 FC 16p    -10,2 VOLT      4 pin on dispay board

U8G2_SED1330_240X128_1_8080 u8g2(U8G2_R0, 6, 7, 8, 9, 10, 11, 12, 13, /*enable=*/ 3, /*cs=*/ 2, /*dc=*/ 4, /*reset=*/ 5); // Connect RD with +5V, FG with GND  NANO / UNO /MEGA
void setup(void) {
u8g2.begin();
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_6x10_tr); 
     
     u8g2.drawStr(40,64,"240x128_bollar4_SAT_NANO_GPS.ino");
  } while ( u8g2.nextPage() );
  delay(5000); 
gpsPort.begin(9600);  
}

/* draw something on the display with the `firstPage()`/`nextPage()` loop*/

void loop(void) {
   while (gps.available( gpsPort )) {
  fix = gps.read();      
  fixCount++;
  if (fix.status >= gps_fix::STATUS_STD){   
  } 
   
  u8g2.firstPage();
  do {
  u8g2.setFont(u8g2_font_6x10_tr);
  u8g2.drawStr(10, 10, "Sat Count:");
  u8g2.drawStr(60, 70, "W");
  u8g2.drawStr(175, 70, "E");
  u8g2.drawStr(117, 10, "N");//34,13
  u8g2.drawStr(117, 124, "S");
  u8g2.drawPixel(120, 64);
  u8g2.drawCircle(120, 64, 16);
  u8g2.drawCircle(120, 64, 32);
  u8g2.drawCircle(120, 64, 48);

  
for (int k = 0; k < MAX_SATELLITES; ++k) 
{
if (gps.satellites[k].snr > 5) // 
{
az = ( gps.satellites[k].azimuth );
el = (gps.satellites[k].elevation );
az_r = az * rad_fac;
 e = 48* ((90 - el) / 90); //  
 x = 120 + (sin(az_r) * e);//
 y = 64 - (cos(az_r) * e);// 
 u8g2.drawCircle(x, y, 2); 
} 
}
u8g2.setCursor(70, 10);
if ( gps.sat_count < 10 ) 
u8g2.print( ' ' ); 
u8g2.print( gps.sat_count );

u8g2.setCursor(160, 10);
  u8g2.print("UTC:");
  if (fix.dateTime.hours < 10) {
  u8g2.print("0");
  }
  u8g2.print(fix.dateTime.hours);
  u8g2.print(":");
  if (fix.dateTime.minutes < 10) {
  u8g2.print("0");
  }
  u8g2.print(fix.dateTime.minutes);
  u8g2.print(":");
  if ( fix.dateTime.seconds < 10 ) {
  u8g2.print("0");
  }
  u8g2.print( fix.dateTime.seconds );  
 
} while ( u8g2.nextPage() );
}
}

olikraus, 2023-02-22:
So you have used U8G2_SED1330_240X128 here, right?
But what is your actual display then? I mean the above linked datasheed is the controller datasheet, not the datasheet of the display itself.

sm5oem, 2023-02-23:
Hi the diplay is a Hyundai HG25604-B

Thanks,Ronny Larsson

 
 

  

|  | Virus-free.www.avg.com |

sm5oem, 2023-02-23:
MVH, Thanks,Ronny Larsson

 

 

 
 

  

|  | Virus-free.www.avg.com |

olikraus, 2023-02-23:
Is this the display? https://www.allelectronics.com/mas_assets/media/allelectronics2018/spec/LCD-101.pdf

olikraus, 2023-02-23:
I have create a new beta release 2.34.15, which includes the constructor U8G2_SED1330_256X128_1_8080 (see examples)
I know, it is long time ago, but are you able to test this constructor somehow?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

sm5oem, 2023-02-23:
HiYes that’s the one. Thanks Sent from my iPhone

olikraus, 2023-02-23:
Wow, thanks for the quick response

sm5oem, 2023-02-23:
Your welcome I am in Arizona usThanks Sent from my iPhone

olikraus, 2023-06-12:
hmmm... closing

closed 2023-06-12 by olikraus