olikraus/u8g2 #1909: No image on "HG-9616W-0.69-8PIN" (SSD1306 96X16 oled)

mahmeh, 2022-07-12:
Hi, i have been struggling to get this display module to work, i always turn to u8g2 when trying a new mono screen but i cant seem to get anything on this display, im in contact with the manufacturer, but in the mean time i wanted to check if this display is compatible with the SSD1306 96X16_ER Constructor

Data sheet: 
[HG-9616W-0.69-8PIN.pdf](https://github.com/olikraus/u8g2/files/9093269/HG-9616W-0.69-8PIN.pdf)
Another data sheet from another supplier on Aliexpress, i think its the same
[HNX-069WH8P01.pdf](https://github.com/olikraus/u8g2/files/9093376/HNX-069WH8P01.pdf)

olikraus, 2022-07-12:
Did you provide the 7v at pin 7?

mahmeh, 2022-07-12:
I have tried, but the manufacturer is unclear if that is needed or not, the
datasheet says a cap should be connected to pin 7 if the internal regulator
is used, but no value is specified and the manufacturer have not got back
to me on that.

I'm gonna do some more tests with 7v applied, but i would prefer to use the
internal regulator if possible

olikraus, 2022-07-12:
Have a look at this display: https://www.buydisplay.com/graphic-i2c-0-69-inch-96x16-oled-display-mp3-module-white-on-black

The corresponding datasheet (https://www.buydisplay.com/download/manual/ER-OLED0.69-1_Series_Datasheet.pdf) clearly states that you need to connect two caps to four dedicated pins. Those pins are not their with your display, so I assume you need an external 7V power supply. 

Additionally u8g2 needs to be modified, because u8g2 will always ativate the internal charge pump.

mahmeh, 2022-07-14:
Thanks, disabling the built in regulator was what i was missing i think, i
manged to get image on the display now! Some weird artefacts but might be
my sketchy test setup

Digim00n, 2022-07-30:
Working flawlessly when controlling the reset pin with the mcu, instead of just pulling it high.
Thanks for helping me solve this

mahmeh, 2022-07-30:
Working flawlessly when controlling the reset pin with the mcu, instead of
just pulling it high.
Thanks for helping me solve this

closed 2023-02-22 by olikraus