olikraus/u8g2 #1910: Board with atmega328pb, pin d12(miso) is piloted by the U8G2 library

tistructor, 2022-07-14:
I made a custom board with atmega328pb and tried the graphics test sample file and it works properly.
I noticed though that pin 12 (MISO) it is controlled by the U8G2 library because in my card I had used this pin to control a buzzer but it does not work properly indeed only by enabling the pin as output with the oscilloscope I detect impulses  of less than 1V without me pilots this pin in any way.
I tried not to use the U8G2 library and pin d12 works properly.
Consequently the library enables the 12 MISO pin I believe with the SPI function even if the library does not need it.

[image]

olikraus, 2022-07-14:
Did you use HW SPI? If so, I assume there is an issue with either Arduino SPI implementation or Atmega328 controller.
With SW SPI do you observer the same problem?

tistructor, 2022-07-14:
I am using HW spi.
I am reading the atmega328pb datasheet and in fact once the MISO pin is enabled if it is not used it becomes an input and cannot be used.
 I didn't know it was so. I believed that from some registers it was possible to disable the piloting of this pin by the SPI interface
In other projects I think I was lucky not to use pin 12.

Now I try SPI software interface.

tistructor, 2022-07-14:
SPI software mode works properly and my pin that controls the buzzer too.

I have to redo the Display board Azzzzzzz

Thanks

closed 2023-02-22 by olikraus