olikraus/u8g2 #1913: Library update for LC7981, 128x128 display

autocrat22, 2022-07-16:
The monochrome display [datasheet](https://www.tme.eu/Document/f3f2f61fd63368304bbb4b73d359cd54/RG128128B-BIW-V.pdf) for **RG128128B**.

How should [u8x8_d_lc7981.c](https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_lc7981.c) modified to support it?

olikraus, 2022-07-18:
I have create a new constructor U8G2_LC7981_128X128_1_6800 and U8G2_LC7981_128X128_F_6800
Note that DC = RS !!!
Can you test the FlipMode example (page buffer method): The outer frame at the border of the display should stay and must not change.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

olikraus, 2022-08-13:
no feedback...
closing...

closed 2022-08-13 by olikraus