olikraus/u8g2 #1914: Request for display support

marsjupiter1, 2022-07-18:
I am having no luck with:
https://ae01.alicdn.com/kf/S67c792a61722402eab70fc7d67f73185f.jpg

which says it has an 11C interface on a sT7567S

Have you any pointers?

TIA

Martin Walker

olikraus, 2022-07-18:
Which constructor did you use? Is there any picture of the backside of the display?

marsjupiter1, 2022-07-19:
I tried a lot of different constructors.
the back of the display is in this image
https://ae01.alicdn.com/kf/S31d4a2d1bfa84bf38e20306ab0a32c46r.jpg

regards

Martin

olikraus, 2022-07-19:
Please use a SW_I2C constructor. Will it work? There should be a contrast test example. Will this example show something?

marsjupiter1, 2022-07-19:
thanks for the reply. I have tried:
//U8X8_ST7567_JLX12864_HW_I2C u8x8(21,22);
//U8X8_ST7567_JLX12864_HW_I2C u8x8(U8X8_PIN_NONE);
//U8X8_ST7567_ENH_DG128064_SW_I2C u8x8(21,22);
//U8X8_ST7567_ENH_DG128064I_SW_I2C u8x8(21,22);
//U8X8_ST7567_OS12864_SW_I2C u8x8(21,22);
//U8X8_ST7567_OS12864_2ND_HW_I2C u8x8(U8X8_PIN_NONE);
//U8X8_ST7567_ENH_DG128064I_2ND_HW_I2C u8x8(U8X8_PIN_NONE);
//U8X8_ST7567_ENH_DG128064I_HW_I2C u8x8(U8X8_PIN_NONE);
U8X8_ST7567_JLX12864_2ND_HW_I2C u8x8(U8X8_PIN_NONE);

on a couple of displays using the hello world example. But nothing so much
as flickers.
I do note a 12c scan gives a device id of 0x3F. e.g. it does find the
display. I can't find any reference to an id of 0x3F though.

thanks again

Martin Walker

marsjupiter1, 2022-07-19:
> If you assign the i2c address manually, you need to multiply with 2 for
> u8g2 (see the manual)

Thank you for your response.

That is indeed the trick.
  u8x8.setI2CAddress(0x3F*2);
has brought things to life.
If you don't mind me saying though. Isn't it a little bit buried. Without
your help I might have not found it.
Even the fact that these devices work off an address you can scan for and
plainly the default address is not always right is a thing I didn't know.
I read the faq and found no reference to this.

thanks again

Martin Walker

olikraus, 2022-07-19:
https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress

* Arguments:
   * u8g2: A pointer to the u8g2 structure.
   * adr: I2C address multiplied with 2 (the lowest bit must be zero)

The topic has been subject of many discussions here in this project
For example here: https://github.com/olikraus/u8g2/issues/821
Especially see the picture where the display has the "so called" incorrect" address on it.

closed 2023-02-22 by olikraus