olikraus/u8g2 #1915: SSD1306 Some I2C packages is getting nacked

spi5498, 2022-07-18:
Hello!

So, I'm continuing to work on my port of the library to a new MCU.

Currently working on the HW I2C implementation and using a simple SSD1306 128x64 OLED as a "test" unit.

Now, I'm struggling to get any output on the display. So I have hooked up the scope with I2C protocol-analyzer running.
And, to start with I only run the u8g2_InitDisplay;

I can confirm that all the 25packets/commands transmitted from the MCU to the OLED.
However, I can see that a few of the packets gets "Data ACK 0" (nacked) on them, and it seems to be consistently the same packets.

I'm not sure if this is expected or not, if there is some timing issue causing it, or if it just the scope making it up.

I will attach the trace below from in open-office format (ODS). Hopefully you can open it.

[oledtrace.ods](https://github.com/olikraus/u8g2/files/9133716/oledtrace.ods)

olikraus, 2022-07-18:
Can you also post your glue and Hal code?

spi5498, 2022-07-18:
Sure!

Now, this is a work in progress, so the I2C initialization is handled outside of the callback procedure. 
```

uint8_t u8x8_byte_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)

{

	static uint8_t buffer[32];
	static uint8_t buf_idx;
	uint8_t *data;

	switch(msg)
  		{
    			case U8X8_MSG_BYTE_SEND:
      			
			i2c_write_blocking(i2c_default, 60, (uint8_t *)arg_ptr, (int)arg_int, false);

					      				      
      			break;
    
    			case U8X8_MSG_BYTE_INIT:
      			/* add your custom code to init i2c subsystem */
      			break;
    
    			case U8X8_MSG_BYTE_SET_DC:
      			/* ignored for i2c */
      			break;
    
    			case U8X8_MSG_BYTE_START_TRANSFER:
      			
      			break;
    
    			case U8X8_MSG_BYTE_END_TRANSFER:
      			
			break;
    
    			default:
      				return 0;
  		}
  return 1;

}
```

```
uint8_t u8x8_gpio_and_delay_rp2040(u8x8_t *u8x8, uint8_t msg,uint8_t arg_int,void *arg_ptr)
{

switch(msg)
  {
    case U8X8_MSG_GPIO_AND_DELAY_INIT:	// called once during init phase of u8g2/u8x8
      break;							// can be used to setup pins
    case U8X8_MSG_DELAY_NANO:			// delay arg_int * 1 nano second
      sleep_us(1);
      break;    
    case U8X8_MSG_DELAY_100NANO:		// delay arg_int * 100 nano seconds
      sleep_us(100);
      break;
    case U8X8_MSG_DELAY_10MICRO:		// delay arg_int * 10 micro seconds
      sleep_us(arg_int *10);
      break;
    case U8X8_MSG_DELAY_MILLI:			// delay arg_int * 1 milli second
      sleep_ms(arg_int * 1);

      break;
    case U8X8_MSG_DELAY_I2C:				// arg_int is the I2C speed in 100KHz, e.g. 4 = 400 KHz
      break;							// arg_int=1: delay by 5us, arg_int = 4: delay by 1.25us
    case U8X8_MSG_GPIO_D0:
	gpio_put(8, arg_int);
         //case U8X8_MSG_GPIO_SPI_CLOCK:
      break;
    case U8X8_MSG_GPIO_D1:				// D1 or SPI data pin: Output level in arg_int
      gpio_put(9,arg_int);
    //case U8X8_MSG_GPIO_SPI_DATA:
      break;
    case U8X8_MSG_GPIO_D2:				// D2 pin: Output level in arg_int
      gpio_put(10,arg_int);
      break;
    case U8X8_MSG_GPIO_D3:				// D3 pin: Output level in arg_int
      gpio_put(11,arg_int);
      break;
    case U8X8_MSG_GPIO_D4:				// D4 pin: Output level in arg_int
      gpio_put(4,arg_int);
      break;
    case U8X8_MSG_GPIO_D5:				// D5 pin: Output level in arg_int
      gpio_put(5,arg_int);
      break;
    case U8X8_MSG_GPIO_D6:				// D6 pin: Output level in arg_int
      gpio_put(6,arg_int);
      break;
    case U8X8_MSG_GPIO_D7:				// D7 pin: Output level in arg_int
      gpio_put(7,arg_int);
      break;
    case U8X8_MSG_GPIO_E:				// E/WR pin: Output level in arg_int
      gpio_put(18,arg_int);
      break;
    case U8X8_MSG_GPIO_CS:				// CS (chip select) pin: Output level in arg_int
      gpio_put(14,arg_int);
      break;
    case U8X8_MSG_GPIO_DC:				// DC (data/cmd, A0, register select) pin: Output level in arg_int
      gpio_put(17,arg_int);
      break;
    case U8X8_MSG_GPIO_RESET:			// Reset pin: Output level in arg_int
      gpio_put(22,arg_int);
      break;
    case U8X8_MSG_GPIO_CS1:				// CS1 (chip select) pin: Output level in arg_int
      gpio_put(15,arg_int);
      break;
    case U8X8_MSG_GPIO_CS2:				// CS2 (chip select) pin: Output level in arg_int
      //gpio_put(1,arg_int);
      break;
    case U8X8_MSG_GPIO_I2C_CLOCK:		// arg_int=0: Output low at I2C clock pin
      break;							// arg_int=1: Input dir with pullup high for I2C clock pin
    case U8X8_MSG_GPIO_I2C_DATA:			// arg_int=0: Output low at I2C data pin
      break;							// arg_int=1: Input dir with pullup high for I2C data pin
    case U8X8_MSG_GPIO_MENU_SELECT:
      u8x8_SetGPIOResult(u8x8, /* get menu select pin state */ 0);
      break;
    case U8X8_MSG_GPIO_MENU_NEXT:
      u8x8_SetGPIOResult(u8x8, /* get menu next pin state */ 0);
      break;
    case U8X8_MSG_GPIO_MENU_PREV:
      u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ 0);
      break;
    case U8X8_MSG_GPIO_MENU_HOME:
      u8x8_SetGPIOResult(u8x8, /* get menu home pin state */ 0);
      break;
    default:
      u8x8_SetGPIOResult(u8x8, 1);			// default return value
      break;
  }
  return 1;

}

```

This is the MCU specific I2C init snippet:

```
i2c_init(i2c_default, 100 * 1000);
gpio_set_function(20,GPIO_FUNC_I2C);
gpio_set_function(21,GPIO_FUNC_I2C);
gpio_pull_up(20);
gpio_pull_up(21);
bi_decl(bi_2pins_with_func(20, 21, GPIO_FUNC_I2C));

u8g2_t u8g2;

u8g2_Setup_ssd1306_128x64_noname_f(&u8g2,U8G2_R0,u8x8_byte_i2c, u8x8_gpio_and_delay_rp2040);
u8g2_InitDisplay(&u8g2);
```

The MCU in question is a Raspberry Pi Pico

olikraus, 2022-07-19:
You should cache the bytes and follow the template from the second example in this section: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-i2c-communication

I also suggest to complete the delay code:
```
    case U8X8_MSG_DELAY_NANO:			// delay arg_int * 1 nano second
      sleep_us(arg_int*1);
      break;    
    case U8X8_MSG_DELAY_100NANO:		// delay arg_int * 100 nano seconds
      sleep_us(arg_int*100);
      break;
```
and
```
    case U8X8_MSG_DELAY_I2C:				// arg_int is the I2C speed in 100KHz, e.g. 4 = 400 KHz
      sleep_us(arg_int*5);
      break;	
```

spi5498, 2022-07-20:
Thanks for your input!

The nano delays are still a work in progress.
Not really sure on how to best implement exactly one nano second delay.
The Pi Pico runs at 125Mhz, and the information I have is that one clock cycle will result in 8ns. 
Not really sure when and where the 1nano is used?

The 100 nanoseconds is just an old typo, I will correct this, however would be great to know a little bit more in regards to the 1ns need / precision.

Furthermore, so the U8X8_MSG_DELAY_I2C is still used, even when a custom HW I2C callback is used?
I was thinking that this was only used when the library own software i2c implementation is used?
(Note that the HW i2c is already configured to use the correct speed)
What kind of delay is expected here? 

Also, is there any specific reason why its needed to cache the bytes in this case? 
Since the i2c_write_blocking takes the same arguments as for example the wire library for Arduino, i.e pointer to data and the size.
As we can see in the trace, all the data is transmitted, but some is nacked, and always the same.
Might be as simple as the U8X8_MSG_DELAY_I2C then, if this is used even for HW usage?

olikraus, 2022-07-20:
> Furthermore, so the U8X8_MSG_DELAY_I2C is still used, even when a custom HW I2C callback is used?
> I was thinking that this was only used when the library own software i2c implementation is used?

correct, it is just in case you want to switch to SW_I2C for testing purpose...
Additionally both nano delays are also not used if I remember correctly. You could add a 1us delay if you want to be on the safe side (more delay is always ok...).

> Also, is there any specific reason why its needed to cache the bytes in this case?

Reason 1: Speed

I do not know your HAL function, but I assume it will sent start, then i2c address, then the provided data and finally a stop.
If this data is cached, then start, stop and address are sent only once in a while.

Reason 2: Multi-Byte Instructions to the display are broken

An instruction for the display may include 2 or more bytes. For example the brightness command could be 0x81 0x20.
The code 0x81 is the display command to change the brightness and 0x20 is the corresponding argument.
Unfortunately U8g2 will split this command into two U8X8_MSG_BYTE_SEND messages with one byte each (actually it is even more complicated, but let's keep it simple here).

So with your code the transmission will be:
`start address 0x81 stop start address 0x20 stop`
The problem is, that the display command processor will receive the 0x81 but will miss the argument, because you abort the transmission with the stop command. Later the display command interpreter will receive the 0x20 and will probably react with some random action on it.
Caching will ensure that the commands are not split and it will send the correct and uninterrupted sequence to the display:
`start address 0x81 0x20 stop`

spi5498, 2022-07-22:
Hello!

Well, lets leave the delays for the time being since they don't seem to be of any value to the task at hand.
They need of course to be finished in the port.

In regards to the picos "i2c_write_blocking" function and "start stop" if the value of "true" is passed along with this, it will not issue a stop and the next transfer will begin with a "restart" instead.

I have attempted a first try to buffer the messages as per the porting guide. 
But I'm struggling a bit with it. It might coming down to the other issue i opened for some request in clarification on how the I2C is expected to work for the library.

Since it has a switch case for: start transfer, end transfer, and byte send, it would be great to have a bit further clarification on what the library actually does, and what it expect the HAL to do and when.

Latest revision: 

switch(msg)
  {
    case U8X8_MSG_BYTE_SEND:
      data = (uint8_t *)arg_ptr;      
      
      while( arg_int > 0 )
      {
	buf_idx++;	
	buffer[buf_idx] = *data;
	data++;
	arg_int--;
		
	
      } 
      
	
	break;

    case U8X8_MSG_BYTE_INIT:
      /* add your custom code to init i2c subsystem */
      
      break;
    case U8X8_MSG_BYTE_SET_DC:
      /* ignored for i2c */
      break;
    
    case U8X8_MSG_BYTE_START_TRANSFER:
      buf_idx = 0;
      

      break;
    
    case U8X8_MSG_BYTE_END_TRANSFER:

      	i2c_write_blocking(i2c_default, 60, buffer, buf_idx,false);
      break;
    
    default:
      return 0;
  }
  return 1;

This causes all data (i.e all of the init commands) to be sent in one packet, and subsequently getting all NACKs.

olikraus, 2022-07-22:
> Also, just found out on the scope that the first byte sent is 00, and that the last command is missing. So something is also up with the buffer-loop..

```
	buffer[buf_idx] = *data;
	buf_idx++;	
```

spi5498, 2022-07-22:
Yeah, that was my first thought, so I tested that, and for some extremely strange reason that only me one package containing "AE".

Meanwhile, leaving it as mentioned, it will get me all the bytes, but without the last one, and a leading 00.
If i leave the code exactly as in your sample:

```
while( arg_int > 0 )
      {
	
	buffer[buf_idx++] = *data;
	data++;
	arg_int--;
	} 
```

It also results in only one package with AE being sent.
Which is the first byte in the init sequence.

A real tricky situation this :smile:

spi5498, 2022-07-23:
i think there might be an issue with the value of arg_int when the i2c_byte function is called..

spi5498, 2022-07-23:
I added this output to the terminal for further debugging:

```
 while( (uint8_t)arg_int > 0 )
      {
	buffer[buf_idx] = *data;
	printf("%x",buffer[buf_idx]);
	printf("\n");
	buf_idx++;
	data++;
	arg_int--;
		} 
```

```
ae
d5
80
a8
3f
d3
0
40
8d
14
20
0
a1
c8
da
12
81
cf
d9
f1
db
40
2e
a4
a6
```

So this seems that all init commands are all buffered OK.

When looking at the source file for the display with init, it seems like it should be sent in pairs, as you also stated.

But this causes all messages to be sent in just one packet, and that "feels" a bit strange.

Checking further, the value of arg_int is always just 1 when entering the buffer loop.

I'm guessing for a command with a value, this should be "2".
Causing the loop to run twice, and then the END_TRANSFER submitting a packet containing two values?

If so, this is broken somewhere.

When is "BYTE_END_TRANSFER" called? 

if this is

spi5498, 2022-07-23:
Further debug output:

(This is for the Init sequence + powersave command):

```
Buf length: 25
aed580a83fd30408d14200a1c8da1281cfd9f1db402ea4a6
Buf length: 1
af

```

The above is generated with this code:
```

 case U8X8_MSG_BYTE_END_TRANSFER:
	
	printf("Buf length: ");
	printf("%d\n",buf_idx);

	for (int i = 0; i < buf_idx; i++){

	printf("%x",buffer[i]);
      	//printf("\n");
	}
	printf("\n");
	i2c_write_blocking(i2c_default, 60, buffer, 25, false);
      break;
```

This shows us that we are generating all the init commands in one single packet. 

And then a new packet for the powersave command.

Is this expected behavior?

olikraus, 2022-07-23:
It looks good from my perspective

spi5498, 2022-07-23:
Alright.

But shouldn't it just send 1 or 2 bytes at a time? 
The init procedure is transmitted in only one packet containing 25 bytes?

olikraus, 2022-07-23:
> But shouldn't it just send 1 or 2 bytes at a time?

You will receive single bytes only and you need to combine them (like you did in the last version of your code). 
Multiple bytes are fine. What confuses me, is that the initial 0x00 byte is missing:
https://github.com/olikraus/u8g2/blob/7395c6857f3ade38a636b67365427178ad528dfa/csrc/u8x8_cad.c#L548
so you should see 00 ae d5 80 ... instead of ae d5 80 ...

spi5498, 2022-07-23:
Hm,

That is interesting.. I have just checked the u8x8_d_ssd1306_128x64_noname.c file in the source directory.
In that one the "u8x8_d_ssd1306_128x64_noname_init_seq[]" does not contain the leading 00 you refer to.
The first command is "AE".

But then again, i might be looking in the wrong files :sweat_smile:

olikraus, 2022-07-24:
The 0x00 is part of the ssd1306 protocol and inserted by the above mentioned u8x8_cad.c procedure.

spi5498, 2022-07-24:
Interesting.

Checking my u8x8_cad.c i do have that call under function: uint8_t u8x8_cad_ssd13xx_fast_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)

Can we be certain that this is called when using "u8g2_Setup_ssd1306_128x64_noname_f" i.e?

spi5498, 2022-07-24:
Well, today I have some progress on this.

After changing buf_idx = 0; to buf_idx = 1; in the BYTE_START_TRANSFER.
I kind of suspected that something is up with the while-loop condition with the greater than ">" sign.

The init sequence was actually then successfully sent towards the display, and the display "lights up" with a gargbled image.
The image is just like an old TV signal. black/white dots in a mess.

However, trying further to send something to display. like:
```

while(1){

	u8g2_FirstPage(&u8g2);
	do
	{
	u8g2_SetFont(&u8g2,u8g2_font_tenthinnerguys_t_all);
	u8g2_DrawStr(&u8g2,0,10,"Hello");
	}while(u8g2_NextPage(&u8g2));	

sleep_ms(2000);
```

This for some rason causes two packets containing 133 bytes. Most of which are 0.

Like: 
```
040100b0fefe3030fefe0f0f8585878700fefe0fefe0f0f81818f8f0000000000000000000000000
Bytes written: 133                                                              
Buf length: 133                                                                 
040100b1330033013333103303301333310000000000000000000000000000000000000000000000
Bytes written: 133                                                              
Buf length: 133                                                                 
040100b2000000000000000000000000000000000000000000000000000000000000000000000000
Bytes written: 133                                                              
Buf length: 133                                                                 
040100b3000000000000000000000000000000000000000000000000000000000000000000000000
Bytes written: 133                                                              
Buf length: 133                                                                 
040100b4000000000000000000000000000000000000000000000000000000000000000000000000
Bytes written: 133                                                              
Buf length: 133                                                                 
040100b5000000000000000000000000000000000000000000000000000000000000000000000000
Bytes written: 133                                                              
Buf length: 133                                                                 
040100b6000000000000000000000000000000000000000000000000000000000000000000000000
Bytes written: 133                                                              
Buf length: 133                                                                 
040100b7000000000000000000000000000000000000000000000000000000000000000000000000
Bytes written: 133       
```

So the loop is causing us grief for some reason.. Any ideas?

olikraus, 2022-07-24:
Here is example which I wrote some time ago. It might serve as an example how it should be: https://github.com/olikraus/lpc804_plu/tree/master/lpc804/u8g2_logo

olikraus, 2022-07-24:
I really wonder how your buffer can exceed 32 bytes. Did you modify the init sequence? Did you modify the cad procedure? The cad procedure is part of the setup. The missing 0 byte can be caused only by a wrong cad procedure.

spi5498, 2022-07-24:
Yeah, it is interesting, I have checked around for other samples.

However, I cannot identify why the initial 00 does not get sent by the library..

The code currently is this:

```
uint8_t u8x8_byte_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)

{

	static uint8_t buffer[34];
	static uint8_t buf_idx;
	uint8_t *data;
	int suc = 0;
	
	switch(msg)
  {
    case U8X8_MSG_BYTE_SEND:
	
	printf("BYTE_SEND: %d",arg_int);
	printf("\n");
	
  
      	data = (uint8_t *)arg_ptr;      
      
      while( arg_int > 0 )
      {
	
	buffer[buf_idx++] = *data;
	
	data++;
	arg_int--;
	} 
       
	break;

    case U8X8_MSG_BYTE_INIT:
      /* add your custom code to init i2c subsystem */
      
      break;
    case U8X8_MSG_BYTE_SET_DC:
      /* ignored for i2c */
      break;
    
    case U8X8_MSG_BYTE_START_TRANSFER:
      buf_idx = 0;
      

      break;
    
    case U8X8_MSG_BYTE_END_TRANSFER:
	
	//Added some debug messages to trace what is happening.
	printf("Buf length: ");
	printf("%d\n",buf_idx);
	if (buf_idx <= 0) break;

	for (int i = 0; i < buf_idx; i++){

	printf("%x",buffer[i]);
      	
	}
	printf("\n");
	suc = i2c_write_blocking(i2c_default, 60, buffer, buf_idx, false);
	printf("Bytes written: %d\n",suc);
	
      break;
    
    default:
      return 0;
  }
  return 1;

}
```

olikraus, 2022-07-24:
I think it IS sent. Better use %02x in the printf statement

spi5498, 2022-07-24:
Sorry, missed your comment regarding the cad procedure.
No, I have not been in any other source file. I even downloaded the library once more just to be sure.

02x did not make any difference. We also can see that the buffer length is only 25. By checking the init procedure in the source file for the display, there are a total of 25 bytes in that procedure alone. So it should be 26 with the initial 00 that should be added from the cad procedure you referred to earlier?

Even if it was not visible in the debug text, it should still be counted when we present the buffer lenght, but it is not.

Also, if I dont "force" a 0 as the initial packet. the SSD1306 is nacking the rest of the commands. So we can be sure that the leading 00 is not added.

And yes, it is really strange that the buffer exceed the 32 limit. 

The init calls from my main.c is:

```
u8g2_t u8g2;

u8g2_Setup_ssd1306_128x64_noname_f(&u8g2,U8G2_R0,u8x8_byte_i2c, u8x8_gpio_and_delay_rp2040);
u8g2_InitDisplay(&u8g2);
sleep_ms(20);
u8g2_SetPowerSave(&u8g2, 0);

```

spi5498, 2022-07-24:
Also,

By appending the 00 in the beginning of: 

```
static const uint8_t u8x8_d_ssd1306_128x64_noname_init_seq[] = {
    
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  
  
  U8X8_C(0x000),
  U8X8_C(0x0ae),		                /* display off */
  U8X8_CA(0x0d5, 0x080),		/* clock divide ratio (0x00=1) and oscillator frequency (0x8) */
  U8X8_CA(0x0a8, 0x03f),		/* multiplex ratio */
  U8X8_CA(0x0d3, 0x000),		/* display offset */
  U8X8_C(0x040),		                /* set display start line to 0 */
  U8X8_CA(0x08d, 0x014),		/* [2] charge pump setting (p62): 0x014 enable, 0x010 disable, SSD1306 only, should be removed for SH1106 */
  U8X8_CA(0x020, 0x000),		/* horizontal addressing mode */
  
  U8X8_C(0x0a1),				/* segment remap a0/a1*/
  U8X8_C(0x0c8),				/* c0: scan dir normal, c8: reverse */
  // Flipmode
  // U8X8_C(0x0a0),				/* segment remap a0/a1*/
  // U8X8_C(0x0c0),				/* c0: scan dir normal, c8: reverse */
  
  U8X8_CA(0x0da, 0x012),		/* com pin HW config, sequential com pin config (bit 4), disable left/right remap (bit 5) */

  U8X8_CA(0x081, 0x0cf), 		/* [2] set contrast control */
  U8X8_CA(0x0d9, 0x0f1), 		/* [2] pre-charge period 0x022/f1*/
  U8X8_CA(0x0db, 0x040), 		/* vcomh deselect level */  
  // if vcomh is 0, then this will give the biggest range for contrast control issue #98
  // restored the old values for the noname constructor, because vcomh=0 will not work for all OLEDs, #116
  
  U8X8_C(0x02e),				/* Deactivate scroll */ 
  U8X8_C(0x0a4),				/* output ram to display */
  U8X8_C(0x0a6),				/* none inverted normal display mode */
    
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

```

We can also force the leading 00 in..

spi5498, 2022-07-24:
So, after further trial and error (By adding the leading 00 manually init section) 

I found that:
u8g2_SetPowerSave(&u8g2, 0);
Is creating only a packet containing one byte "af". Not sure if this should also have a 00 before it?

Anyway, proceeding further with the next command:
u8g2_ClearDisplay(&u8g2);

That function gets the same issue with getting a 128 arg_int.
Maybe you are aware of any difference between these functions that might cause it? Maybe that helps to identify the issue with the large buffer issue with the other commands.

But still no clue as to why the initial 00 does not get added..

olikraus, 2022-07-24:
> 02x did not make any difference.

but we might see better what is going on there.

You mentioned this code:
``` C++
u8g2_Setup_ssd1306_128x64_noname_f(&u8g2,U8G2_R0,u8x8_byte_i2c, u8x8_gpio_and_delay_rp2040);
u8g2_InitDisplay(&u8g2);
sleep_ms(20);
u8g2_SetPowerSave(&u8g2, 0);
```
which should be correct (it is identical to my own code):
https://github.com/olikraus/lpc804_plu/blob/897824574dadb5a1fb76979284b6e48bec20acd6/lpc804/u8g2_logo/u8g2_logo.c#L135-L137

I now see, that you have used the wrong setup procedure. I think it is also not that clear from the description (the none Arduino code is not that good documented), but you need a different setup procedure:

https://github.com/olikraus/u8g2/blob/7395c6857f3ade38a636b67365427178ad528dfa/csrc/u8g2_d_setup.c#L320

``` C++
u8g2_Setup_ssd1306_i2c_128x64_noname_f(&u8g2,U8G2_R0,u8x8_byte_i2c, u8x8_gpio_and_delay_rp2040);
u8g2_InitDisplay(&u8g2);
sleep_ms(20);
u8g2_SetPowerSave(&u8g2, 0);
```

There is that "i2c" after "ssd1306 and before 128x64. Changing this should fix your problems.

olikraus, 2022-07-24:
Hmm, I think I should update https://github.com/olikraus/u8g2/wiki/u8g2setupc

olikraus, 2022-07-24:
Doc update:
https://github.com/olikraus/u8g2/wiki/u8g2setupc#setup-naming-convention

Maybe this makes it more clear to the reader.

spi5498, 2022-07-24:
Wow! Yes! This solved everything :sweat_smile: 

The update to the documentation will really assist anyone further with porting the library to a platform.

Looking at it now, without adding this to the constructor, there is of course no way for the library to choose the correct procedures for I2C. At least I can laugh at it now. Looking back at all the confusion and stupid changes I made in order to try to diagnose the issue is extremely funny :rofl: 

Finally I can go back to working on the port. After this I2C clean up, i only have SPI left.

I will document it as best as I can do and get back to you if you would like to publish the port.

Again, thanks for all the assistance and this great library!

Br,

olikraus, 2022-07-24:
Super. I already did some doc updates. Hope that will be good enough. Thanks for this discussion here.

closed 2022-09-03 by olikraus