olikraus/u8g2 #1916: SSD1309

NA7KR, 2022-07-20:
The display is only shown the top 1/2 of the words.
```
//U8G2_SSD1309_128X64_NONAME0_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 18, /* data=*/ 23, /* cs=*/ 5, /* dc=*/ 16, /* reset=*/ 17);  
U8G2_SSD1309_128X64_NONAME0_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 5, /* dc=*/ 16, /* reset=*/ 17); 

void setup(void) {
  Serial.begin(115200);
  u8g2.begin();
}

void loop(void) {
  currentState = digitalRead(BUTTON_PIN);
  if (boot == false)
    {
      boot = false;
      u8g2.setFont(u8g2_font_ncenB14_tr);
      u8g2.setCursor(0, 14);
      u8g2.print("Hello 1");
      u8g2.setCursor(0, 28);
      u8g2.print("Hello 2");
      u8g2.sendBuffer();
    }
```
Tried different fonts and Cursor settings 
Screen is https://www.amazon.com/dp/B09LND6QJ1?psc=1&ref=ppx_yo2ov_dt_b_product_details

olikraus, 2022-07-20:
Use the F (full buffer) constrictor.

NA7KR, 2022-07-20:
I have tried all 4 1309

NA7KR, 2022-07-20:
I see U8G2_SSD1309_128X64_NONAME0_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 5, /* dc=*/ 16, /* reset=*/ 17); 
I have to edit 1 to F 

Thanks

closed 2022-07-20 by NA7KR

olikraus, 2022-07-20:
Further reading: 

https://github.com/olikraus/u8g2/blob/7395c6857f3ade38a636b67365427178ad528dfa/doc/faq.txt#L134-L142

full buffer drawing:
https://github.com/olikraus/u8g2/wiki/u8g2reference#sendbuffer
page buffer drawing:
https://github.com/olikraus/u8g2/wiki/u8g2reference#firstpage