olikraus/u8g2 #1917: Display shows empty line

klerone, 2022-07-22:
Hello Oliver, great job with this library thank you for sharing your work with us.

I have a problem with an lcd PCD8544 that leave an empty line when printing text or anything other than drawFrame and drawLine.

Here i attach a picture to show you the problem I already look over the internet but cannot find anything related. 

It is connected to 3.3V, hardware SPI, 3.3V @8Mhz atmega328P. 

[image]
[image]
[image]

Also i have the same problem with an LcD with ST7588 over I2C 

I am not sure if this may be a hardware related problem.

olikraus, 2022-07-22:
strange... I have never seen such bugs

klerone, 2022-07-29:
Well, I will get some samples soon to test. I will let you know.

Best regards.

Fran Calero.

closed 2023-02-22 by olikraus