olikraus/u8g2 #1918: Adafruit 128x128 OLED not working with u8g2 on Adafruit Feather ESP32-S3

spiff72, 2022-07-25:
Hello,

I am at my wits end trying to figure out how to get Adafruit's 1.5" 128x128 OLED (SSD1327) working with U8G2 and a new Feather ESP32-S3.  I initially tried using the Adafruit SSD1327 library and I haven't had any success with that one either.

I also tried running Adafruit's SSD1327 library on an ESP32 TTGO TDisplay board (not using the built in TFT display), and I got that to work fine - so I know the OLED board functions.

For U8G2 I am using the Full Buffer demo, and I have tried all of the I2C constructors for SSD1327 128x128 displays.  I have tried each of these "as-is", and with the SCL and SDA pins explicitly defined (4 and 3).

```
//U8G2_SSD1327_EA_W128128_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 5, /* data=*/ 4, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SSD1327_EA_W128128_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);  /* Uno: A4=SDA, A5=SCL, add "u8g2.setBusClock(400000);" into setup() for speedup if possible */
//U8G2_SSD1327_MIDAS_128X128_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE); /* Uno: A4=SDA, A5=SCL, add "u8g2.setBusClock(400000);" into setup() for speedup if possible */

```
Additionally, I saw some mentions of needing to set the I2C address (which for this board is 0x3D - confirmed with WireScan sketch), so I tried adding the following to the setup section (I also tried using 0x7A since the docs say it is supposed to be multiplied by 2):
`u8g2.setI2CAddress(0x3D); // this is placed after the u8g2.begin(); line`  

I don't get any errors, but the display is always blank...

I also added a Serial.println() statement to the loop so I could confirm the sketch was running (it is)...

I'm thinking this is a quirk of this S3 variant of the ESP32 feather and I am just missing something??

Any suggestions?

spiff72, 2022-07-25:
Update:
I was able to get the Adafruit SSD1327 library working now - for some reason, adding an 8 second delay in the setup section (immediately after the Serial.begin(115200) line got it working.

I no longer need to revert to u8g2 - but it would be nice to have more flexibility on fonts and sizes, since the adafruit library text sizes are very limited.  Adding the same delay to this u8g2 sketch didn't make a difference.

Thanks...

olikraus, 2022-07-25:
0x7a should be correct. If Adafruit lib works, then usually the same is true for u8g2.

spiff72, 2022-07-25:
I used 0x3D when using the Adafruit SSD1327 library - but I couldn't get u8g2 to work with either address.

olikraus, 2022-07-25:
Hmmm also wiring could be an issue.

olikraus, 2023-02-22:
Remember to multiply the I2C address with 2, when using the SetI2CAddress function of u8g2!

closed 2023-02-22 by olikraus