olikraus/u8g2 #1921: mui_u8g2_draw_button_utf_pi not defined

loopj, 2022-07-31:
Hey Oli, great library thanks for building this.

I'm trying to define my own custom mui field which will render a "pi" style button, but I notice that `mui_u8g2_draw_button_utf_pi` is not actually implemented in `mui_u8g2.c`, even though it is defined in the header.

Is this intentional? I would love access to some of the "private" drawing functions from mui_u8g2.c to create my own fields.

Best
- j

olikraus, 2022-08-01:
Mui is work in progress. I did a lot of renaming so it might be happened due to that. It is at least not intentional. 
Thanks for the hint. I will look into this.

loopj, 2022-08-01:
Thanks! I’m using Mui on a couple of my projects already and love it.

olikraus, 2022-09-03:
ah, your request is to have "mui_u8g2_draw_button_utf_pi" because it is not there.

olikraus, 2022-09-03:
The goto button is there: "mui_u8g2_btn_goto_w1_pi" what exactly do you need?
Actually your function should be there alread, it is "mui_u8g2_draw_button_pi".
I actually removed all the utf strings, because MUI will only use UTF8.

Ah, I see the problem. Yes indeed it looks like a renaming problem. I will update the header.

closed 2022-09-04 by olikraus

loopj, 2022-09-04:
Thank you! This will allow me to remove a ton of duplication.

olikraus, 2022-09-04:
Great. Note that I have just released 2.33.15, so the update should be there soon in Arduino IDE.