olikraus/u8g2 #1923: (Seems to be pretty basic) nothing shown on the display, STM32L476VGT6 - SSD1306(128*64) - on SPI

Junseok-O, 2022-08-04:
First of all, thanks for your kind and such professional work helping out throughout the world

I may be one of the newest newbie in this field and it seems that the problem is very basic, but I can't just figure what the problem is.

I read many examples based on your u8g2 and this is my configuration. It builds alright but the display doesn't show a thing.

I also tried other guy's simple library just to check if the SSD1306 is broke, but nothing happened. But I'm kinda sure it's not because my boss(went to a business trip and I've got no help. Stayed up all night dealing with this problem) gave me the device.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

below is the main.c
```
/* USER CODE BEGIN Header */
/**
  ******************************************************************************
  * @file           : main.c
  * @brief          : Main program body
  ******************************************************************************
  * @attention
  *
  * Copyright (c) 2022 STMicroelectronics.
  * All rights reserved.
  *
  * This software is licensed under terms that can be found in the LICENSE file
  * in the root directory of this software component.
  * If no LICENSE file comes with this software, it is provided AS-IS.
  *
  ******************************************************************************
  */
/* USER CODE END Header */
/* Includes ------------------------------------------------------------------*/
#include "main.h"

/* Private includes ----------------------------------------------------------*/
/* USER CODE BEGIN Includes */
#include "u8g2.h"
/* USER CODE END Includes */

/* Private typedef -----------------------------------------------------------*/
/* USER CODE BEGIN PTD */

/* USER CODE END PTD */

/* Private define ------------------------------------------------------------*/
/* USER CODE BEGIN PD */
/* USER CODE END PD */

/* Private macro -------------------------------------------------------------*/
/* USER CODE BEGIN PM */

/* USER CODE END PM */

/* Private variables ---------------------------------------------------------*/
RTC_HandleTypeDef hrtc;

SPI_HandleTypeDef hspi1;

UART_HandleTypeDef huart4;

static u8g2_t u8g2;
/* USER CODE BEGIN PV */

/* USER CODE END PV */
[... 358 lines cut ...]

#ifdef  USE_FULL_ASSERT
/**
  * @brief  Reports the name of the source file and the source line number
  *         where the assert_param error has occurred.
  * @param  file: pointer to the source file name
  * @param  line: assert_param error line source number
  * @retval None
  */
void assert_failed(uint8_t *file, uint32_t line)
{
  /* USER CODE BEGIN 6 */
  /* User can add his own implementation to report the file name and line number,
     ex: printf("Wrong parameters value: file %s on line %d\r\n", file, line) */
  /* USER CODE END 6 */
}
#endif /* USE_FULL_ASSERT */
```

I barely fixed any other configurations 'cause I don't know much..
I had to put RTC because eventually I'm planning to show a RTC I coded formerly.

If the question is too ambiguous due to my lack, is there a simple way to check if the SSD1306 is broke?
sorry if this seems like a gibberish, I barely know much about this field and I haven't slept much for weeks.
I totally understand if you can't give any fine answers, I already learned and earned a lot thanks to you!

olikraus, 2022-08-10:
U8g2 is build on top of the Arduino Environment. Maybe you can run STM32duino (https://github.com/stm32duino/Arduino_Core_STM32) with your controller. U8g2 will work without problems under STM32duino.

Junseok-O, 2022-08-23:
Thanks. I struggled on this project for more than 3 weeks now and finally made it!
Actually, the problem seems to be the pins of the boards.. I only fixed the configurations of GPIO pins and it worked

I didn't want to make the project heavier and the STM32duino seemed hard to understand, but your works helped my butt out of my situation. Thanks a lot.

closed 2022-08-23 by Junseok-O

olikraus, 2022-08-23:
:)