olikraus/u8g2 #1925: Can't compile sys/arm-linux/examples/cpp-examples/u8g2_4wire_hw_spi_cpp/u8g2_4wire_hw_spi.cpp

MikePooh, 2022-08-05:
`..u8g2/sys/arm-linux/examples/cpp-examples/u8g2_4wire_hw_spi_cpp/../../../../../sys/arm-linux/port/u8g2port.h:26:2: error: ‘gpio_t’ does not name a type;`
I can't find declaration of gpio_t. Could you please advice where to search?
Also in this file was
```
#include "gpio.h"
#include "i2c.h"
#include "spi.h"
```
Which was changed to
```
#include "linux/gpio.h"
#include "linux/i2c.h"
#include "linux/spi/spidev.h"
```
Is that correct?

olikraus, 2022-08-05:
for reference: https://github.com/olikraus/u8g2/blob/master/sys/arm-linux/port/u8g2port.h

At least I can't say anthing here. This is contributed code...

MikePooh, 2022-08-05:
The problem was in submodule. 
`git submodule update --init --recursive --jobs 8`

closed 2022-08-08 by MikePooh