olikraus/u8g2 #1926: How to send bitmap without using u8g2 library?

gary00k, 2022-08-05:
How can I upload bitmap to ST7567 just using SPI? Could you tell me what data packages to send?

olikraus, 2022-08-05:
This is described on page 30 of the ST7567 data sheet (https://www.crystalfontz.com/controllers/Sitronix/ST7567/)

gary00k, 2022-08-07:
Yes, I firstly looked into datasheet, but honestly, I don't understand it :/

olikraus, 2022-08-07:
I think this is the main purpose of a library: one person tries to understand the technical details, while the users of that library can just benefit from it without knowing the technical details.

I mean you have the choice: use u8g2, then no technical details are required or access your display directly, which requires to learn technical details.

closed 2023-02-22 by olikraus