olikraus/u8g2 #1928: Secondary I2C peripheral does not work on RPi Pico (SH1107)

AlexanderMandera, 2022-08-06:
Hi,

I wanted to use u8g2 with an existing sketch I found on GitHub (originally made for ESP32, but I ported it to the RP2040).
I already designed a board using GPIO14 and 15 which uses the secondary I2C peripheral.

I used the FlipMode sketch to test the functionality before.

To check if the display actually works at all with this library, I connected the display to the primary I2C peripheral with the default pins 4 and 5:
```c
U8G2_SH1107_64X128_1_HW_I2C u8g2(U8G2_R1, /* reset=*/ U8X8_PIN_NONE);
```

The sketch with this constructor **does work** without any further modification.

But if I connect SDA to GPIO14 and SCL to GPIO15 and use this constructor:
```c
U8G2_SH1107_64X128_1_2ND_HW_I2C u8g2(U8G2_R1, /* reset=*/ U8X8_PIN_NONE);
```
and using this setup routine to set the secondary I2C pins:
```c
void setup(void) {
  Serial.begin(115200);
  
  delay(100);

  Wire1.setSDA(14);
  Wire1.setSCL(15);
  u8g2.begin();
  u8g2.setFont(u8g2_font_6x10_tr);
}
```

It does not work at all.
I am using the Arduino Pico Core (https://github.com/earlephilhower/arduino-pico).

I also verified that the display on the secondary peripheral actually works with the Adafruit_SH110X library.
Before I try to change the existing sketch to use a whole new library, I wanted to checkout if you have a solution for this library for me :)

Thank you very much in advance.
~ Alexander

olikraus, 2022-08-06:
The 2nd hw is available if the arduino core provider has set the number of available i2c devices correctly. 
As an alternative solution you might try to use the primary i2c and reassign the pin ins like you did in the example above.
Another (maybe slower) alternative solution might be the use of the software emulated i2c (SW_I2C) u8g2 constructor.

bsramsen, 2022-08-08:
HI Alexander , HI Oli
I am Bernd Schmitt from germany, retired engineer and fan of rpi PICO and the EARLE F PHILHOWER arduino-core..
I did a lot of tests during the last days and my results are the same as Alexanders.
The 2nd HW is not recognized for the PICO

`
#include <Arduino.h>  // Graphics and font library for ST7789 driver chip    waveshare shield pico TFT1.3
#include <TFT_eSPI.h> // Graphics and font library for ST7789 driver chip    waveshare shield pico TFT1.3
#include <SPI.h>
#include <Wire.h>
#include <U8g2lib.h>

/// OLED DEFINITIONS ///
U8G2_SSD1306_128X64_NONAME_F_HW_I2C oled0(U8G2_R0, /* reset=*/U8X8_PIN_NONE);
#define U8X8_HAVE_2ND_HW_SPI
// U8G2_SSD1306_128X64_NONAME_F_2ND_HW_I2C oled1(U8G2_R0, /* reset=*/U8X8_PIN_NONE);
U8G2_SSD1306_128X64_NONAME_F_HW_I2C oled1(U8G2_R0, /* reset=*/U8X8_PIN_NONE);

// the pin definitions for UART,I2C,SPI for the earlephilehower arduino-core can be found here
// C:\Users\User\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.3.3\variants\rpipico/pins_arduino.h       for Arduino IDE
// C:\Users\User\.platformio\packages\framework-arduinopico\variants\rpipico/pins_arduino.h                          for platformIO 
// this is how Wire and Wire1 are defined in the above mentioned datasheets
// #define PIN_WIRE_SDA 4
// #define PIN_WIRE_SCL 5
// #define PIN_WIRE1_SDA 26
// #define PIN_WIRE1_SCL 27
`

I have build a test environment with a nice clock on ST7789 TFT of the waveshare shield pico TFT 1.3 and on a small additional board  there are two headers with I2C0 and I2C1 connection for direct plugin of Oled displays SSD1306.
Oled0 is working fine - Oled 1 is not working when i use the constructor for 2ND HW I2C.
I have checked with a PCF8574 extender , that Wire1 is working correctly - the function is absolutely okay.

I also tried to force the 2ND_HW  function with
#define U8X8_HAVE_2ND_HW_SPI   before calling the 2ND_HW  constructor
but this has no effect - i think your library  is  following its logic and overwrites my definition. 
( but it would be nice to force the 2ND_HW function easy like that ! )

C:\Users\User\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.3.3\variants\rpipico/pins_arduino.h       for Arduino IDE
C:\Users\User\.platformio\packages\framework-arduinopico\variants\rpipico/pins_arduino.h                          for platformIO 

The Earle Philhower core defines the variable GPIOs only in the above mentioned files - you cannot change the definitions in the code itself.. 

So what Alexander mentions in his code   Wire1.setSDA(14) doesnt work and is not accepted by the compiler.
You really have to change the change the definitions in the text-file.
These are the definitions you will find there:
`
#pragma once

// Pin definitions taken from:
//    https://datasheets.raspberrypi.org/pico/pico-datasheet.pdf

// LEDs
#define PIN_LED        (25u)

// Serial
#define PIN_SERIAL1_TX (0u)
#define PIN_SERIAL1_RX (1u)

#define PIN_SERIAL2_TX (8u)
#define PIN_SERIAL2_RX (9u)

// SPI
#define PIN_SPI0_MISO  (16u)
#define PIN_SPI0_MOSI  (19u)
#define PIN_SPI0_SCK   (18u)
#define PIN_SPI0_SS    (17u)

#define PIN_SPI1_MISO  (12u)
#define PIN_SPI1_MOSI  (15u)
#define PIN_SPI1_SCK   (14u)
#define PIN_SPI1_SS    (13u)

// Wire
#define PIN_WIRE0_SDA  (8u)  // 4
#define PIN_WIRE0_SCL  (9u)  // 5

#define PIN_WIRE1_SDA  (6u)  //26
#define PIN_WIRE1_SCL  (7u)  //27

#define SERIAL_HOWMANY (3u)
#define SPI_HOWMANY    (2u)
#define WIRE_HOWMANY   (2u)

#include "../generic/common.h"
` 

And there is also the definition for the number of the I2Cs available on the PICO board:
#define WIRE_HOWMANY   (2u)

I think the U8g2lib should look there for the number of I2Cs  to activate the 2ND_HW  functionality.

As far as i could see - OLI has already integrated this procedure for the Teensy 3.2
It would be fine , to have a solution also for the PICO , which is an UPCOMING STAR in my opinion.

Have a good time OLI and Alexander

[image]
[image]

olikraus, 2022-08-08:
Here is a kind of an "official" use of that macro:
https://github.com/arduino/ArduinoCore-samd/blob/89833081ef18aea430b8943ab653dbbd510d43ce/variants/arduino_zero/variant.h#L150

bsramsen, 2022-08-08:
HI Oli  t
hank you for the quick response.
I added the macro to my codee like that

`
/// OLED DEFINITIONS ///
U8G2_SSD1306_128X64_NONAME_F_HW_I2C oled0(U8G2_R0, /* reset=*/U8X8_PIN_NONE);
#define WIRE_INTERFACES_COUNT 2
U8G2_SSD1306_128X64_NONAME_F_2ND_HW_I2C oled1(U8G2_R0, /* reset=*/U8X8_PIN_NONE);
`
same result - the second displpay on I2c1 is dead.
Where is my fault ???

bsramsen, 2022-08-08:
HI OLI
thank you for you quick response

I added the macro to my code like that:

/// OLED DEFINITIONS ///
U8G2_SSD1306_128X64_NONAME_F_HW_I2C oled0(U8G2_R0, /* reset=*/U8X8_PIN_NONE);
#define WIRE_INTERFACES_COUNT 2
U8G2_SSD1306_128X64_NONAME_F_2ND_HW_I2C oled1(U8G2_R0, /* reset=*/U8X8_PIN_NONE);

Same result - the second display on I2C1 doesnt work .  where is my fault ?

olikraus, 2022-08-08:
The macro Definition needs to be there in the variant.h file of the lib files for your controller

bsramsen, 2022-08-08:
THANKS VERY MUCH OLI
it works perfectly.

I just put the macro at the end of the definition file
 C:\Users\User.platformio\packages\framework-arduinopico\variants\rpipico/pins_arduino.h        in platformIO

[image]

olikraus, 2022-08-08:
🤗👍

nordseele, 2022-11-11:
In case this is useful to anyone: with Platformio, you can also add the 'define' directive in the `platformio.ini`  file of your project, adding the following. 

`build_flags = -DWIRE_INTERFACES_COUNT=2`

Full file:

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
board_build.core = earlephilhower
framework = arduino
lib_deps = olikraus/U8g2@^2.34.5
build_flags = -DWIRE_INTERFACES_COUNT=2
```

closed 2023-02-22 by olikraus