olikraus/u8g2 #1931: Help Request

lewisjluck, 2022-08-12:
Hi Oli,

I am a university student from Australia and am trying to use your library in a project of mine. We are making an electric scale using a strain gauge etc. and are trying to communicate between an LCD and a microcontroller to display various information... thus far with no success.

We are using the Atmega328P microcontroller and are attempting to communicate with this specific LCD display: https://www.buydisplay.com/lcd-display-serial-graphic-display-128x64-st7920-white-on-blue. As far as we can tell this is a standard ST7920 and is supported by your library through SPI protocol. 

We have followed this tutorial (https://www.instructables.com/ST7920-LCD-With-ATmega328-in-Atmel-Studio-Using-SP/) but nothing has been displayed to our LCD. I have poured over it for hours, and multiple different people have reproduced every step independently - I am sure we are following the steps correctly. 

We have connected the pins as described in that tutorial, and have used the following setup command in the main file found on the as7 page of your Github: u8g2_Setup_st7920_s_128x64_f(&u8g2, U8G2_R0, u8x8_byte_4wire_sw_spi, u8x8_avr_gpio_and_delay).

We are honestly lost as to what the issue may be. If you have any guidance we would be very appreciative.

olikraus, 2022-08-12:
There are multiple aspects to consider, here are some:
- Controller Output Pin Level must be 5V
- The ST7920 must be configured for SPI mode
- Wires should be short and soldered (ST7920 is a little bit sensitive here)
- The ST7920 usually requires 5V input voltage
- u8x8_avr_gpio_and_delay needs to implement the SPI messages and route those level to your pins

lewisjluck, 2022-08-12:
Hi, thanks very much for the response!

1. 
2. Could you clarify as to how we would configure it for SPI mode? We have connected the pins appropriately per that tutorial and used the specified setup instructions. 
3. We have been using jumper cables such as those found below in a breadboard. Could this potentially be the problem?
https://core-electronics.com.au/jumper-wires-standard-7-m-m-pack-of-30.html?gclid=CjwKCAjw9NeXBhAMEiwAbaY4ln7OG7wnQIq5-_IcvUoXe5RdsXgdQW0LZu4GV6Tbno_WiQSk_TpIgBoCaKAQAvD_BwE. 
4. We have connected to 5V (though originally we were connecting to 3.3V which we thought might be an issue). 
5. We have set it up as specified, though I am unaware of any steps that involved u8x8_avr_gpio_and_delay. Is internal configuration of that function requried too? 

I have attached the entire main.c file we are using below in Atmel Studio. Thank you so much for your help!

`     /*
  main.c
	
  AVR Test Project
  This project will use 4-Wire SW SPI
  Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)
  Copyright (c) 2018, olikraus@gmail.com
  All rights reserved.
  Redistribution and use in source and binary forms, with or without modification, 
  are permitted provided that the following conditions are met:
  * Redistributions of source code must retain the above copyright notice, this list 
    of conditions and the following disclaimer.
    
  * Redistributions in binary form must reproduce the above copyright notice, this 
    list of conditions and the following disclaimer in the documentation and/or other 
    materials provided with the distribution.
  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND 
  CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, 
  INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF 
  MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE 
  DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR 
  CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, 
  SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT 
  NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; 
  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER 
  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, 
  STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) 
  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF 
  ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.  
	
*/

#include <avr/io.h>
#include <u8g2.h>
#include <util/delay.h>

#define DISPLAY_CLK_DIR DDRB
#define DISPLAY_CLK_PORT PORTB
#define DISPLAY_CLK_PIN 5

#define DISPLAY_DATA_DIR DDRB
#define DISPLAY_DATA_PORT PORTB
#define DISPLAY_DATA_PIN 3

#define DISPLAY_CS_DIR DDRB
#define DISPLAY_CS_PORT PORTB
#define DISPLAY_CS_PIN 2

#define DISPLAY_DC_DIR DDRB
#define DISPLAY_DC_PORT PORTB
#define DISPLAY_DC_PIN 1

#define DISPLAY_RESET_DIR DDRB
#define DISPLAY_RESET_PORT PORTB
#define DISPLAY_RESET_PIN 0

#define P_CPU_NS (1000000000UL / F_CPU)

u8g2_t u8g2;
[... 108 lines cut ...]

}

int main(void)
{
	/*
		Select a setup procedure for your display from here: https://github.com/olikraus/u8g2/wiki/u8g2setupc
		1. Arg: Address of an empty u8g2 structure
		2. Arg: Usually U8G2_R0, others are listed here: https://github.com/olikraus/u8g2/wiki/u8g2reference#carduino-example
		3. Arg: Protocol procedure (u8g2-byte), list is here: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#communication-callback-eg-u8x8_byte_hw_i2c
		4. Arg: Defined in this code itself (see above)
	*/
	
	
	/* full buffer example, setup procedure ends in _f */
	 
	while(1){
		u8g2_Setup_st7920_s_128x64_f(&u8g2, U8G2_R0, u8x8_byte_4wire_sw_spi, u8x8_avr_gpio_and_delay);
		u8g2_InitDisplay(&u8g2);
		u8g2_SetPowerSave(&u8g2, 0);
		u8g2_ClearBuffer(&u8g2);
		u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
		u8g2_DrawStr(&u8g2, 1, 18, "U8g2 on AVR");
		u8g2_SendBuffer(&u8g2);
	}`

olikraus, 2022-08-13:
>  Could you clarify as to how we would configure it for SPI mode? 

There is this PSB pin if I remember correctly, which needs to be set to the correct value

> We have been using jumper cables such as those found below in a breadboard.

These breadboards are very unreliable. Users had been searchin for hours and finally there was a connection issue in the breadboard.

> I am unaware of any steps that involved u8x8_avr_gpio_and_delay.

You need to update the DISPLAY_xxx macros according to your needs and pin connection.

One more thing to consider: Did you add a proper varpot for the contrast setup?

Maybe you could sent a full picture of your setup.

Did you check my tutorial here: https://github.com/olikraus/u8g2/wiki/setup_tutorial
It uses the ST7920...

lewisjluck, 2022-08-17:
Hi Oli, 

Thanks for your response. We have kept trying and still no luck. The PSB pin is set to ground as needed for serial. We have measured the signals sent by the atmega328p and the clock, data etc all seems visually correct at 3.3V communicating. The RST pin has no signal to it, and the RS pin has an seemingly an occasional ping. I believe the connections are correct. 

The macro pins we have configured as above in the code. I think it corresponds correctly to the pins we have connected on PORTB according to the atmega pinout. I have attached a photo of our setup below. We are using a seeduino to power the setup at 3.3V, I don’t believe this should be affecting it in any other way. 

[image]

lewisjluck, 2022-08-17:
Hi Oli, we have fixed it! The potentiometer we were using was faulty. Thank you for your help!

olikraus, 2022-08-17:
👍

lewisjluck, 2022-09-02:
Hi Oli! 

Hope you've been well. We are loving using our display and your library - everything is working great. The only problem is currently we are experiencing significant lag. We have the constructor set to "F" mode (see above). Each time we draw text to the display (we have lines to draw) we call ClearBuffer at the start, DrawStr for each of the lines, then send buffer. Doing this set of functions calls back to back, the display pauses for about 1 second in between showing the two different versions. Our task requires an update rate of at least 2Hz. Any suggestions to make it faster? We are communicating over serial. 

Thanks so much for your help.

olikraus, 2022-09-02:
Difficult to say something without seeing the code.

lewisjluck, 2022-09-02:
I should clarify - even us executing ClearBuffer, DrawStr, Send Buffer, Clear Buffer, Draw Str, Send Buffer immediately after each other (as the only calls other than setup in main) still results in an about 1 second delay in between the first string and udpating to the second. 

Is there a default delay built in?

olikraus, 2022-09-02:
Maybe the 100NANO delay is too large. Can you reduce the delay caused here by reducing the 100UL factor. As long as the display still works it could be reduced.

lewisjluck, 2022-09-07:
Hi Oli,

Unfortunately that did not seem to change much. I took the value down to 1 with no noticeable difference. 

For reference, this is the only code in our main function (with the setup as shown above):
`
int main(void) {
	u8g2_Setup_st7920_s_128x64_f(&u8g2, U8G2_R0, u8x8_byte_4wire_sw_spi, u8x8_avr_gpio_and_delay);
	u8g2_InitDisplay(&u8g2);
	u8g2_SetPowerSave(&u8g2, 0);
	u8g2_SetFont(&u8g2, u8g2_font_pressstart2p_8f);
	while(1) {
		u8g2_ClearBuffer(&u8g2);
		
		u8g2_DrawStr(&u8g2, 1, 9, "LINE 1");
		u8g2_SendBuffer(&u8g2);
		
		u8g2_DrawStr(&u8g2, 1, 19, "LINE 2");
		u8g2_SendBuffer(&u8g2);
		
		u8g2_DrawStr(&u8g2, 2, 29, "LINE 3");
		u8g2_SendBuffer(&u8g2);
	}
	
}
`

I will attach a video of the result. 

 Still seems to be an update rate of just under 1Hz and we require 2Hz. It seems like the issue arises in SendBuffer? As there is a pause between each line displaying.

We really appreciate your help and would love some further suggestions if you can think of anything as the details of the library is still quite confusing to us. 

Lewis

lewisjluck, 2022-09-07:
https://user-images.githubusercontent.com/42509364/188775389-4b828c4d-d0e7-42f5-bb81-eeba18897c7e.MOV

olikraus, 2022-09-07:
Did you try to remove the code from U8X8_MSG_DELAY_100NANO?
I still think that there is an issue with the delays.

lewisjluck, 2022-09-07:
I changed the 100UL in the function to 1, to no effect. Should I delete all the code under case: U8X8_MSG_DELAY_100NANO?

olikraus, 2022-09-07:
It's worth a try

lewisjluck, 2022-09-09:
Hey Oli!
We have made some progress in the speed problem. We hace decided to use updateDisplayArea to make it fast by not sending the whole buffer every time - however, the area we specify only seems to be updating every second line of pixels. Very weird. Code shown below. We are using the update display area function to only update the top left quarter of our LCD (which is 128 pixels wide and 64 pixels high). 

`
u8g2_DrawStr(&u8g2, 1, 9, "Hello");
u8g2_UpdateDisplayArea(&u8g2, 0, 0, 8, 4); 
`

Have you seen this before/have any suggestions?

olikraus, 2022-09-09:
I am not sure whether UpdateArea is fully supported for the st7920

ChrisEasyScan9699, 2024-12-25:
@lewisjluck  have you try  u8g2_UpdateDisplayArea for ST7920 ??