olikraus/u8g2 #1932: Can it be adapted to JLX384160G?

sddtly, 2022-08-12:
ST7586S

sddtly, 2022-08-12:
`
u8g2.drawLine(0, 0, 383, 159);
u8g2.drawLine(0, 159, 383, 0);

u8g2.setFont(u8g2_font_ncenB24_te);
u8g2.drawStr(30,60,"Hello World!");

u8g2.drawFrame(10, 10, 363, 139);
`

U8G2_ST7586S_S028HN118A_F_4W_HW_SPI u8g2(U8G2_R0, 5, 4 ,2);

[image]

sddtly, 2022-08-12:
U8G2_ST7586S_ERC240160_F_4W_HW_SPI u8g2(U8G2_R0, 5, 4 ,2);

[image]

sddtly, 2022-08-12:
U8G2_ST7586S_YMC240160_F_4W_HW_SPI u8g2(U8G2_R0, 5, 4 ,2);

[image]

sddtly, 2022-08-12:
All photos were taken after power off and restart

olikraus, 2022-08-12:
So your request is to get a new constructor for your display, right?
We could use the first constructor as a base for this: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7586s_s028hn118a.c

Is there any datasheet for your display available?

sddtly, 2022-08-12:
thank you but it doesn't seem to work

[u8x8_d_st7586s_s028hn118a.c.txt](https://github.com/olikraus/u8g2/files/9316224/u8x8_d_st7586s_s028hn118a.c.txt)

sddtly, 2022-08-12:
> 谢谢，但它似乎不起作用

谢谢，但它似乎不起作
[u8x8_d_st7586s_s028hn118a.c.txt](https://github.com/olikraus/u8g2/files/9316254/u8x8_d_st7586s_s028hn118a.c.txt)
用

sddtly, 2022-08-12:
> So your request is to get a new constructor for your display, right? We could use the first constructor as a base for this: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7586s_s028hn118a.c
> 
> Is there any datasheet for your display available?

thank you but it doesn't seem to wor
[u8x8_d_st7586s_s028hn118a.c.txt](https://github.com/olikraus/u8g2/files/9316257/u8x8_d_st7586s_s028hn118a.c.txt)
k

sddtly, 2022-08-12:
[JLX384160G-973-PN(不带字库).pdf](https://github.com/olikraus/u8g2/files/9316295/JLX384160G-973-PN.pdf)

sddtly, 2022-08-12:
> 所以你的要求是为你的显示器获取一个新的构造函数，对吧？ 我们可以使用第一个构造函数作为基础：[https ://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7586s_s028hn118a.c](https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7586s_s028hn118a.c)
> 
> 是否有适用于您的显示器的数据表？

[image]
[u8x8_d_st7586s_s028hn118a.c.txt](https://github.com/olikraus/u8g2/files/9328912/u8x8_d_st7586s_s028hn118a.c.txt)

sddtly, 2022-08-12:
已解决，感谢，修改u8g2_m_48_17_f的参数

olikraus, 2022-08-12:
> Solved, thanks, modify the parameters of u8g2_m_48_17_f

It is more complicated. My plan is to update the code in u8g2 accordingly and create a new constructor for you.

sddtly, 2022-08-13:
> 

thank you very much

olikraus, 2022-08-13:
I have created new beta release 2.33.10
It includes the new constructor U8G2_ST7586S_JLX384160...
Can you test the FlipMode example for the page mode examples list? There should be a solid frame at the display border all the time. Remember to put u8g2 into 16 bit mode (see FAQ)

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

sddtly, 2022-08-14:
U8G2_ST7586S_S028HN118A_F_4W_HW_SPI u8g2(U8G2_R0, 5, 4 ,2);

u8g2.drawLine(0, 0, 383, 159);
    u8g2.drawLine(0, 159, 383, 0);

    u8g2.setFont(u8g2_font_ncenB24_te);
    u8g2.drawStr(30,30,"Hello World!");

    u8g2.drawFrame(10, 10, 363, 139);
    u8g2.drawFrame(0, 0, 384, 160);

    u8g2.setFont(u8g2_font_6x13_mr);
    u8g2.setCursor(15, 70);
    u8g2.printf("getDisplayHeight: %d",u8g2.getDisplayHeight());
    u8g2.setCursor(15, 85);
    u8g2.printf("getDisplayWidth:  %d",u8g2.getDisplayWidth());

    u8g2.setCursor(15, 100);
    u8g2.printf("getBufferTileHeight: %d",u8g2.getBufferTileHeight());
    u8g2.setCursor(15, 115);
    u8g2.printf("getBufferTileWidth:  %d",u8g2.getBufferTileWidth());
[image]

sddtly, 2022-08-14:
U8G2_ST7586S_JLX384160_F_4W_HW_SPI u8g2(U8G2_R0, 5, 4 ,2);

[image]
[main.txt](https://github.com/olikraus/u8g2/files/9332665/main.txt)

sddtly, 2022-08-14:
u8x8_d_st7586s_jlx384160.c    
line 173            U8X8_A(0x087), // should be 0x87 but caused flicker 0x9F 
Normal after changing 0x87 to 0x9f

sddtly, 2022-08-14:
> FlipMode
谢谢你，现在看起来一且正常
[image]

olikraus, 2022-08-14:
Ah, I will change this, thanks

sddtly, 2022-08-14:
> thanks

thanks

sddtly, 2022-08-14:
> Ah, I will change this, thanks

> thanks

> Ah, I will change this, thanks

thanks

sddtly, 2022-08-14:
> Ah, I will change this, thanks

thanks

olikraus, 2022-08-14:
I have created beta 2.33.11

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

sddtly, 2022-08-15:
> I have created beta 2.33.11
>
> You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip Arduino IDE:
>
> 1. Remove the existing U8g2_Arduino [...]

olikraus/U8g2@^2.33.11
very perfect，thank you for your dedication
[image]

olikraus, 2022-08-15:
nice, thanks for testing

closed 2022-09-03 by olikraus

mspudi, 2023-07-30:
> > 我已经创建了测试版 2.33.11
> > 您可以从这里下载最新的U8g2测试版：https://github.com/olikraus/U8g2_Arduino/archive/master.zip Arduino IDE：
> > 
> > 1. 删除现有的U8g2_Arduino库 （https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
> > 2. 通过Arduino IDE安装U8g2_Arduino Zip文件，添加zip库菜单（https://www.arduino.cc/en/Guide/Libraries）。
> > 
> > PlatformIO：（https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps）应包括`platformio.ini`
> > ```
> > lib_deps =
> >   u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
> > ```
> 
> 奥利克劳斯/U8g2@^2.33.11 非常完美，感谢您的奉献 [image]

这个怎么操作成功的，我始终点不亮，能分享一下TEST么